evcc-io/evcc #13264: Support for Fritz!Powerline Smart Plug 546E

Kugelfang666, 2024-04-02:
**Is your feature request related to a problem? Please describe.**
Support for FRITZ!Powerline 546E Smart Plug with energy metering

**Describe the solution you'd like**
It would be great to have these supported similar to the FRITZ!Dect based plugs. 

**Describe alternatives you've considered**
 cannot reach the garage with Wifi based plugs. 

**Additional context**
I'm using several FRITZ!Powerline 546E based plugs to control the charge of my e-bikes.

andig, 2024-04-02:
I guess they should follow exactly the same API and hence be supported already?

Kugelfang666, 2024-04-02:
Thanks for looking into this @andig 
If I compare the AIN in the docue with what I see in the GUI of the Powerline adapter is looks different. The AIN of my smart Plug looks more like a mac address 6xHEX with two digits and : as separator

closed 2024-04-03 by andig

Kugelfang666, 2024-04-03:
You were right, seems to be working!