evcc-io/evcc #13274: Abweichende kWh-Werte in der Auflistung der Ladevorgänge

SolarPowerEV, 2024-04-03:
### Describe the bug

Wenn man sich die Ladevorgänge anschaut, dann weichen die Werte für die geladenen kWh, die von evcc angezeigt werden, von den tatsächlichen Werten sichtbar ab ohne dass ich einen guten Grund dafür erkennen kann.

Hier mal ein Screenshot als Beispiel. evcc zeigt unter "geladen" einen Wert von 10,2 kWh an. Tatsächlich ist der geladene Wert aber laut AlfenWallbox 10,72 kWh gewesen. Dies weiß evcc eigentlich auch, wie die beiden Werte unter Zählerstand zeigen, denn 96,6 kWh minus 85,9 kWh ergibt eine Differenz von 10,7 kWh. Wie also evcc auf die Idee kommt, nur 10,2 kWh anzuzeigen, ist für mich nicht nachvollziehbar. Wenn ein Ladevorgang halbwegs stabil gelaufen ist, ist diese Abweichung noch halbwegs im Rahmen. Gab es aber Probleme (z.B. weil die Alfen Schwierigkeiten mit der Phasenumschaltung hatte und evcc dann ständig zwischen einer dreiphasigen 4,2 kW-Ladung und einem Abschalten wegen nicht ausreichendem PV-Überschuss hin- und hergesprungen ist, dann werden die Abweichungen sogar noch größer. Da lag die Abweichung dann bei 1,1 kWh.

Hier mal zwei Screenshots zur Verdeutlichung:
[image]

[image]

Was ich nicht ganz verstehe ist, wieso evcc nicht einfach die von der Wallbox ermittelte Differenz aus den beiden Zählerständen nimmt, sondern sich den Wert anscheinend selbst irgendwie zusammenrechnet. Und dabei eben anscheinend doch größere Abweichungen entstehen können. Davon abgeleitete Werte wie z.B. der Anteil der Sonne am Ladestrom sind dann vermutlich entsprechend auch nicht ganz korrekt, denn in dem Fall, wo die Wallbox bei Phasenumschaltung rumgezickt hat, ist leider überwiegend kein Überschuss geladen worden, weswegen evcc ja dann den Ladevorgang auch immer wieder abgebrochen hat.

### Steps to reproduce

1. Ein paar Mal mit evcc laden und die Abweichungen tauchen in der Ladeübersicht auf (zumindest bei Alfen-Wallboxen)
2.
3.
...

### Configuration details

```yaml
Vermutlich nicht relevant für diese Frage
```

### Log details

```text
Vermutlich nicht relevant für diese Frage
```

### What type of operating system are you running?

Docker container

### Version

v0.124.10

andig, 2024-04-03:
https://github.com/evcc-io/evcc/issues/12980

closed 2024-04-03 by andig

SolarPowerEV, 2024-04-03:
Ok, offensichtlich haben also auch andere dasselbe Problem. Sogar mit einer anderen Wallbox. 
Nur kurz die Frage: wieso verwendet evcc für das Logfile nicht einfach die Differenz aus dem Startwert und dem Endwert des Zählers der Wallbox? Die Werte sollten doch eigentlich immer stimmen und wenn die Wallbox hier Unsinn produzieren sollte, dann hat der Nutzer noch ganz andere Probleme und auf jeden Fall ist es dann auch kein Problem von evcc.