evcc-io/evcc #13292: Volvo classic API broken

oskarfessel, 2024-04-04:
### Describe the bug

The Volvo classic/legacy API has problems with the SSL certificate.

### Steps to reproduce

1. connect Volvo legacy hybrid vehicle (V60 2016)
2. 
3.
...

### Configuration details

```yaml
[main  ] INFO 2024/04/04 16:13:04 evcc 0.124.10 (d3ae3350)
[main  ] INFO 2024/04/04 16:13:04 using config file: /etc/evcc.yaml
[db    ] INFO 2024/04/04 16:13:04 using sqlite database: /root/.evcc/evcc.db
[mqtt  ] INFO 2024/04/04 16:13:04 connecting evcc-1753000401 at tcp://192.168.100.138:1883
[main  ] ERROR 2024/04/04 16:13:04 creating vehicle ev1 failed: cannot create vehicle type 'template': cannot create vehicle type 'volvo': cannot get vehicles: Get "https://vocapi.wirelesscar.net/customerapi/rest/v3.0/customerac
counts": tls: failed to verify certificate: x509: certificate is valid for *.qmatic.cloud, *.de.qmatic.cloud, *.se.qmatic.cloud, *.us.qmatic.cloud, *.uk.qmatic.cloud, *.nl.qmatic.cloud, *.be.qmatic.cloud, *.dk.qmatic.cloud, *.no
.qmatic.cloud, *.fr.qmatic.cloud, *.es.qmatic.cloud, *.fi.qmatic.cloud, qmatic.cloud, not vocapi.wirelesscar.net
  
Konfiguration (/etc/evcc.yaml):

network:
  # schema is the HTTP schema
  # setting to `https` does not enable https, it only changes the way URLs are generated
  schema: http
  # host is the hostname or IP address
  # if the host name contains a `.local` suffix, the name will be announced on MDNS
  # docker: MDNS announcements don't work. host must be set to the docker host's name.
  host: evcc.local
  # port is the listening port for UI and api
  # evcc will listen on all available interfaces
  port: 7070
  
interval: 10s # control cycle interval

log: info

mqtt:
  broker: 192.168.100.138:1883
  topic: evcc # root topic for publishing, set empty to disable publishing
  # clientid: foo
  # user: *****
  # password: *****
  
meters:
- type: template
  template: homematic
  usage: grid
  host: 83.97.41.196  # IP-Adresse oder Hostname
  device: 'SEQ1375070' # Homematic Ger\xc3\xa4te Id, wie im CCU Webfrontend angezeigt.
  user: *****
  password: *****
  name: my_grid

- name: sma_grid
  type: template
  template: sma-home-manager
  usage: grid
  host: 192.168.100.253 # IP-Adresse oder Hostname
  interface: xennet1

  
- type: template
  template: sma-hybrid
  name: pv2bat
  host: 192.168.100.99
  usage: battery
  modbus: tcpip
  capacity: 16.6
  id: 3 # Configured Modbus Device ID (SMA default ID 3 = 126)
      
- type: template
  template: sma-hybrid
  id: 3
  host: 192.168.100.99
  port: 502
  usage: pv
  modbus: tcpip
  name: pv3

hems:
  type: sma 
  deviceid: *****
  allowcontrol: false
  
  
chargers:
- type: template
  template: wallbe-meter
  host: 192.168.0.8
  port: 502
  name: wallbox2

vehicles:
- type: template
  template: volvo
  title: UNIX43E
  icon: car
  user: ***** 
  password: *****
  vin: *****
  mode: pv
  capacity: 11
  phases: 1
  maxcurrent: 14
  name: ev1
  
  
  
loadpoints:
- title: Garage
  charger: wallbox2
  mode: pv
  
  
  
tariffs:
  currency: EUR # three letter ISO-4217 currency code (default EUR)
  grid: 
    # either static grid price (or price zones)
    type: fixed
    price: 0.2820 # EUR/kWh
    zones:
      - days: Mo-Fr
        hours: 2-5
        price: 0.2820 # EUR/kWh
      - days: Sa,So
        price: 0.2820 # EUR/kWh
  feedin:
    # rate for feeding excess (pv) energy to the grid
    type: fixed
    price: 0.08 # EUR/kWh
  co2:
    # co2 tariff provides co2 intensity forecast and is for co2-optimized target charging if no variable grid tariff is specified
    type: gr\xc3\xbcnstromindex # Gr\xc3\xbcnStromIndex (Germany only)
    zip: 59427

    # type: electricitymaps # https://app.electricitymaps.com/map
    # uri: <uri>
    # token: *****
    # zone: DE

site:
  title: Mein Zuhause
  residualPower: 100
  meters:
    grid: sma_grid
    battery:
      - pv2bat
    pv:
      - pv3

Version: `0.124.10 (d3ae3350)`
```

### Log details

```text
evcc --log debug vehicle
[main  ] INFO 2024/04/04 16:07:34 evcc 0.124.10 (d3ae3350)
[main  ] INFO 2024/04/04 16:07:34 using config file: /etc/evcc.yaml
[db    ] INFO 2024/04/04 16:07:34 using sqlite database: /root/.evcc/evcc.db
[mqtt  ] INFO 2024/04/04 16:07:34 connecting evcc-1620153774 at tcp://192.168.100.138:1883
[mqtt  ] DEBUG 2024/04/04 16:07:34 tcp://192.168.100.138:1883 connected
[main  ] ERROR 2024/04/04 16:07:34 creating vehicle ev1 failed: cannot create vehicle type 'template': cannot create vehicle type 'volvo': cannot get vehicles: Get "https://vocapi.wirelesscar.net/customerapi/rest/v3.0/customeraccounts": tls: failed to verify certificate: x509: certificate is valid for *.qmatic.cloud, *.de.qmatic.cloud, *.se.qmatic.cloud, *.us.qmatic.cloud, *.uk.qmatic.cloud, *.nl.qmatic.cloud, *.be.qmatic.cloud, *.dk.qmatic.cloud, *.no.qmatic.cloud, *.fr.qmatic.cloud, *.es.qmatic.cloud, *.fi.qmatic.cloud, qmatic.cloud, not vocapi.wirelesscar.net
ev1
---
.Soc:     vehicle not available: cannot create vehicle type 'template': cannot create vehicle type 'volvo': cannot get vehicles: Get "https://vocapi.wirelesscar.net/customerapi/rest/v3.0/customeraccounts": tls: failed to verify certificate: x509: certificate is valid for *.qmatic.cloud, *.de.qmatic.cloud, *.se.qmatic.cloud, *.us.qmatic.cloud, *.uk.qmatic.cloud, *.nl.qmatic.cloud, *.be.qmatic.cloud, *.dk.qmatic.cloud, *.no.qmatic.cloud, *.fr.qmatic.cloud, *.es.qmatic.cloud, *.fi.qmatic.cloud, qmatic.cloud, not vocapi.wirelesscar.net
Capacity: 11.0kWh
Features: [Offline Retryable]
```

### What type of operating system are you running?

Linux

### Version

0.124.10 (d3ae3350)

andig, 2024-04-04:
Is there any open source solution that still works for Volvo? Our request for partner account has not been answered by Volvo. I'd suggest to reach out to your dealer and complain...

0x3d13f, 2024-04-04:
I can connect to URL https://vocapi.wirelesscar.net/customerapi/rest/v3.0/customeraccounts with curl here without a TLS problem.

closed 2024-04-04 by oskarfessel