evcc-io/evcc #13297: ENTSO-e Fehler: No data for resolution: PT60M

arboeh, 2024-04-05:
### Describe the bug

Ich bekomme beim Aktivieren des Tarfis "entsoe" folgenden Fehler im Log:

```
[entsoe] ERROR 2024/04/05 07:20:54 no data for resolution: PT60M
[main  ] ERROR 2024/04/05 07:20:54 failed configuring grid tariff: cannot create tariff type 'entsoe': no data for resolution: PT60M
```

### Steps to reproduce

1. Aktivieren des flexiblem Tarifs "entsoe" in der Konfiguration

### Configuration details

```yaml
tariffs:
  currency: EUR
  grid:
    type: entsoe
    domain: BZN|DE-L
    securitytoken: 956XXXXXXXXXX
    charges: 0.11698
    tax: 0.19
  feedin:
    type: fixed
    price: 0.086 # EUR/kWh
  planner:
    type: grünstromindex
    zip: 5XXX
```

### Log details

```text
[entsoe] ERROR 2024/04/05 07:20:54 no data for resolution: PT60M
[main  ] ERROR 2024/04/05 07:20:54 failed configuring grid tariff: cannot create tariff type 'entsoe': no data for resolution: PT60M
```

### What type of operating system are you running?

Linux

### Version

0.124.10

andig, 2024-04-05:
Bitte trace log

arboeh, 2024-04-05:
Ja, korrekt. current_electricity_market_price war undefined.