evcc-io/evcc #13301: Docu Easee wallbox should be changed

jekruege, 2024-04-05:
### Describe the bug

I need a lot of time as new user to identify the problems with the easee wallbox.

The website and initial settings of the easee wallbox should be changed. Timeout 10s is the problem.

chargers:
  - name: my_charger
    type: template
    template: easee
    user: # Emailadresse
    password: # wie Login für Easee App oder Web Portal (https://easee.cloud)
    charger: EH______`
    timeout: **1m** # optional
    authorize: # Steuert ob evcc die Authentifizierung am Charger vornimmt. Vorteil ist

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
-
```

### Log details

```text
-
```

### What type of operating system are you running?

Linux

GrimmiMeloni, 2024-04-06:
I agree that the current default of 10s is too short. 
1m on the flipside though is too long. The timeout is for a blocking operation, i.e. given 1m this might mean the entire system gets stuck for 1m if easee has an issue.
I switched to 20s timeout sometime in summer last year and this made things pretty stable, even on days when the easee cloud is seriously acting up.

jekruege, 2024-04-06:
Easee blocked my wallbox, if I change the timeout below 1m. Thats happened if the charging rate changes often during cloudy weather. With 1m it works perfect. 

I have then to cut the fuse for an hour. 

I tried a lot if times with 20s, 30s and 40s. 

My global evcc timeout is 15s.

GrimmiMeloni, 2024-04-07:
I think there's a misunderstanding here.
The timeout does not change how often evcc talks to the wallbox. That is controlled by the interval.
The timeout defines how long evcc will wait for an answer from the wallbox for any given command.

PR #13321 (2024-04-07): Easee: change default timeout for cloud commands to 20s
fix #13301
closed 2024-04-07 by andig