evcc-io/evcc #13303: Easee: inconsistent charger state after restart

GrimmiMeloni, 2024-04-05:
> > Scheint also irgendwas in der Kommunikation EVCC mit der Easee Wallbox zu sein.
> 
> Nein, das ist so nicht korrekt. evcc kann nicht direkt mit der Easee Wallbox kommunizieren, da sie kein lokales Interface hat. Kommandos werden von evcc an die Easee Cloud gesendet, und diese leitet diese an die Wallboxen weiter. Wenn die Box vom Strom getrennt und wieder angeschlossen wird, verändert sie Ihren Zustand (in deinem Beispiel also z.b. Wechsel von 1p auf 3p (default)). Wenn die Easee Ihren Zustand nach Neustart korrekt zurück an die Cloud announced, bekommt evcc das auch mit, und würde dann zumindest den Mismatch erkennen. Wann die Wallbox Ihren Zustand nach Neustart ankündigt, kannst du in den Logs erkennen. Ich habe nach einem Reset der Box schon alles zwischen 3 und 15 Minuten gesehen. Da kommt dann _irgendwann_ ein ganzer Schwall an PRODUCT_UPDATE Nachrichten. Du wirst _Dein_ Problem vermutlich lösen können, in dem Du einfach längere Zeit zwischen dem Einschalten der WB und dem Ladestart verstreichen lässt. Alternativ kannst du diese Fehlerquelle komplett ausschließen, indem Du die Box mit der Easee Cloud verbunden lässt.

Ich wollte da jetzt gar niemanden angreifen, dachte eher, dass es im Sinne der **Robustheit** vielleicht hilft den **UseCase „die Wallbox ist nicht dauerversorgt“** mit zu betrachten. Tut mir leid wenn das falsch rüberkam. 
Mir ist klar, dass die Kommunikation über die Cloud geht, wollte im Sinne der Vereinfachung nicht die ganze Kommunikationskette aufschreiben. Korrekt hätte ich sagen müssen in der Kommunikation zwischen Easee-Cloud und lokalem EVCC. Zwischen Neustart der Wallbox und Ladebeginn EVCC-gesteuert liegen locker 30-60 Minuten in meiner Konfiguration. 
Ich habe gestern Abend für mich eine Lösung gefunden, indem ich EVCC automatisiert neu starte nachdem die Wallbox erneut eingeschaltet/mit Spannung versorgt wird. Ich kann sehr gut damit leben, dass es **mein Problem** ist, danke aber für den deutlichen Hinweis. Mein Ansinnen war eher zur Robustheit von EVCC beizutragen für Real-Life UseCases. Habe ich verstanden, ist nicht von Interesse, damit fine.
Damit funktioniert dann auch die Phasenumschaltung. 
Versteh mich nicht falsch: die **Easee Ansteuerung** per EVCC grundsätzlich **hat ja funktioniert**, **die Phasenumschaltung jedoch nicht**, weshalb ich’s **_hier_** mit angemerkt habe.  
**Auffällig war dass der EVCC-Neustart die Phasenumschaltung repariert hat.**

_Originally posted by @MrIngenieur in https://github.com/evcc-io/evcc/discussions/13258#discussioncomment-9018505_

andig, 2024-04-05:
@GrimmiMeloni was wäre hier der Bug?

GrimmiMeloni, 2024-04-05:
@TheFreaker86 bitte neues Issue.

@MrIngenieur ich hab das hier getestet. Das ist ein Easee Problem. Auch meine Box (die sogar authentifizieren muss! 😱 ) lädt nach dem Reboot einfach los. Und das sogar ohne irgendwelche Status Updates wie SignalR. Im evcc Log kann ich es 
lediglich erkennen weil mein Grid Meter auf 11kW++ steil geht. 

Ich mache hier direkt zu, denn da können wir mal so gar nix machen.

closed 2024-04-05 by GrimmiMeloni