evcc-io/evcc #13312: KIA Charge status: not available with nightly build evcc 0.124.10 (f78913ea)

ilion1, 2024-04-05:
### Describe the bug

After updating to the current nightly build the vehicle status cannot be detected anymore. This results in missing car identification during charging (guest vehicle) and error messages. Attached are log files for the different versions.

```
evcc 0.124.10 (f78913ea)
...
Soc:           34%
Capacity:      64.0kWh
Charge status: not available
Range:         145km
Odometer:      47539km
Finish time:   2024-04-06 00:40:00 +0200 CEST
Position:      12.345,12.345
Limit Soc:     80%
Features:      []
[eebus ] DEBUG 2024/04/05 21:00:33 mdns: stop announcement

evcc 0.124.10
...
Soc:           34%
Capacity:      64.0kWh
Charge status: A
Range:         145km
Odometer:      47539km
Finish time:   2024-04-06 00:40:00 +0200 CEST
Position:      12.345,12.345
Target Soc:    80%
Features:      []
[eebus ] DEBUG 2024/04/05 21:00:09 mdns: stop announcement

```
[log-0.124.10+1712283176.txt](https://github.com/evcc-io/evcc/files/14889935/log-0.124.10%2B1712283176.txt)
[log-0.124.10.txt](https://github.com/evcc-io/evcc/files/14889936/log-0.124.10.txt)

### Steps to reproduce

Run evcc vehicle with different versions of evcc

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: info
levels:
  cache: error
#  eebus: trace

# unique installation id
plant: xxxx

interval: 10s # control cycle interval

meters:
- type: template
  template: sonnenbatterie 
  usage: grid  
  host: 192.168.189.38  
  port: 8080  
  name: grid1
- type: template
  template: sonnenbatterie 
  usage: pv  
  host: 192.168.189.38  
  port: 8080  
  name: pv2
- type: template
  template: sonnenbatterie 
  usage: battery  
  host: 192.168.189.38  
  port: 8080  
  name: battery3

chargers:
- type: template
  template: ellipro 
  ski: D9A2 03D9 E918 F950 EC97 E593 7251 E179 79A3 905E  
  ip: 192.168.189.34  
  name: wallbox5

vehicles:
- type: template
  template: kia 
  title: KIA e-Niro  
  user: xyz@xyz.com  
  password: xxxxxxxxx  
  capacity: 64  
  language: en  
  name: ev4

loadpoints:
- title: Carport
  charger: wallbox5
  mode: pv

site:
  title: Home
  residualPower: 100
  meters:
    grid: grid1
    pv:
    - pv2
    battery:
    - battery3

eebus:
  certificate:
....
```

### Log details

```text
see attached files above
```

### What type of operating system are you running?

Linux

### Version

evcc 0.124.10 (f78913ea)

pandamouse84, 2024-04-05:
Having a similar problem with charging, having a KIA and Elli Wallbox, too.

I think the problem is the connection to the Elli wallbox, as I can find the following entries in the log (repeating every 30 seconds):

`
lp-1  ] ERROR 2024/04/05 23:17:28 charge meter: data not available
[lp-1  ] ERROR 2024/04/05 23:17:28 charge meter: not available
[lp-1  ] ERROR 2024/04/05 23:17:58 charge meter: data not available
[lp-1  ] ERROR 2024/04/05 23:17:58 charge meter: not available
`

pandamouse84, 2024-04-05:
Additional Debug Log:

``` Log
[eebus ] DEBUG 2024/04/05 23:36:50 Send: d:_i:47859_Elli-Wallbox-XXXXXXXXXX notify 168 DeviceDiagnosisHeartbeatData
[eebus ] DEBUG 2024/04/05 23:36:51 Send: d:_i:47859_Elli-Wallbox-XXXXXXXXXX notify 169 DeviceDiagnosisHeartbeatData
[eebus ] DEBUG 2024/04/05 23:36:52 Send: d:_i:47859_Elli-Wallbox-XXXXXXXXXX notify 170 DeviceDiagnosisHeartbeatData
[eebus ] DEBUG 2024/04/05 23:36:53 Send: d:_i:47859_Elli-Wallbox-XXXXXXXXXX notify 171 DeviceDiagnosisHeartbeatData
[eebus ] DEBUG 2024/04/05 23:36:53 Send: d:_i:47859_Elli-Wallbox-XXXXXXXXXX notify 172 DeviceDiagnosisHeartbeatData
[eebus ] DEBUG 2024/04/05 23:36:54 Send: d:_i:47859_Elli-Wallbox-XXXXXXXXXX notify 173 DeviceDiagnosisHeartbeatData
[eebus ] DEBUG 2024/04/05 23:36:55 Send: d:_i:47859_Elli-Wallbox-XXXXXXXXXX notify 174 DeviceDiagnosisHeartbeatData
[eebus ] DEBUG 2024/04/05 23:36:56 Send: d:_i:47859_Elli-Wallbox-XXXXXXXXXX notify 175 DeviceDiagnosisHeartbeatData
[eebus ] DEBUG 2024/04/05 23:36:57 Send: d:_i:47859_Elli-Wallbox-XXXXXXXXXX notify 176 DeviceDiagnosisHeartbeatData
[eebus ] DEBUG 2024/04/05 23:36:57 Send: d:_i:47859_Elli-Wallbox-XXXXXXXXXX notify 177 DeviceDiagnosisHeartbeatData
[eebus ] DEBUG 2024/04/05 23:36:58 Send: d:_i:47859_Elli-Wallbox-XXXXXXXXXX notify 178 DeviceDiagnosisHeartbeatData
[eebus ] DEBUG 2024/04/05 23:36:59 Send: d:_i:47859_Elli-Wallbox-XXXXXXXXXX notify 179 DeviceDiagnosisHeartbeatData
[eebus ] DEBUG 2024/04/05 23:37:00 Send: d:_i:47859_Elli-Wallbox-XXXXXXXXXX notify 180 DeviceDiagnosisHeartbeatData
[eebus ] DEBUG 2024/04/05 23:37:01 Send: d:_i:47859_Elli-Wallbox-XXXXXXXXXX notify 181 DeviceDiagnosisHeartbeatData
[eebus ] DEBUG 2024/04/05 23:37:01 Send: d:_i:47859_Elli-Wallbox-XXXXXXXXXX notify 182 DeviceDiagnosisHeartbeatData
[eebus ] DEBUG 2024/04/05 23:37:02 Send: d:_i:47859_Elli-Wallbox-XXXXXXXXXX notify 183 DeviceDiagnosisHeartbeatData
[eebus ] DEBUG 2024/04/05 23:37:03 Send: d:_i:47859_Elli-Wallbox-XXXXXXXXXX notify 184 DeviceDiagnosisHeartbeatData
[eebus ] DEBUG 2024/04/05 23:37:04 Send: d:_i:47859_Elli-Wallbox-XXXXXXXXXX notify 185 DeviceDiagnosisHeartbeatData
[eebus ] DEBUG 2024/04/05 23:37:05 Send: d:_i:47859_Elli-Wallbox-XXXXXXXXXX notify 186 DeviceDiagnosisHeartbeatData
[eebus ] DEBUG 2024/04/05 23:37:05 Send: d:_i:47859_Elli-Wallbox-XXXXXXXXXX notify 187 DeviceDiagnosisHeartbeatData
[eebus ] DEBUG 2024/04/05 23:37:06 Send: d:_i:47859_Elli-Wallbox-XXXXXXXXXX notify 188 DeviceDiagnosisHeartbeatData
[eebus ] DEBUG 2024/04/05 23:37:07 Send: d:_i:47859_Elli-Wallbox-XXXXXXXXXX notify 189 DeviceDiagnosisHeartbeatData
[eebus ] DEBUG 2024/04/05 23:37:08 Send: d:_i:47859_Elli-Wallbox-XXXXXXXXXX notify 190 DeviceDiagnosisHeartbeatData
[eebus ] DEBUG 2024/04/05 23:37:09 Send: d:_i:47859_Elli-Wallbox-XXXXXXXXXX notify 191 DeviceDiagnosisHeartbeatData
[eebus ] DEBUG 2024/04/05 23:37:09 Send: d:_i:47859_Elli-Wallbox-XXXXXXXXXX notify 192 DeviceDiagnosisHeartbeatData
[eebus ] DEBUG 2024/04/05 23:37:10 Send: d:_i:47859_Elli-Wallbox-XXXXXXXXXX notify 193 DeviceDiagnosisHeartbeatData
[eebus ] DEBUG 2024/04/05 23:37:11 Send: d:_i:47859_Elli-Wallbox-XXXXXXXXXX notify 194 DeviceDiagnosisHeartbeatData
[eebus ] DEBUG 2024/04/05 23:37:12 Send: d:_i:47859_Elli-Wallbox-XXXXXXXXXX notify 195 DeviceDiagnosisHeartbeatData
[eebus ] DEBUG 2024/04/05 23:37:13 Send: d:_i:47859_Elli-Wallbox-XXXXXXXXXX notify 196 DeviceDiagnosisHeartbeatData
[eebus ] DEBUG 2024/04/05 23:37:13 Send: d:_i:47859_Elli-Wallbox-XXXXXXXXXX notify 197 DeviceDiagnosisHeartbeatData
[eebus ] DEBUG 2024/04/05 23:37:14 Send: d:_i:47859_Elli-Wallbox-XXXXXXXXXX notify 198 DeviceDiagnosisHeartbeatData
[eebus ] DEBUG 2024/04/05 23:37:15 Send: d:_i:47859_Elli-Wallbox-XXXXXXXXXX notify 199 DeviceDiagnosisHeartbeatData
[eebus ] DEBUG 2024/04/05 23:37:16 Send: d:_i:47859_Elli-Wallbox-XXXXXXXXXX notify 200 DeviceDiagnosisHeartbeatData
[eebus ] DEBUG 2024/04/05 23:37:17 Send: d:_i:47859_Elli-Wallbox-XXXXXXXXXX notify 201 DeviceDiagnosisHeartbeatData
[eebus ] DEBUG 2024/04/05 23:37:17 Send: d:_i:47859_Elli-Wallbox-XXXXXXXXXX notify 202 DeviceDiagnosisHeartbeatData
[eebus ] DEBUG 2024/04/05 23:37:18 Send: d:_i:47859_Elli-Wallbox-XXXXXXXXXX notify 203 DeviceDiagnosisHeartbeatData
[eebus ] DEBUG 2024/04/05 23:37:19 Send: d:_i:47859_Elli-Wallbox-XXXXXXXXXX notify 204 DeviceDiagnosisHeartbeatData
[site  ] DEBUG 2024/04/05 23:37:20 ----
[eebus ] DEBUG 2024/04/05 23:37:20 Send: d:_i:47859_Elli-Wallbox-XXXXXXXXXX notify 205 DeviceDiagnosisHeartbeatData
[lp-1  ] ERROR 2024/04/05 23:37:20 charge meter: data not available
[site  ] DEBUG 2024/04/05 23:37:20 pv power: 0W
[site  ] DEBUG 2024/04/05 23:37:20 battery soc: 0%
[site  ] DEBUG 2024/04/05 23:37:20 battery power: 0W
[site  ] DEBUG 2024/04/05 23:37:20 grid meter: 402W
[site  ] DEBUG 2024/04/05 23:37:20 site power: 502W
[lp-1  ] ERROR 2024/04/05 23:37:20 charge meter: not available
[lp-1  ] DEBUG 2024/04/05 23:37:20 charger status: B
[lp-1  ] DEBUG 2024/04/05 23:37:20 limitSoc reached: 90.0% > 80%
```

VolkerK62, 2024-04-06:
Bei mir (Hyundai) ist der Status ebenfalls nicht vorhanden.
Da ist wohl mit https://github.com/evcc-io/evcc/pull/13259 etwas schiefgegangen.

VolkerK62, 2024-04-06:
@andig 
kann es sein dass du bei https://github.com/evcc-io/evcc/commit/84fb6b1580fef49ce62f48cc05a9ebd3940d8907
vergessen hast, diese Zeile ebenfalls wieder rauszunehmen?
https://github.com/evcc-io/evcc/blob/c00a5b14c23f4abba959997cb4d9fd4f84e2a9ed/vehicle/bluelink/provider.go#L107

ilion1, 2024-04-06:
> Having a similar problem with charging, having a KIA and Elli Wallbox, too.
> 
> I think the problem is the connection to the Elli wallbox, as I can find the following entries in the log (repeating every 30 seconds):
> 
> ```
> [lp-1  ] ERROR 2024/04/05 23:17:28 charge meter: data not available
> [lp-1  ] ERROR 2024/04/05 23:17:28 charge meter: not available
> [lp-1  ] ERROR 2024/04/05 23:17:58 charge meter: data not available
> [lp-1  ] ERROR 2024/04/05 23:17:58 charge meter: not available
> ```
> 
> @ilion1: Can you find the same entries in your log, too?

I have similar entries but only once when connecting the car. This is somewhat normal and can be fixed by unplugging the car if it happens more than once. If it stays I would stop evcc, restart the wallbox, wait 5 minutes and start evcc again all without a car being connected. Once charging starts I am getting the vehicle status error. From my point of view this is not related to the Elli wallbox as it happens also with the car disconnected when using the latest nightly build.

```
[lp-1  ] ERROR 2024/04/05 20:36:06 charge meter: data not available
[lp-1  ] ERROR 2024/04/05 20:36:06 charge meter: not available
[main  ] ERROR 2024/04/05 20:36:10 vehicle status: not available
[main  ] ERROR 2024/04/05 20:36:15 vehicle status: not available
[main  ] ERROR 2024/04/05 20:36:26 vehicle status: not available
[main  ] ERROR 2024/04/05 20:36:35 vehicle status: not available
[main  ] ERROR 2024/04/05 20:36:45 vehicle status: not available
[main  ] ERROR 2024/04/05 20:36:56 vehicle status: not available
[main  ] ERROR 2024/04/05 20:37:06 vehicle status: not available
[main  ] ERROR 2024/04/05 20:37:16 vehicle status: not available

```

VolkerK62, 2024-04-06:
this has nothing to do with Elli. It is only a bluelink problem.

andig, 2024-04-06:
Bitte mal

    evcc vehicle —log trace

ilion1, 2024-04-06:
ist oben schon angefügt in der ersten Nachricht für beide Versionen

andig, 2024-04-06:
Ich hab jetzt in beide rein geschaut und sehe kein Problem. Mir ist nicht klar was hier falsch sein soll.

andig, 2024-04-06:
Ahhh, sorry. Fixed in https://github.com/evcc-io/evcc/commit/c32d49094dd566c53cabbf8047ce26db002d5d97.

closed 2024-04-06 by andig