evcc-io/evcc #13315: Weird behaviour with multiple loadpoints in PV-mode

Henrik0x7F, 2024-04-06:
### Describe the bug

Hallo,

ich habe zwei Ladepunkte:

1. Go-echarger 11 kW, Priorität 10
2. Einen dreiphasigen Heizstab mit eingebauter Regelung. Kommuniziert über Modbus TCP, in evcc als einphasiger Ladepunkt modelliert, Priorität 0

Der Go-echarger funktioniert einwandfrei. Beim Heizstab gibt es Probleme:

Im PV-Modus wird dem Heizstab eine zu hohe Leistung zur Verfügung gestellt. Zum Beispiel ist ein Überschuss von 3 kW vorhanden. Dieser Wert wird dem Heizstab auch zunächst zu Verfügung gestellt. Nachdem der Heizstab allerdings einschaltet wird plötzlich ein höherer Wert gesendet (z.B. 4 kW).

Ausschnitt aus Log:
```
[lp-2  ] DEBUG 2024/04/06 14:37:05 max charge current: 27A
[site  ] DEBUG 2024/04/06 14:37:15 ----
[lp-1  ] DEBUG 2024/04/06 14:37:15 charge power: 0W
[lp-2  ] DEBUG 2024/04/06 14:37:15 charge power: 3000W
[main  ] DEBUG 2024/04/06 14:37:15 lp Ladesäule at prio 10 gets additional 3000W from Heizstab at prio 0, total 3000W
[site  ] DEBUG 2024/04/06 14:37:15 pv power: 3994W
[site  ] DEBUG 2024/04/06 14:37:15 battery soc: 100%
[site  ] DEBUG 2024/04/06 14:37:15 battery power: 123W
[site  ] DEBUG 2024/04/06 14:37:15 grid meter: -542W
[site  ] DEBUG 2024/04/06 14:37:15 grid currents: [4.7 3.65 -5.2]A
[site  ] DEBUG 2024/04/06 14:37:15 giving loadpoint priority for additional: 3000W
[site  ] DEBUG 2024/04/06 14:37:15 site power: -3319W
[lp-1  ] DEBUG 2024/04/06 14:37:15 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2024/04/06 14:37:15 charge total import: 1215.211kWh
[lp-1  ] DEBUG 2024/04/06 14:37:15 charger status: A
[site  ] DEBUG 2024/04/06 14:37:25 ----
[lp-1  ] DEBUG 2024/04/06 14:37:25 charge power: 0W
[lp-2  ] DEBUG 2024/04/06 14:37:25 charge power: 3000W
[site  ] DEBUG 2024/04/06 14:37:25 pv power: 3872W
[site  ] DEBUG 2024/04/06 14:37:25 battery soc: 100%
[site  ] DEBUG 2024/04/06 14:37:25 battery power: 64W
[site  ] DEBUG 2024/04/06 14:37:25 grid meter: -372W
[site  ] DEBUG 2024/04/06 14:37:26 grid currents: [4.79 3.74 -4.97]A
[site  ] DEBUG 2024/04/06 14:37:26 site power: -208W
[lp-2  ] DEBUG 2024/04/06 14:37:26 charger status: C
[lp-2  ] DEBUG 2024/04/06 14:37:26 pv charge current: 27.9A = 27A + 0.907A (-208W @ 1p)
```

Gleichzeitig wird dem Heizstab manchmal auch deutlich weniger Leistung zur Verfügung gestellt als Überschuss vorhanden ist. All das passiert während kein Auto am Go-echarger angeschlossen ist.

### Steps to reproduce

s.o.

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

interval: 10s # control cycle interval

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: false

meters:
- type: template
[... 1021 lines cut ...]
[site  ] DEBUG 2024/04/06 14:55:15 giving loadpoint priority for additional: 2000W
[site  ] DEBUG 2024/04/06 14:55:15 site power: -4301W
[lp-1  ] DEBUG 2024/04/06 14:55:15 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2024/04/06 14:55:15 charge total import: 1215.211kWh
[lp-1  ] DEBUG 2024/04/06 14:55:15 charger status: A
[site  ] DEBUG 2024/04/06 14:55:25 ----
[lp-1  ] DEBUG 2024/04/06 14:55:25 charge power: 0W
[lp-2  ] DEBUG 2024/04/06 14:55:25 charge power: 2000W
[site  ] DEBUG 2024/04/06 14:55:25 pv power: 4834W
[site  ] DEBUG 2024/04/06 14:55:25 battery soc: 100%
[site  ] DEBUG 2024/04/06 14:55:25 battery power: -18W
[site  ] DEBUG 2024/04/06 14:55:25 grid meter: -2138W
[site  ] DEBUG 2024/04/06 14:55:25 grid currents: [-6.02 2.55 -5.68]A
[site  ] DEBUG 2024/04/06 14:55:25 site power: -2056W
[lp-2  ] DEBUG 2024/04/06 14:55:25 charger status: C
[lp-2  ] DEBUG 2024/04/06 14:55:26 pv charge current: 37.9A = 29A + 8.94A (-2056W @ 1p)
```

### What type of operating system are you running?

Linux

### Version

0.124.6

andig, 2024-04-06:
Es ist unklar, was das Problem sein soll?

Henrik0x7F, 2024-04-06:
> Es ist unklar, was das Problem sein soll?

Hallo,

erstmal danke für die schnelle Antwort.

Problem ganz kurz gefasst: 3 kW Überschuss, evcc sendet aber höhere Werte über Modbus an Heizstab. Zum Beispiel 22 A = 5 kW. Die Batterie ist so eingestellt, dass sie nicht zum Laden benutzt werden soll und gleichzeitig ist an anderen Ladepunkten kein Auto angeschlossen. Im Log ist zu sehen:

```
[lp-2  ] DEBUG 2024/04/06 14:53:05 max charge current: 29A
...
[lp-1  ] DEBUG 2024/04/06 14:55:25 charge power: 0W
[lp-2  ] DEBUG 2024/04/06 14:55:25 charge power: 2000W
[site  ] DEBUG 2024/04/06 14:55:25 pv power: 4834W
[site  ] DEBUG 2024/04/06 14:55:25 battery soc: 100%
[site  ] DEBUG 2024/04/06 14:55:25 battery power: -18W
[site  ] DEBUG 2024/04/06 14:55:25 grid meter: -2138W
[site  ] DEBUG 2024/04/06 14:55:25 grid currents: [-6.02 2.55 -5.68]A
[site  ] DEBUG 2024/04/06 14:55:25 site power: -2056W
[lp-2  ] DEBUG 2024/04/06 14:55:25 charger status: C
[lp-2  ] DEBUG 2024/04/06 14:55:26 pv charge current: 37.9A = 29A + 8.94A (-2056W @ 1p)
```
Also:
- lp-2 max charge current = 29 A = 6670 W
- lp-1, kein Auto angeschlossen = 0 W
- lp-2, Heizstab = 2000 W
- PV-Leistung = 4834 W
- Batterie: 100 %
- Grid Messung: 2138 W Einspeisung
- Heizstab Status C
- Heizstab Charge Current = 37,9 A = 29 A + 8,94 A (entspricht 8717 W, obwohl nur 4138 W Überschuss)

Warum wird also ein höherer Strom an den Heizstab gesendet als verfügbar ist?

VolkerK62, 2024-04-06:
>Heizstab Charge Current = 37,9 A = 29 A + 8,94 A (entspricht 8717 W, obwohl nur 4138 W Überschuss)

`[lp-2  ] DEBUG 2024/04/06 14:55:26 pv charge current: 37.9A = 29A + 8.94A (-2056W @ 1p)`
Evcc geht davon aus, dass mit 29A geladen wird und wenn dann trotzdem noch 2kW Überschuss vorhanden sind, wird dementsprechend erhöht.
Das Problem ist, dass dein Heizstab nicht das macht, was ihm gesagt wird.
Kann der evtl nur in 1000W Stufen geregelt werden?

Henrik0x7F, 2024-04-06:
> Kann der evtl nur in 1000W Stufen geregelt werden?

Das ist richtig. Allerdings verstehe ich nicht, warum dies zu Problemen führen sollte. Der Go-echarger kann ja auch je nach Zahl der Phasen nur in ~230 W / ~690 W schritten regeln (ganz davon abgesehen, dass das Auto ja letztendlich den tatsächlichen Ladestrom bestimmt). Die Heizstab Regelung funktioniert wie folgt: evcc überträgt den maximal zulässigen Ladestrom. Der Heizstab schaltet dann (mit eigener Hysterese etc.) in die höchste Stufe, bei welcher der vorgegebene Ladestrom nicht überschritten wird. Die tatsächlich verbrauchte Leistung wird auch korrekt zurückgemeldet:
````
[lp-2  ] DEBUG 2024/04/06 14:55:25 charge power: 2000W
````
Funktioniert die Regelung in evcc tatsächlich so, dass die gemeldete Leistung quasi ignoriert wird?

VolkerK62, 2024-04-06:
mit den 1000W Schritten war nur eine Verständnisfrage. sollte kein Problem sein.

Dann hätte aber hier auf 3000W erhöht werden müssen.
`[lp-2  ] DEBUG 2024/04/06 14:52:05 max charge current: 14A`

Henrik0x7F, 2024-04-07:
> mit den 1000W Schritten war nur eine Verständnisfrage. sollte kein Problem sein.
> 
> Dann hätte aber hier auf 3000W erhöht werden müssen. `[lp-2 ] DEBUG 2024/04/06 14:52:05 max charge current: 14A`

Das hat er auch ;) Allerdings hat die Regelung des Heizstabs eine Verzögerung ähnlich der guardduration in evcc. Heißt es kann einige Minuten dauern, bis der Heizstab schaltet.

Mir stellt sich immer noch die Grundsatzfrage: Ist das ein Bug oder ein Feature? Also das evcc anscheinend nach einer Weile den Ladestrom über den Überschuss hinaus erhöht, wenn der Ladepunkt nicht die vorgegebene Leistung verbraucht. Ich kann mir gerade kein Szenario vorstellen, wo dieses Verhalten erforderlich ist.

Nach welcher Zeit und ab welcher Differenz tritt diese Überhöhung auf (falls Feature)? Ggf. kann ich die Verzögerung im Heizstab reduzieren. Möchte aber auch nicht, dass mir die Relais nach einem Jahr alle kaputt gehen.

andig, 2024-04-07:
Geregelt wird nach Netzübergabepunkt. Springende Lasten sind daher ein Problem.