evcc-io/evcc #13334: Elli charger not responding to stop command

anduz, 2024-04-07:
### Describe the bug

Charger is showing wrong charging power and does not stop charging.

### Steps to reproduce

1. Solar: SUN2000 + LUNA2000
2. Charger: VW ID Charger Connect Elli single phase, 
     - internal CT connected but disabled with DIP A 0000
     - max current set to 15 with DIP B 01010 (setting 11110 for 20A single fase showed INVALID in WB config, but no other error)
4. Car: eUP
5. Plug in Car with charger setting "PV"
6. Click "AUS" does not stop charging but an error message is displayed:
      Garage: charger out of sync: expected disabled, got enabled

Got also this error earlier:
     Garage: charger logic error: current mismatch (got 0A, expected 2.2A)

### Configuration details

```yaml
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: NNNNNNNNNNNNNNNNNNNNNNNNNNNNNNNN

interval: 30s # control cycle interval

meters:
- type: template
  template: huawei-dongle-powersensor 
  id: 1  
  host: 192.168.0.99  
  port: 5021  
  usage: grid  
  modbus: tcpip  
  name: grid1
- type: template
  template: huawei-dongle-powersensor 
  id: 1  
  host: 192.168.0.99  
  port: 5021
  usage: pv  
  modbus: tcpip  
  name: pv2
- type: template
  template: huawei-dongle-powersensor 
  id: 1  
  host: 192.168.0.99  
  port: 5021
  usage: battery  
  modbus: tcpip  
  name: battery3

chargers:
- type: template
  template: elliconnect
  ski: NNNNNNNNNNNNNNNNNNNNNNNNNNNNNNNN
  ip: 192.168.0.174
  name: wallbox1

loadpoints:
- title: Garage
[... 121 lines cut ...]
[lp-1  ] DEBUG 2024/04/07 21:02:43 set session energy limit: 0
[site  ] DEBUG 2024/04/07 21:02:58 ----
[lp-1  ] DEBUG 2024/04/07 21:02:58 charge power: 0W
[site  ] DEBUG 2024/04/07 21:03:01 pv power: 0W
[site  ] DEBUG 2024/04/07 21:03:05 battery soc: 67%
[site  ] DEBUG 2024/04/07 21:03:05 battery power: 778W
[site  ] DEBUG 2024/04/07 21:03:05 grid meter: 15W
[site  ] DEBUG 2024/04/07 21:03:10 grid currents: [-0.52 -0 -0]A
[site  ] DEBUG 2024/04/07 21:03:11 site power: 893W
[lp-1  ] DEBUG 2024/04/07 21:03:11 charger status: A
[site  ] DEBUG 2024/04/07 21:03:28 ----
[lp-1  ] DEBUG 2024/04/07 21:03:28 charge power: 0W
[site  ] DEBUG 2024/04/07 21:03:29 pv power: 0W
[site  ] DEBUG 2024/04/07 21:03:32 battery soc: 67%
[site  ] DEBUG 2024/04/07 21:03:32 battery power: 767W
[site  ] DEBUG 2024/04/07 21:03:34 grid meter: -21W
```

### What type of operating system are you running?

Docker container

### Version

evcc version 0.125.0

StefanSchoof, 2024-04-07:
I think Elli does not work if the CT Spule is connected: https://github.com/evcc-io/evcc/discussions/2396#discussioncomment-3038835

StefanSchoof, 2024-04-07:
Das war die.aussage von Elli: "Bitte beachten Sie, dass unsere Wallboxen an eine Photovoltaikanlage angeschlossen werden können, indem Sie die installierten TC-Spulen entfernen und die DIP-Schalter auf den Werkseinstellungen 00000 - 00000 belassen."

anduz, 2024-04-07:
Danke, werde es testen!