evcc-io/evcc #13335: Garo Wallbox- Error "[lp-1 ] ERROR charge timer: modbus: length in response header '12081' must not be zero or greater than '254'"

m03pchen, 2024-04-07:
### Describe the bug

Hello everyone,

I have a Garo wallbox ((GLBDCMW-T211FC-VCC) connected to my homeassistant via wifi.
The garo_wallbox addon is running on the homeassistant and also displays all values.
I can also access the wallbox's web interface via the local IP, change the mode (on/off) and an other things (like loadpower 6-16 ampere)

After the 3 Minute start this error pops into thelog:

"[lp-1 ] ERROR charge timer: modbus: length in response header '12081' must not be zero or greater than '254'"

The error comes every minute again.
No wallbox is displayed under the device settings - only the Volvo.
On the overview page, however, the Volvo is always shown as "not connected" even if the charging cable is connected and I start the wallbox via Home Assistant or web interface.

Config from http://wallbox-ip:8080/servlet/rest/chargebox/config

{
"ocppConnected": false,
"maxChargeCurrent": 16,
"productId": "119",
"programVersion": "8.4",
"firmwareVersion": 8,
"firmwareRevision": 4,
"lbVersion2": true,
"serialNumber": 2180591,
"meterSerialNumber": "122145V",
"meterType": 341,
"factoryChargeLimit": 32,
"switchChargeLimit": 16,
"rfidReaderPresent": false,
"rfidMode": "RFID_DISABLED",
"powerbackupStatus": 2,
"lastTemperature": 29,
"warningTemperature": 59,
"cutoffTemperature": 64,
"reducedIntervalsEnabled": false,
"reducedCurrentIntervals": [],
"softwareVersion": 189,
"availableVersion": 0,
"updateUrl": "",
"networkMode": 1,
"networkType": 0,
"networkSSID": "WIFIname",
"webNetworkPassword": "",
"networkAPChannel": 6,
"ethNetworkMode": 0,
"gcConfigTimestamp": null,
"gcloudActivated": false,
"gcActivatedFrom": null,
"ethGateway": "",
"ethDNS": "",
"ethIP": "",
"ethMask": 24,
"localLoadBalanced": false,
"groupLoadBalanced": false,
"groupLoadBalanced101": false,
"energyReportEnabled": false,
"master": true,
"timezone": null,
"slaveList": [
{
"reference": "Name-Wallbox",
"serialNumber": 2180591,
"lastContact": 1644515054363,
"online": false,
"loadBalanced": false,
"phase": 4,
"productId": 132,
[... 367 lines cut ...]
[lp-1  ] INFO 2024/04/07 21:23:39 loadpoint 1:
[lp-1  ] INFO 2024/04/07 21:23:39   mode:        off
[lp-1  ] INFO 2024/04/07 21:23:39   charger:     power ✗ energy ✗ currents ✗ phases ✗ wakeup ✗
[lp-1  ] INFO 2024/04/07 21:23:39   meters:      charge ✗
[lp-1  ] DEBUG 2024/04/07 21:23:39 phase timer inactive
[lp-1  ] DEBUG 2024/04/07 21:23:39 pv timer inactive
[lp-1  ] INFO 2024/04/07 21:23:39 vehicle updated: unknown -> VolvoV60
[lp-1  ] ERROR 2024/04/07 21:24:39 charger enabled: modbus: length in response header '12081' must not be zero or greater than '254'
[site  ] WARN 2024/04/07 21:24:39 interval <30s can lead to unexpected behavior, see https://docs.evcc.io/docs/reference/configuration/interval
[site  ] DEBUG 2024/04/07 21:24:39 ----
[lp-1  ] DEBUG 2024/04/07 21:24:39 charge power: 0W
[site  ] DEBUG 2024/04/07 21:24:39 pv power: 0W
[site  ] DEBUG 2024/04/07 21:24:39 battery soc: 76%
[site  ] DEBUG 2024/04/07 21:24:39 battery power: 512W
[site  ] DEBUG 2024/04/07 21:24:39 grid meter: 1W
[site  ] DEBUG 2024/04/07 21:24:39 site power: 613W
```

### What type of operating system are you running?

Linux

### Version

0.125.0

andig, 2024-04-08:
Try having only one system talking to the box.

m03pchen, 2024-04-08:
> Try having only one system talking to the box.

I tried it today. I deactivated the Garo add-on and completely restarted Homeassistant (Power shutdown)
Then I started HA with EVCC only - log looks exactly the same.
3 minutes until EVCC is completely started and one minute later the modbus: length error appears

andig, 2024-04-08:
@premultiply es geht um Bender- da musst Du mal bitte schauen.

premultiply, 2024-04-08:
Was sagt denn `evcc charger --log trace --diagnose`?

m03pchen, 2024-04-09:
> Was sagt denn `evcc charger --log trace --diagnose`?

Sorry aber wo genau gebe ich den Befehl ein ?
Bei mir läuft HA auf Proxmox - glaube da komme ich nicht in die Console dafür oder?

premultiply, 2024-04-09:
Einfach auf einem normalen System Windows, macOS, Linux etc. in der Konsole ausführen.
evcc ist eine einzelne ausführbare Datei.

m03pchen, 2024-04-09:
Ehrlich gesagt komme ich so nicht weiter.
Habe gerade mal versucht EVCC über Windows zu konfigurieren, PV Anlage und Fahrzeug klappt.
Bei der Wallbox kommt folgendes (gleiches):

"Teste die Konfiguration von Garo GLB, GLB+, LS4, LS4 compact ...
  Fehler: modbus: length in response header '12081' must not be zero or greater than '254'

 Der Test von Garo GLB, GLB+, LS4, LS4 compact ist fehlgeschlagen. Soll es trotzdem in die Konfiguration aufgenommen 
 werden? Yes

 Garo GLB, GLB+, LS4, LS4 compact Garo GLB, GLB+, LS4, LS4 compact wurde erfolgreich hinzugefügt."

Ich habe dann aus Unwissenheit deinen Befehl eingegeben:

C:\Download\evcc>evcc charger --log trace --diagnose
[main  ] INFO 2024/04/09 20:28:51 evcc 0.125.0
[main  ] INFO 2024/04/09 20:28:51 using config file: C:\Download\evcc\evcc.yaml
[db    ] INFO 2024/04/09 20:28:51 using sqlite database: C:\Users\*****\.evcc\evcc.db
[db    ] TRACE 2024/04/09 20:28:51 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="settings" -1 <nil>
[db    ] TRACE 2024/04/09 20:28:51 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "settings" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2024/04/09 20:28:51 SELECT * FROM `settings` LIMIT 1 -1 <nil>
[db    ] TRACE 2024/04/09 20:28:51 SELECT * FROM `settings` 0 <nil>
[db    ] TRACE 2024/04/09 20:28:51 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="device_details" -1 <nil>
[db    ] TRACE 2024/04/09 20:28:51 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="devices" -1 <nil>
[db    ] TRACE 2024/04/09 20:28:51 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="configs" -1 <nil>
[db    ] TRACE 2024/04/09 20:28:51 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "configs" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2024/04/09 20:28:51 SELECT * FROM `configs` LIMIT 1 -1 <nil>
[db    ] TRACE 2024/04/09 20:28:51 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="config_details" -1 <nil>
[db    ] TRACE 2024/04/09 20:28:51 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "config_details" AND sql IS NOT NULL order by type = "table" desc 2 <nil>
[db    ] TRACE 2024/04/09 20:28:51 SELECT * FROM `config_details` LIMIT 1 -1 <nil>
[db    ] TRACE 2024/04/09 20:28:51 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%CONSTRAINT ""fk_configs_details"" %" OR sql LIKE "%CONSTRAINT fk_configs_details %" OR sql LIKE "%CONSTRAINT `fk_configs_details`%" OR sql LIKE "%CONSTRAINT [fk_configs_details]%" OR sql LIKE "%CONSTRAINT         fk_configs_details      %") -1 <nil>
[db    ] TRACE 2024/04/09 20:28:51 SELECT count(*) FROM sqlite_master WHERE type = "index" AND tbl_name = "config_details" AND name = "idx_unique" -1 <nil>
[db    ] TRACE 2024/04/09 20:28:51 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%CONSTRAINT ""fk_devices_details"" %" OR sql LIKE "%CONSTRAINT fk_devices_details %" OR sql LIKE "%CONSTRAINT `fk_devices_details`%" OR sql LIKE "%CONSTRAINT [fk_devices_details]%" OR sql LIKE "%CONSTRAINT         fk_devices_details      %") -1 <nil>
[db    ] TRACE 2024/04/09 20:28:51 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%""device_id"" %" OR sql LIKE "%device_id %" OR sql LIKE "%`device_id`%" OR sql LIKE "%[device_id]%" OR sql LIKE "%   device_id       %") -1 <nil>
[db    ] TRACE 2024/04/09 20:28:51 SELECT * FROM `configs` WHERE `configs`.`class` = 1 ORDER BY id 0 <nil>

Ansonsten bin ich mit meinem Können ohne Anleitung leider am Ende

premultiply, 2024-04-09:
```yaml
chargers:
  - name: Garo-Wallbox
    type: template
    template: bender
    host: *.*.*.* # IP-Adresse oder Hostname
    port: 8080
```

Port 8080 klingt falsch.
Was passt am Default 502 nicht?

m03pchen, 2024-04-09:
Habe mal den Port 8080 gegen 502 ausgetauscht:

[main  ] INFO 2024/04/09 21:11:51 evcc 0.125.0
[main  ] INFO 2024/04/09 21:11:51 using config file: /config/evcc.yaml
[main  ] INFO 2024/04/09 21:11:51 starting ui and api at :7070
[lp-1  ] WARN 2024/04/09 21:11:52 poll mode '{connected 5m0s}' may deplete your battery or lead to API misuse. USE AT YOUR OWN RISK.
[site  ] INFO 2024/04/09 21:11:52 site config:
[site  ] INFO 2024/04/09 21:11:52   meters:      grid ✓ pv ✓ battery ✓
[site  ] INFO 2024/04/09 21:11:52     grid:      power ✓ energy ✓ currents ✗
[site  ] INFO 2024/04/09 21:11:52     pv 1:      power ✓ energy ✓ currents ✗
[site  ] INFO 2024/04/09 21:11:52     battery 1: power ✓ energy ✓ currents ✗ soc ✓ capacity ✗
[site  ] INFO 2024/04/09 21:11:52   vehicles:
[site  ] INFO 2024/04/09 21:11:52     vehicle 1: range ✓ finish ✓ status ✓ climate ✗ wakeup ✗
[lp-1  ] INFO 2024/04/09 21:11:52 loadpoint 1:
[lp-1  ] INFO 2024/04/09 21:11:52   mode:        off
[lp-1  ] INFO 2024/04/09 21:11:52   charger:     power ✗ energy ✗ currents ✗ phases ✗ wakeup ✗
[lp-1  ] INFO 2024/04/09 21:11:52   meters:      charge ✗
[lp-1  ] DEBUG 2024/04/09 21:11:52 phase timer inactive
[lp-1  ] DEBUG 2024/04/09 21:11:52 pv timer inactive
[lp-1  ] INFO 2024/04/09 21:11:52 vehicle updated: unknown -> VolvoV60
[lp-1  ] ERROR 2024/04/09 21:11:52 charger enabled: dial tcp 192.168.178.10:502: connect: connection refused

premultiply, 2024-04-09:
https://docs.evcc.io/docs/devices/chargers#garo-glb-glb-ls4-ls4-compact

> Der 'Modbus TCP Server für Energiemanagement-Systeme' muss aktiviert sein. 'Registersatz' darf NICHT auf 'Phoenix' oder 'TQ-DM100' eingestellt sein. Die dritte Auswahlmöglichkeit 'Ebee', 'Bender', 'MENNEKES' etc. ist richtig. 'UID Übertragung erlauben' muss aktiviert sein.

Ist das alles gegeben?