evcc-io/evcc #13339: Solaredge Home EV Charger nur max. Leistung, Einstellungen evcc werden ignoriert

Shoe-Me, 2024-04-08:
### Describe the bug

Der Home EV Charger ist derzeit nicht im Gerätemanager der Solaredge App eingebunden und somit nicht aktiv über die App steuerbar, jedoch in der App als Charger sichtbar. 
Einstellungen von evcc wie max. Ladestrom werden vom Charger ignoriert (unabhängig davon, ob ich die Solaredge oder Keba App zum steuern installiert habe oder darüber Einstellungen vornehme). Stelle ich den max. Ladestrom in der Keba App ein, wird dieser berücksichtigt. 
Schließe ich den Charger ans Fahrzeug startet der Ladevorgang zunächst mit den von evcc vorgegebenen Parametern (bspw. Min+PV). Nach einer gewissen Ladedauer (variiert von Ladevorgang zu Ladevorgang) schaltet die Wallbox auf max. Ladeleistung und bleibt dabei, bis der Vorgang unterbrochen wird. Fehlermeldungen in evcc werden nicht angezeigt. 

### Steps to reproduce

1. Ich habe den Fehler mehrfach reproduziert, indem ich den Ladevorgang neu gestartet habe. 
2. Ich habe die Wallbox mit und ohne Einbindung in Solaredge / Keba App versucht über evcc zu steuern. Gleiches Ergebnis! 
3. Konfigurieren einer neuen yaml hat nicht geholfen. Auch das Zurücksetzen der Wallbox auf Werkseinstellungen half nicht. 
...

### Configuration details

```yaml
[main  ] INFO 2024/04/08 08:40:32 evcc 0.125.0
[main  ] INFO 2024/04/08 08:40:32 using config file: /etc/evcc.yaml
[db    ] INFO 2024/04/08 08:40:33 using sqlite database: /home/michaelschumann/.evcc/evcc.db
[lp-1  ] DEBUG 2024/04/08 08:40:37 charge total import: 960.767kWh
[site  ] WARN 2024/04/08 08:40:37 battery configured but residualPower is missing or <= 0 (add residualPower: 100 to site), see https://docs.evcc.io/en/docs/reference/configuration/site#residualpower

Konfiguration (/etc/evcc.yaml):

# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: *****

interval: 30s # control cycle interval

sponsortoken: *****

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: true

meters:
- type: template
  template: solaredge-hybrid 
  id: 1  
  host: 192.168.178.63  
  port: 1502  
  usage: grid  
  modbus: tcpip  
  name: grid1
- type: template
  template: solaredge-hybrid 
  id: 1  
  host: 192.168.178.63  
  port: 1502  
  usage: pv  
  modbus: tcpip  
  name: pv2
- type: template
  template: solaredge-hybrid 
  id: 1  
  host: 192.168.178.63  
  port: 1502  
  usage: battery  
  modbus: tcpip  
  name: battery3

chargers:
- type: template
  template: keba-modbus 
  id: 255  
  host: 192.168.178.40  
  port: 502  
  modbus: tcpip  
  name: wallbox5

vehicles:
- type: template
  template: hyundai 
  title: Ioniq 5  
  user: *****
  password: *****
  vin: *****
  capacity: 58  
  language: de  
  name: ev4

loadpoints:
- title: Garage
  charger: wallbox5
  vehicle: ev4
  mode: off

site:
  title: Theodor
  meters:
    grid: grid1
    pv:
    - pv2
    battery:
    - battery3
    
tariffs:
  currency: EUR # (default EUR)
  grid:
    # static grid price
    type: fixed
    price: 0.284 # [currency]/kWh

  feedin:
    # rate for feeding excess (pv) energy to the grid
    type: fixed
    price: 0.082 # [currency]/kWh

Version: `0.125.0`
```

### Log details

```text
[main  ] INFO 2024/04/08 08:44:48 evcc 0.125.0
[main  ] INFO 2024/04/08 08:44:48 using config file: /etc/evcc.yaml
[main  ] INFO 2024/04/08 08:44:48 starting ui and api at :7070
[db    ] INFO 2024/04/08 08:44:49 using sqlite database: /home/michaelschumann/.evcc/evcc.db
[lp-1  ] DEBUG 2024/04/08 08:44:53 charge total import: 960.767kWh
[site  ] WARN 2024/04/08 08:44:53 battery configured but residualPower is missing or <= 0 (add residualPower: 100 to site), see https://docs.evcc.io/en/docs/reference/configuration/site#residualpower
[site  ] INFO 2024/04/08 08:44:53 site config:
[site  ] INFO 2024/04/08 08:44:53   meters:      grid ✓ pv ✓ battery ✓
[site  ] INFO 2024/04/08 08:44:53     grid:      power ✓ energy ✗ currents ✗
[site  ] INFO 2024/04/08 08:44:53     pv 1:      power ✓ energy ✗ currents ✗
[site  ] INFO 2024/04/08 08:44:53     battery 1: power ✓ energy ✗ currents ✗ soc ✓ capacity ✗
[site  ] INFO 2024/04/08 08:44:53   vehicles:
[site  ] INFO 2024/04/08 08:44:53     vehicle 1: range ✓ finish ✓ status ✓ climate ✗ wakeup ✓
[lp-1  ] INFO 2024/04/08 08:44:53 loadpoint 1:
[lp-1  ] INFO 2024/04/08 08:44:53   mode:        off
[lp-1  ] INFO 2024/04/08 08:44:53   charger:     power ✓ energy ✓ currents ✓ phases ✗ wakeup ✗
[lp-1  ] INFO 2024/04/08 08:44:53   meters:      charge ✓
[lp-1  ] INFO 2024/04/08 08:44:53     charge:    power ✓ energy ✓ currents ✓
[lp-1  ] DEBUG 2024/04/08 08:44:53 phase timer inactive
[lp-1  ] DEBUG 2024/04/08 08:44:53 pv timer inactive
[lp-1  ] INFO 2024/04/08 08:44:53 vehicle updated: unknown -> Ioniq 5
[site  ] DEBUG 2024/04/08 08:44:53 ----
[lp-1  ] DEBUG 2024/04/08 08:44:53 charge power: 0W
[site  ] DEBUG 2024/04/08 08:44:53 pv power: 587W
[site  ] DEBUG 2024/04/08 08:44:53 battery soc: 52%
[site  ] DEBUG 2024/04/08 08:44:53 battery power: -369W
[site  ] DEBUG 2024/04/08 08:44:53 grid meter: -4W
[site  ] DEBUG 2024/04/08 08:44:53 site power: -373W
[lp-1  ] DEBUG 2024/04/08 08:44:54 vehicle odometer: 5515km
[lp-1  ] DEBUG 2024/04/08 08:44:54 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2024/04/08 08:44:54 charge total import: 960.767kWh
[lp-1  ] DEBUG 2024/04/08 08:44:54 charger status: A
[lp-1  ] INFO 2024/04/08 08:44:54 car disconnected
[lp-1  ] DEBUG 2024/04/08 08:44:54 set session soc limit: 0
[lp-1  ] DEBUG 2024/04/08 08:44:54 set session energy limit: 0
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.125.0

andig, 2024-04-08:
Im Logfile ist davon nichts zu erkennen. Kannst du das bitte mit

    evcc charger

überprüfen.

Shoe-Me, 2024-04-08:
Momentan ist das Fahrzeug nicht verbunden. 
```
[main  ] INFO 2024/04/08 11:08:41 evcc 0.125.0
[main  ] INFO 2024/04/08 11:08:41 using config file: /etc/evcc.yaml
[db    ] INFO 2024/04/08 11:08:42 using sqlite database: /home/michaelschumann/.evcc/evcc.db
Power:          0W
Energy:         960.8kWh
Current L1..L3: 0A 0A 0A
Charge status:  A
Enabled:        false
```
Ich würde heute Nachmittag (wenn ich eine Ladesitzung starte) noch mal ein Log erstellen? Evtl. auch das aus der Wallbox, wenn es Euch hilft?

Shoe-Me, 2024-04-14:
[evcc (2).log](https://github.com/evcc-io/evcc/files/14970092/evcc.2.log)

Shoe-Me, 2024-04-14:
Ich habe das Fahrzeug an die Walbox angeschlossen und die Wallbox startet mit 11,2kw unabhängig was ich in evcc konfiguriert habe. Konfiguriert waren max 10A (=7,6kw) und Min+PV. schalte ich manuell auf Schnell, drosselt er auf 7,6kw, schalte ich dann auf Min+PV regelt die Wallbox auf 4,2kw, da momentan kaum Solarertrag vorhanden ist. Schalte ich aus, geht die Wallbox aus, schaltet kurz darauf aber wieder ein und ist auf 11,2 kw. Dann wieder aus, wieder auf 11,2kw... PV Modus geht derzeit gar nicht.

premultiply, 2024-04-14:
Bitte alle SE-Verknüpfungen vollständig entfernen.
Es kann nur eine steuernde Instanz geben.