evcc-io/evcc #13341: Daheimladen  Faktor 10 falscher charge total import: 54050.600kWh

kohbi, 2024-04-08:
### Describe the bug

Daheimladen  Faktor 10 falscher charge total import 

auf dem Display der Box steht der richtige Wert 5405,06 Kwh evcc ließt aber 54050.600kWh aus. 

### Steps to reproduce

1.restart ... 

2.
3.
...

### Configuration details

```yaml
[main  ] INFO 2024/04/08 09:46:52 evcc 0.125.0 (5d963e2d)
[main  ] INFO 2024/04/08 09:46:52 using config file: /home/pi/evcc.yaml

Konfiguration (/home/pi/evcc.yaml):

# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: error
levels:
  cache: error

# unique installation id
plant: *****

interval: 10s # control cycle interval

tariffs:
  grid:
    # either
    type: tibber
    token: *****
    homeid: "ab267ab3-e032-4107-a185-1c8b056de57e" # optional if multiple homes associated to account

#tariffs:
 # currency: EUR
 # grid:
 #   type: fixed
 #   price: 0.39
 # feedin:
 #   type: fixed
 #   price: 0.085

meters:
- type: template
  template: senec-home 
  usage: grid  
  host: 192.168.178.138  
  name: grid1
  schema: https
- type: template
  template: senec-home 
  usage: pv  
  host: 192.168.178.138  
  name: pv2
  schema: https
- type: template
  template: senec-home 
  usage: battery  
  host: 192.168.178.138  
  name: battery3
  schema: https
chargers:
- type: template
  template: daheimladen-mb 
  host: 192.168.178.54  
  port: 502  
  name: wallbox5

vehicles:
- type: template
  template: renault 
  title: ZOE  
  user: *****
  password: *****
  vin: *****
  capacity: 50  
  name: ev4

- name: my_car
  type: template
  template: mg2mqtt
  title: MG5-MQTT # Wird in der BenutzeroberflÃ¤che angezeigt (Optional)
  user: *****
  vin: *****
  capacity: 61 # AkkukapazitÃ¤t in kWh (Optional) 
loadpoints:
- title: WALLBOX
  charger: wallbox5
  mode: pv
  phases: 3
  mincurrent: 8
  maxcurrent: 16
#  resetOnDisconnect: true  
 # disable:
      #delay: 1m
      #threshold: 200
  soc:
    poll:
      mode: charging
      interval: 5m
    estimate: true

site:
  title: HOME
  residualPower: 100
  maxGridSupplyWhileBatteryCharging: 50
#  prioritySoC: 30
 # bufferSoc: 50
  meters:
    grid: grid1
    pv:
    - pv2
    battery:
    - battery3

# mqtt message broker
mqtt:
  broker: localhost:1883
  topic: evcc # root topic for publishing, set empty to disable publishing
  # clientid: foo
  user: *****
  password: *****

Version: `0.125.0 (5d963e2d)`
```

### Log details

```text
pi@iobroker-pi:~ $ sudo journalctl -fau evcc
Apr 08 09:37:14 iobroker-pi evcc[1332371]: [site  ] DEBUG 2024/04/08 09:37:14 ba                                                                                                             ttery soc: 13%
Apr 08 09:37:14 iobroker-pi evcc[1332371]: [site  ] DEBUG 2024/04/08 09:37:14 ba                                                                                                             ttery power: -1752W
Apr 08 09:37:14 iobroker-pi evcc[1332371]: [site  ] DEBUG 2024/04/08 09:37:14 gr                                                                                                             id meter: 14W
Apr 08 09:37:14 iobroker-pi evcc[1332371]: [site  ] DEBUG 2024/04/08 09:37:14 ba                                                                                                             ttery has priority at soc 13% (< 60%)
Apr 08 09:37:14 iobroker-pi evcc[1332371]: [site  ] DEBUG 2024/04/08 09:37:14 si                                                                                                             te power: 114W
Apr 08 09:37:14 iobroker-pi evcc[1332371]: [lp-1  ] DEBUG 2024/04/08 09:37:14 ch                                                                                                             arge voltages: [235 238 236]V
Apr 08 09:37:14 iobroker-pi evcc[1332371]: [lp-1  ] DEBUG 2024/04/08 09:37:14 de                                                                                                             tected connected phases: 3p
Apr 08 09:37:14 iobroker-pi evcc[1332371]: [lp-1  ] DEBUG 2024/04/08 09:37:14 ch                                                                                                             arge currents: [0 0 0]A
Apr 08 09:37:14 iobroker-pi evcc[1332371]: [lp-1  ] DEBUG 2024/04/08 09:37:14 ch                                                                                                             arge total import: 54050.600kWh
Apr 08 09:37:14 iobroker-pi evcc[1332371]: [lp-1  ] DEBUG 2024/04/08 09:37:14 ch                                                                                                             arger status: A
Apr 08 09:38:14 iobroker-pi evcc[1332371]: [site  ] DEBUG 2024/04/08 09:38:14 ----
Apr 08 09:38:14 iobroker-pi evcc[1332371]: [lp-1  ] DEBUG 2024/04/08 09:38:14 charge power: 0W
Apr 08 09:38:14 iobroker-pi evcc[1332371]: [site  ] DEBUG 2024/04/08 09:38:14 pv power: 1984W
Apr 08 09:38:14 iobroker-pi evcc[1332371]: [site  ] DEBUG 2024/04/08 09:38:14 battery soc: 13%
Apr 08 09:38:14 iobroker-pi evcc[1332371]: [site  ] DEBUG 2024/04/08 09:38:14 battery power: -1571W
Apr 08 09:38:14 iobroker-pi evcc[1332371]: [site  ] DEBUG 2024/04/08 09:38:14 grid meter: 109W
Apr 08 09:38:14 iobroker-pi evcc[1332371]: [site  ] DEBUG 2024/04/08 09:38:14 battery has priority at soc 13% (< 60%)
Apr 08 09:38:14 iobroker-pi evcc[1332371]: [site  ] DEBUG 2024/04/08 09:38:14 site power: 209W
Apr 08 09:38:14 iobroker-pi evcc[1332371]: [lp-1  ] DEBUG 2024/04/08 09:38:14 charge voltages: [234 238 236]V
Apr 08 09:38:14 iobroker-pi evcc[1332371]: [lp-1  ] DEBUG 2024/04/08 09:38:14 detected connected phases: 3p
Apr 08 09:38:14 iobroker-pi evcc[1332371]: [lp-1  ] DEBUG 2024/04/08 09:38:14 charge currents: [0 0 0]A
Apr 08 09:38:14 iobroker-pi evcc[1332371]: [lp-1  ] DEBUG 2024/04/08 09:38:14 charge total import: 54050.600kWh
Apr 08 09:38:14 iobroker-pi evcc[1332371]: [lp-1  ] DEBUG 2024/04/08 09:38:14 charger status: A
Apr 08 09:39:14 iobroker-pi evcc[1332371]: [site  ] DEBUG 2024/04/08 09:39:14 ----
Apr 08 09:39:14 iobroker-pi evcc[1332371]: [lp-1  ] DEBUG 2024/04/08 09:39:14 charge power: 0W
Apr 08 09:39:14 iobroker-pi evcc[1332371]: [site  ] DEBUG 2024/04/08 09:39:14 pv power: 2093W
Apr 08 09:39:14 iobroker-pi evcc[1332371]: [site  ] DEBUG 2024/04/08 09:39:14 battery soc: 13%
Apr 08 09:39:14 iobroker-pi evcc[1332371]: [site  ] DEBUG 2024/04/08 09:39:14 battery power: -1658W
Apr 08 09:39:14 iobroker-pi evcc[1332371]: [site  ] DEBUG 2024/04/08 09:39:14 grid meter: -3W
Apr 08 09:39:14 iobroker-pi evcc[1332371]: [site  ] DEBUG 2024/04/08 09:39:14 battery has priority at soc 13% (< 60%)
Apr 08 09:39:14 iobroker-pi evcc[1332371]: [site  ] DEBUG 2024/04/08 09:39:14 site power: 97W
Apr 08 09:39:14 iobroker-pi evcc[1332371]: [lp-1  ] DEBUG 2024/04/08 09:39:14 charge voltages: [234 238 236]V
Apr 08 09:39:14 iobroker-pi evcc[1332371]: [lp-1  ] DEBUG 2024/04/08 09:39:14 detected connected phases: 3p
Apr 08 09:39:14 iobroker-pi evcc[1332371]: [lp-1  ] DEBUG 2024/04/08 09:39:14 charge currents: [0 0 0]A
Apr 08 09:39:14 iobroker-pi evcc[1332371]: [lp-1  ] DEBUG 2024/04/08 09:39:14 charge total import: 54050.600kWh
Apr 08 09:39:14 iobroker-pi evcc[1332371]: [lp-1  ] DEBUG 2024/04/08 09:39:14 charger status: A
```

### What type of operating system are you running?

Linux

### Version

0.125.0

andig, 2024-04-08:
@VolkerK62 hast du die Box auch und kannst das bestätigen?

mdkeil, 2024-04-08:
Im Code selbst wird der ausgelesene  Wert korrekterweise um Faktor 10 reduziert.. 
https://github.com/evcc-io/evcc/blob/c1829fd494ced8bdd457c9fc2a4cd5bb7577ec10/charger/daheimladen-mb.go#L230-L237

Wurde das Register ggfs. im Rahmen eines Fw-Updates geändert?

kohbi, 2024-04-08:
Ich habe gleichzeitig ein Ticket bei Daheimladen aufgemacht. 
In der WebGUI von Daheimladen, sind die Werte richtig.  

Ich habe vor einigen Tage eine neue Version 3.19Da drauf bekommen, wegen Wifi Problemen. 
Das könnte also sein, dass die da was angepasst haben.