evcc-io/evcc #13344: ERROR vehicle soc: not available

klausj62, 2024-04-08:
### Describe the bug

Since a few weeks, the vehicle SOC does not appear in the UI, resulting in Fast Charging, since the minimum SOC is not reached. Before, during the last few months, everything was fine. 
Log Entry is: ERROR 2024/04/08 10:38:40 vehicle soc: not available (error is repeated every 10 seconds)
Vehicle is a Renault Zoe
going back to v0.124.5 does not solve the issue
no modifications to evcc.yaml (can be provided here, if helpful)

### Steps to reproduce

1. start evcc
2. plug in vehicle
3. Error "SOC is not available"
4.
...

### Configuration details

```yaml
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: ******

interval: 10s # control cycle interval

tariffs:
  currency: EUR # (default EUR)
  grid:
    type: tibber
    token: ******
  feedin:
    type: fixed
    price: 0.24

meters:
- type: template
  template: victron-energy 
  usage: grid  
  host: 192.168.2.139  
  port: 502  
  name: grid1
- type: template
  template: victron-energy 
  usage: pv  
  host: 192.168.2.139  
  port: 502  
  name: pv2
- type: template
  template: victron-energy 
  usage: battery  
  host: 192.168.2.139  
  port: 502  
  name: battery3

site:
  title: NM4
#  prioritySoc: 40
#  bufferSoc: 90
  residualPower: 100
  meters:
    grid: grid1
    pv:
    - pv2
    battery:
    - battery3

vehicles:
  - name: my_car
    type: template
    template: renault
    title: Zoe
    user: ******
    password: ******
    vin: ******
    capacity: 52
# minSoc: 20
# targetSoc: 70
    phases: 3
    cache: 20m

chargers:
- name: go-e_charger
  type: template
  template: go-e-v3
  host: 192.168.2.126

loadpoints:
- title: Stellplatz
  charger: go-e_charger
  vehicle: my_car
#  resetOnDisconnect: true
  mode: pv # off, pv, minpv, now
  phases: 0
  minCurrent: 9
  maxCurrent: 16
  enable:
    delay: 45s
  disable:
    delay: 90s
  guardduration: 0m

messaging:
  events:
    start:
      title: charging started
      msg: "charging started, ${vehicleSoc}"
    stop:
      title: charging stopped
      msg: "charging stopped, ${vehicleSoc}" 
    connect:
      title: car connected
      msg: "car connected"
    disconnect:
      title: car disconnected
      msg: "car disconnnected ${vehicleSoc}"

  services:
  - type: telegram
    token: *****
    chats: 
    - ******

#sponsortoken: 
sponsortoken: ******
```

### Log details

```text
Log Entry is: ERROR 2024/04/08 10:38:40 vehicle soc: not available
```

### What type of operating system are you running?

Linux

### Version

v0.125.0

mdkeil, 2024-04-08:
related to #13167 #13296

andig, 2024-04-08:
Please add trace log.

andig, 2024-04-08:
Logfile shows that indeed soc *is* not available:

    Apr 08 20:13:53 umweltkontrolle evcc[1197]: [renault] TRACE 2024/04/08 20:13:53 {"data":{"type":"Car","id":"***","attributes":{"timestamp":"2024-04-08T17:45:15Z","batteryTemperature":20,"batteryCapacity":0,"chargingRemainingTime":270,"chargingInstantaneousPower":22.5}}}

So the error message is expected.

klausj62, 2024-04-08:
Which means, Problem is on Renault's server?
But why is the SOC available immediately after restarting evcc?

andig, 2024-04-08:
Sieht so aus. Manchmal kommt nur eine halbe Antwort. Ein Muster kann ich aus dem kurzen Schnipsel nicht erkennen. Vielleicht haben andere open source Projekte das Problem auch und es findet sich dort eine Lösung.

klausj62, 2024-04-08:
Danke für die Analyse und die Bewertung.
Glücklicherweise kann ich über das evcc ui einen Neustart durchführen.

closed 2024-04-11 by andig

stna1981, 2024-04-12:
Ich habe das gleiche Problem mit meinem Renault Megane E-Tech...

mdkeil, 2024-04-12:
Auch hier kann ohne Log nichts gemacht werden.

klausj62, 2024-04-12:
Wieviel Log braucht Ihr denn? Bei meinem o.a. Trace sprach andig von einem Schnipsel. Ich kann das Problem jederzeit reproduzieren. Wenn es an Renault liegen sollte, verstehe ich nach wie vor nicht, dass nach einem Neustart der SOC unmittelbar zur Verfügung steht. Aktuell ist evcc für mich leider nicht mehr nutzbar, da ohne SOC bei jedem einstecken bis 100% geladen wird.

andig, 2024-04-12:
OT: wie lädst Du denn dann ohne evcc?

klausj62, 2024-04-12:
Bitte nicht falsch verstehen: Ich finde evcc an sich grossartig. Es ist für mich aber nur von Nutzen wenn es unbeaufsichtigt funktioniert. Meine Frau fährt fast ausschließlich das Auto und ich hatte ihr gerade beigebracht: Schließ immer an und der Rest geht automatisch.
Ohne evcc nutze ich zum Laden bei niedrigem Strompreis (Tibber) einfach meine alte 22 kW Wallbox und nutze den Timer direkt in der ZOE. Ärgerlich ist natürlich, wenn das PV geführte Laden nicht funktioniert. Ansonsten müsste ich evcc unter ständiger Beobachtung halten, per Telegram werde ich ja informiert, wenn das Auto eingestöpselt wird, und dann muss ich halt aus der Ferne einen Restart machen. Richtig gut machbar ist das aber nicht, wenn ich an der Arbeit oder auch mal grade mit dem Rad unterwegs bin.
Ich probiere heute Abend noch ein paar ältere Versionen von evcc. Vielleicht findet sich ja dort eine Lösung.
Irgendwann wird sicher auch eine Lösung für die aktuelle Version gefunden. Wenn ich dabei unterstützen kann, mache ich das gerne.
Vielleicht richte ich auch einen cron job ein, der evcc alle 15 Minuten neu startet. Solange einen Mindest-SOC ausgestellt habe, sollte das im PV Modus eigentlich funktionieren.

stna1981, 2024-04-12:
In meinem Fall würde das aber auch nicht helfen, da es bei mir gesagt auch mit einem Neustart nicht getan ist. Log kann ich in Kürze nachreichen, bin aktuell nicht Zuhause.

klausj62, 2024-04-12:
Habe gerade mal die Version v0.123.0 damit wird der SOC beim Anstecken zuverlässig erkannt.
Als nächstes muss ich mal das Changelog durchgehen, was mir jetzt an Funktionalität fehlt und ob ich damit leben kann.

stna1981, 2024-04-12:
Wenn ich raten müsste würde ich sagen, der Fehler kam mit 1.123.8 durch diese Änderung:
https://github.com/evcc-io/evcc/

klausj62, 2024-04-12:
Schlechte Nachrichten: War gerade noch einmal einkaufen und nach dem einstecken wurde 2x hintereinander kein SOC erkannt. Nach Neustart war es sofort ok. Ich probiere dann mal die Lösung mit dem cron job (regelmäßiger Neustart).

klausj62, 2024-04-12:
> Wenn ich raten müsste würde ich sagen, der Fehler kam mit 1.123.8 durch diese Änderung: https://github.com/evcc-io/evcc/

Der Link zeigt leider auf keine Änderung.

andig, 2024-04-12:
> Wenn ich raten müsste 

Musst Du nicht. Am besten auf alte Version zurück gehen und verifizieren!

stna1981, 2024-04-12:
Jetzt passen die Links, sorry. Manche Dinge macht man doch besser nicht am Handy. Ich werde mal testen welche der beiden Änderungen verantwortlich ist, die zweite wurde thematisch deutlich eher passen.

klausj62, 2024-04-19:
Neuigkeiten:
Auto war ein paar Tage in der Werkstatt und ist jetzt wieder da.
Auto angestöpselt, es wird kein SOC im UI gezeigt.
Jedoch bringt ein "evcc vehicle" auf der Kommandozeile folgendes zu Tage:
```
[main  ] INFO 2024/04/19 18:38:14 evcc 0.125.0
[main  ] INFO 2024/04/19 18:38:14 using config file: /etc/evcc.yaml
[db    ] INFO 2024/04/19 18:38:15 using sqlite database: /home/klausjerwan/.evcc/evcc.db
Soc:            33%
Capacity:       52.0kWh
Charge status:  B
Range:          100km
Odometer:       0km
Finish time:    2024-04-19 21:11:00 +0200 CEST
Climate active: false
Position:       unexpected status: 404 (Not Found)
Features:       []
```

D.h. der Soc ist bekannt. Aber das UI verarbeitet es einfach nicht.
Ist das jetzt ein Anhaltspunkt für eine Fehlersuche?

BTW: Ein ständiger Restart per cron-job ist natürlich auch keine Lösung, da dabei ja die Einstellungen zurückgesetzt werden.

andig, 2024-04-19:
Wie immer bei Hersteller-Apis: Trace log…

klausj62, 2024-04-19:
Bild aus dem UI:
[image]
Fehlermeldung Stellplatz: vehicle soc: not available
evcc -l trace vehicle auf der Kommandozeile bringt folgendes zu Tage:
```
[main  ] INFO 2024/04/19 20:48:34 evcc 0.125.0
[main  ] INFO 2024/04/19 20:48:34 using config file: /etc/evcc.yaml
[db    ] INFO 2024/04/19 20:48:34 using sqlite database: /home/klausjerwan/.evcc/evcc.db
[db    ] TRACE 2024/04/19 20:48:34 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="settings" -1 <nil>
[db    ] TRACE 2024/04/19 20:48:34 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "settings" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2024/04/19 20:48:34 SELECT * FROM `settings` LIMIT 1 -1 <nil>
[db    ] TRACE 2024/04/19 20:48:34 SELECT * FROM `settings` 6 <nil>
[db    ] TRACE 2024/04/19 20:48:34 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="devices" -1 <nil>
[db    ] TRACE 2024/04/19 20:48:34 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="device_details" -1 <nil>
[db    ] TRACE 2024/04/19 20:48:34 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="configs" -1 <nil>
[db    ] TRACE 2024/04/19 20:48:34 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "configs" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2024/04/19 20:48:34 SELECT * FROM `configs` LIMIT 1 -1 <nil>
[db    ] TRACE 2024/04/19 20:48:34 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="config_details" -1 <nil>
[db    ] TRACE 2024/04/19 20:48:34 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "config_details" AND sql IS NOT NULL order by type = "table" desc 2 <nil>
[db    ] TRACE 2024/04/19 20:48:34 SELECT * FROM `config_details` LIMIT 1 -1 <nil>
[db    ] TRACE 2024/04/19 20:48:34 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%CONSTRAINT ""fk_configs_details"" %" OR sql LIKE "%CONSTRAINT fk_configs_details %" OR sql LIKE "%CONSTRAINT `fk_configs_details`%" OR sql LIKE "%CONSTRAINT [fk_configs_details]%" OR sql LIKE "%CONSTRAINT 	fk_configs_details	%") -1 <nil>
[db    ] TRACE 2024/04/19 20:48:34 SELECT count(*) FROM sqlite_master WHERE type = "index" AND tbl_name = "config_details" AND name = "idx_unique" -1 <nil>
[db    ] TRACE 2024/04/19 20:48:34 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%CONSTRAINT ""fk_devices_details"" %" OR sql LIKE "%CONSTRAINT fk_devices_details %" OR sql LIKE "%CONSTRAINT `fk_devices_details`%" OR sql LIKE "%CONSTRAINT [fk_devices_details]%" OR sql LIKE "%CONSTRAINT 	fk_devices_details	%") -1 <nil>
[db    ] TRACE 2024/04/19 20:48:34 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%""device_id"" %" OR sql LIKE "%device_id %" OR sql LIKE "%`device_id`%" OR sql LIKE "%[device_id]%" OR sql LIKE "%	device_id	%") -1 <nil>
[db    ] TRACE 2024/04/19 20:48:34 SELECT * FROM `configs` WHERE `configs`.`class` = 3 ORDER BY id 0 <nil>
[renault] TRACE 2024/04/19 20:48:34 GET https://renault-wrd-prod-1-euw1-myrapp-one.s3-eu-west-1.amazonaws.com/configuration/android/config_de_DE.json
[renault] TRACE 2024/04/19 20:48:34 {
    "instabug": {
        "enabled": false
    },
    "visibility": {
        "sections": {
            "smart_route_planner": true,
            "pairing_ivi": true,
            "pairing_ze": true,
            "ots": false,
            "ots_retention": "86400"
        },
        "subsections": {},
        "views": {
            "profile_menu_alertes": true,
            "car_maintenance_tab_title": true,
            "car_offers_tab_title": true,
            "car_menu_status_pressure": false,
            "car_status_quotation": true,
            "help_menu_contact": true,
            "help_FAQ_tab_title": true,
            "help_ADVICES_tab_title": false,
            "help_notice_title": true,
            "profile_menu_infos": true,
            "profile_menu_infos_modify_email": false,
            "profile_menu_settings": true,
            "profile_menu_settings_delete_cta": true,
            "profile_menu_edition_region_layout": false,
            "carpage_tab_title_status": true,
            "navigation_myr_off": true,
            "navigation_myr": true,
            "navigation_menu_map_parking_btn": false,
            "navigation_menu_map_more_btn": false,
            "navigation_menu_poi_bottom_ctas": true,
            "navigation_menu_poi_details_sendto_cta": true,
            "help_myr": true,
            "profile_myr": true,
            "signup_firstname": true,
            "add_car_type_layout": false,
            "input_layout_add_car_reg": false,
            "profile": {
                "input_layout_civility": true,
                "input_layout_firstname": true,
                "input_layout_lastname": true,
[... 388 lines cut ...]
  "oldestDataUpdatedTimestamp": 1584722847567,
  "password": {
    "created": "2022-03-25T09:50:26.165Z"
  },
  "profile": {
    "email": "***"
  },
  "registered": "2020-03-20T16:47:27.700Z",
  "socialProviders": "site",
  "verified": "2020-03-20T16:47:53.534Z",
  "verifiedTimestamp": 1584722873534
}
[renault] TRACE 2024/04/19 20:48:37 GET https://api-wired-prod-1-euw1.wrd-aws.com/commerce/v1/accounts/033ea4f1-63e9-4006-861a-46e0e29501b2/kamereon/kca/car-adapter/v1/cars/***/location?country=DE
[... cut ...] :"{\"errors\":[{\"errorCode\":404,\"errorMessage\":\"There is no data for this vin and uid\",\"errorLevel\":\"error\",\"errorType\":\"functional\"}],\"error_reference\":\"rrt-00ace1ea7a3a5252f-a-eu-18091-9471054-49\"}"}],"error_reference":"FUNCTIONAL"}
Soc:            34%
Capacity:       52.0kWh
Charge status:  B
Range:          105km
Odometer:       0km
Finish time:    2024-04-19 23:18:00 +0200 CEST
Climate active: false
Position:       unexpected status: 404 (Not Found)
Features:       []

```

klausj62, 2024-04-19:
Im unteren Drittel des o.a. Trace steht u.a. "batteryLevel"
```
[renault] TRACE 2024/04/19 20:48:35 GET https://api-wired-prod-1-euw1.wrd-aws.com/commerce/v1/accounts/033ea4f1-63e9-4006-861a-46e0e29501b2/kamereon/kca/car-adapter/v2/cars/***/battery-status?country=DE
[renault] TRACE 2024/04/19 20:48:36 {"data":{"type":"Car","id":"***","attributes":{"timestamp":"2024-04-19T18:43:35Z","batteryLevel":34
``` ....

andig, 2024-04-19:
Warum nicht- soc ist ja auch da?

stna1981, 2024-04-19:
Und warum zeigt ihr ihn dann am UI nicht an, sondern nur nur "--"?

andig, 2024-04-20:
Du hast evcc vehicle aufgerufen. Da gibts kein UI. Siehe https://github.com/evcc-io/evcc/issues/13344#issuecomment-2043516884. Offensichtlich sind das zwei unterschiedliche Logs?!

klausj62, 2024-04-20:
Ok, ich versuche es noch einmal. Ich habe geglaubt, ich könnte auf der Kommandozeile durch Eingabe von evcc <befehl> mehr darüber erfahren, was gerade im System so vor sich geht, ggf. auch das eine oder andere triggern.

Hier noch einmal ganz konkret was ich zeigen möchte:
1. evcc läuft als Dienst
2. Ich zeige das UI um zu verdeutlichen: "Auto ist angeschlossen", aber Ladestand ist "--"
3. Meine Interpretation: evcc behauotet den Ladestand des angeschlossenen Fahrzeugs nicht zu kennen.
4. Ich rufe auf der Kommandozeile "evcc -l trace vehicle" auf um mehr darüber zu erfahren, was evcc ggf. schon weiß oder nicht weiß
5. Der Aufruf von "evcc -l trace vehicle" zeigt, so meine Interpretation, dass der soc/Baterriestand sehr wohl bekannt ist, in diesem Fall 34%.
6. Im UI ist zu sehen dass diese Information nicht genutzt wird.

andig, 2024-04-20:
Das ist schon richtig- beweist aber nur dass es nach der Anmeldung funktioniert. Das wussten wir bereits. Nach einiger Zeit kommt aber kein Soc mehr wie Dein Log oben beweist. Es bleibt unklar warum das so ist.

klausj62, 2024-04-20:
Was meinst Du mit "dass es nach der Anmeldung funktioniert"?
Meine Aussage ist die, dass evcc der SOC zwar bekannt ist, aber nicht genutzt wird.
Die Anzeige im UI ändert sich nach dem "evcc vehicle" auf der Kommandozeile nicht.

Dass hier und da die Abfrage bei Renault keine Werte liefert, kann ich mir gut vorstellen.
Wenn jedoch die Werte geliefert werden, was man, so meine Interpretation, aus dem Log nach "evcc -l trace vehicle" sehen kann, evcc diese Werte nicht verwendet, das wiederum sieht man im UI, und kann man auch an der Reaktion von evcc erkennen (Mindestladung wird gestartet), dann liegt es, auch wieder nach meiner Interpretation, an evcc.

Interpretiere ich da irgendwas falsch?

stna1981, 2024-04-20:
Für mich sieht es so aus, als erwartet evcc den Parameter soc in der API Antwort, es kommt aber nur batteryLevel und damit kann er irgendwie nic anfangen?

andig, 2024-04-20:
> Für mich sieht es so aus, als erwartet evcc den Parameter soc in der API Antwort, es kommt aber nur batteryLevel und damit kann er irgendwie nix anfangen?

Bitte nicht raten. Der Soc wir ja angezeigt.

> Interpretiere ich da irgendwas falsch?

Die Kommandozeilenabfrage hat mit der laufenden evcc Instanz einfach nix zu tun.

> Die Anzeige im UI ändert sich nach dem "evcc vehicle" auf der Kommandozeile nicht.

Ist damit erwartet.

> Dass hier und da die Abfrage bei Renault keine Werte liefert, kann ich mir gut vorstellen.

Und **GENAU** das ist das Problem.

stna1981, 2024-04-20:
@andig Nicht ganz. Siehe mein Beitrag weiter oben.

Der Fehler "vehicle soc: not available" tritt bei mir mit 0.124.10 **NICHT** (also niemals) auf, sondern erst seit 0.125.0. Also müsst ihr ja irgendwas geändert haben, was den Fehler auf Seiten evcc verursacht. Unabhängig davon, ob der Wert nun manchmal vom Server kommt oder nicht. Das liegt natürlich bei Renault.

andig, 2024-04-20:
Verstehe ich nicht. Du hast doch selbst geschrieben:

> going back to v0.124.5 does not solve the issue

Dann bitte nochmal Log alte/neue Version.

stna1981, 2024-04-20:
Das war @klausj62, wobei ich nicht weiß, ob sich sein Zitat auf die Fehlermeldung bezug oder auf die Tatsache, dass der SOC teilweise nicht angezeigt wird. Letztere kann auch an Renault liegen. Zu ersterem kann ich für mich sagen, dass der Fehler mit 0.124.10 nicht auftritt und der SOC sporadisch, wenn Renault mitspielt, angezeigt wird., Logs kann ich später anhängen, bin grade unterwegs.

CarstenPV, 2024-04-29:
Hi, anscheinend bin ich richtig. Seit ca. 1 Woche zeigt er kein SOC mehr an. ZOE-2018-41KW, im Auto nichts einstellbar.

Die SOC Anzeige im ZOE zeigt was an, natürlich, aber online zeigt er nichts an(% da waren auch mal keine KM zu sehen). Scheint bei Renault der Fehler zu sein.
Er fängt sofort an <25 % (weil er keine Anzeige hat) zuladen. Hatte aber noch 83 % (steht auf 85 %).
Also habe ich KEIN Fahrzeug genommen, 2x 5KW reingepumpt. In der Nacht hat er dann einfach nochmal angefangen zu laden 100 % und den Hausakku leer gezogen.

Meine Frage wäre, wenn man die Anfangsprozente im EVCC einstellen könnte (bei Renault Fahrzeugen), würde er dann vorschriftsmäßig laden?

Es besteht anscheinend keine Kommunikation zwischen ZOE und EVCC wegen des SOC %.
Alte Versionen haben bei mir nichts gebracht.

deadrabbit87, 2024-04-29:
Renault bombardieren.

Bei mir gings dann mal kurz wieder, jetzt aber auch nicht mehr. 

In der App von Renault wird auch nichts mehr angezeigt.

premultiply, 2024-04-29:
Related https://github.com/jamesremuscat/pyze/issues/118#issuecomment-2082663777 ?
/cc @andig

andig, 2024-04-29:
Ich denke nicht. Hier gibts um soc, nicht mileage?

premultiply, 2024-04-29:
Stimmt. :/

CarstenPV, 2024-04-29:
https://www.renault.de/kontakt.html unter Beschwerde einreichen, mal schauen, was passiert.

stna1981, 2024-04-29:
Also ich sehe in der App meistens den SOC, in evcc aber trotzdem nicht.

CarstenPV, 2024-04-30:
Was für ein Renault hast du denn? Auch ZE40?

stna1981, 2024-04-30:
Megane E-TECH Electric

CarstenPV, 2024-04-30:
Tja, ist das wirklich vergleichbar (2016 R-Link)? Wir haben kein Button, wo man vor klimatisiert etc.
Gerade habe ich festgestellt das in der App 2x MyY Z.E. In und 2x MY Z.E.Con steht.

stna1981, 2024-04-30:
Naja ich denke das API von Renault dürfte für alle Autos gleich sein. Was einzelne Fahrzeuge oder Apps unterstützen, ist ja was anderes.

CarstenPV, 2024-04-30:
Hoi, ich habe das Fahrzeugmal getrennt und wieder verbunden. Na wo ist der Aktivierungscode???
Jetzt ist der Button mit der Klimatisierung da, krass. Aber immer noch eine Prozentzahl vorhanden.

msteinbauer, 2024-05-02:
Ich habe auch seit Wochen seltsames Verhalten der Renault API, selbst die offizielle MyRenault App zeigt nur noch den Kilometerstand an. In HomeAssistant dagegen sehe ich die Reichweite, Außentemperatur und noch paar Sachen mehr, aber keinen SoC. Damit ist halt auch in EVCC kein SOC-genaues Laden möglich.

Pihupt, 2024-05-02:
> Hoi, ich habe das Fahrzeugmal getrennt und wieder verbunden. Na wo ist der Aktivierungscode??? Jetzt ist der Button mit der Klimatisierung da, krass. Aber immer noch eine Prozentzahl vorhanden.

Mir wurde auch geraten das Fzg aus der App zu löschen.
Ohne Hinweis auf den Akt.-Code (verlegt/verloren).

Nun steh ich da, ich da .......

CarstenPV, 2024-05-03:
Zum Glück hatte ich die Mails nicht gelöscht.

Bei Fragen oder wenn Sie Ihre Dienste deaktivieren möchten, wenden Sie sich bitte an unseren Kundenservice:
Mit Wohnsitz in Deutschland:
- Unter der Telefonnummer: 02232 / 737 600
Montag bis Donnerstag: Von 8:00 Uhr bis 18:00 Uhr
Freitags: von 8:00 Uhr bis 17:00 Uhr. 

Das muss doch gehen.

stna1981, 2024-05-05:
Hast du mal die 0.126.2 getestet? Damit scheint das Problem nicht mehr aufzutreten...

CarstenPV, 2024-05-05:
Hatte ich gestern, heute nochmal: .evcc vehicle schreibt: Soc: not available. Wenigstens sind die KM (ODO)jetzt wieder da.

stna1981, 2024-05-05:
Hm ok, dann muss es wohl doch was individuelles sein, entweder Zoe allgemein oder bei deinem Auto, bei mir liefert er:

Soc:            38%
Capacity:       60.0kWh
Charge status:  B
Range:          150km
Odometer:       13769km
Finish time:    2024-05-06 19:31:00 +0200 CEST
Climate active: false
Position:       49.09048805555556,8.263316388888889
OnIdentified:   Mode:pv, MinCurrent:6, MaxCurrent:16
Features:       []

Sieht also soweit alles gut aus und im UI wirds auch angezeigt.

CarstenPV, 2024-05-06:
War den Tag unterwegs und habe mal jetzt Zuhause abgefragt.

Soc:            78%
Capacity:       41.0kWh
Charge status:  A
Range:          241km
Odometer:       58603km
Finish time:    not available
Climate active: false
Position:       unexpected status: 404 (Not Found)

Jetzt warte ich auf die Zoe, aber wenn er das schon anzeigt, dann sollte es funktionieren. Heureka alle Jahre (Monate) wieder.....