evcc-io/evcc #13355: [GUI] Amps dropdown list should be limited based on underlying configuration of loadpoint

hyperbart, 2024-04-09:
### Describe the bug

EVCC with Alfen Pro Line 22KW (32A)
When you configure EVCC and set a maximum amps as below:

[image]

The GUI does not limit the list of amps you can set (so you can get wrong expectations).
GUI should limit the list of amps to the max amps configured in GUI/YAML.

[image]

### Steps to reproduce

EVCC with Alfen Pro Line 22KW (32A)
When you configure EVCC and set a maximum amps as below:

[image]

The GUI does not limit the list of amps you can set (so you can get wrong expectations).
GUI should limit the list of amps to the max amps.

[image]

### Configuration details

```yaml
loadpoints:
- title: Voortuin # display name for UI
  charger: ChargingPointAlfen # Wallbox Referenz
  mode: minpv # charge mode (off, now, minpv, pv)
  phases: 0 
  mincurrent: 6 
  maxcurrent: 22
  guardduration: 2m
  enable:
    threshold: 0
    delay: 1m
  disable:
    threshold: 0 # maximum import power (W)
    delay: 3m
```

### Log details

```text
N/A
```

### What type of operating system are you running?

Docker container

### Version

125

mdkeil, 2024-04-09:
mincurrent, maxcurrent and phases are depricates parameters in yaml-config and should be configured via UI only.

hyperbart, 2024-04-09:
> `mincurrent, maxcurrent` and `phases` are depricated parameters in yaml-config and should be configured via UI only.
> 
> `guardduration` is obsolete too.

Ok, valid feedback but not relevant for this point I think, because regardless of where you configure that limit (be that via YML or WebGUI) the list can/should still be limited according to config in my opinion.

mdkeil, 2024-04-09:
ah ok, your opinion is (after initial configuration) that the shown GUI-settings are limited to the real device parameter limits.

VolkerK62, 2024-04-09:
the list in the (WebGUI) loadpoint settings is the replacement of the parameter `mincurrent, maxcurrent` of the yaml.

naltatis, 2024-04-12:
This is a topic we already talked about a couple of times. Currently, the loadpoint only has one(!) set of min-/maxcurrent limits. Previously it was defined in the evcc.yaml, now it's in the loadpoints setting screen. What you're looking for is to have  two sets:

1. installation level: you or an installer defines min/max while installation/configuration based on the device's capability and fuse protection.
2. user level: ability to adjust min/max current on a daily basis for specify demands (manual peak shaving, manual optimization, ...). Maybe also mode specific (later). Here, it should be only possible to specify values that are inside the boundaries defined in 1.

With Config UI https://github.com/evcc-io/evcc/pull/12958 we introduce min/max current settings in the loadpoint setup/configuration interface. When we are done with that, we can introduce the second set of limits in core and limit the Main UI settings accordingly.

We already have (planned) two distinct endpoints for changing the limits, but right now, both modify the same "installation level" value.

```yaml
PUT /api/config/loadpoints/1 {maxcurrent: 16} # used in Config UI (see PR)
POST /api/loadpoints/1/maxcurrent/16 # used in Main UI
```

In short, yes, we should implement this once https://github.com/evcc-io/evcc/pull/12958 is live.

bjoernwuest, 2025-05-31:
Please also add this feature. For me, it is somewhat crucial. Because of cable length, I want to use 13A on 3~, while on 1~ I am eager to go to 15A. https://github.com/evcc-io/evcc/issues/14661