evcc-io/evcc #13360: PV-Laden trotz Netzabschaltung

chefinen, 2024-04-09:
Hallo,

ich hab da mal einen Feature-Request.
Mein Netzbetreiber schaltet meine Anlage (Netzeinspeisung) immer wieder mal ab (Gewährleistung der Netzstabilität. Das ist dann immer für des Rest des Tages.)
Die PV-Erzeugung wird auf den Hausverbrauch gedrosselt.
evcc (Modus „PV“) reagiert mit einem Stop des Ladevorgang auf die Abschaltung des Netzbetreibers.
Selbst mein SENEC-Speicher lädt dann nicht mehr (konnte ich aber noch nicht 100% nachvollziehen).

Manuelles übersteuern ist möglich:
mit „PV Min.“ und manuellem setzten des Ladestromes.
Nach kurzer Zeit (30 sek. bis 1 Min) regelt die PV-Erzeugung nach und das Auto kann wieder mit PV-Energie geladen werden.
Bei vorbeiziehenden Wolken ist man allerdings ständig am Nachregeln des Ladestroms. 

Nun zum Feature-Request:
Wäre es möglich einen neuen Modus einzuführen der diese Manuelle Übersteuerung (Überwachung der PV-Erzeugung über mehrere Minuten) dann automatisch macht. Geringer Bezug aus dem Netz wäre mir durch die verzögerte PV-Regelung in Ordnung.
Mein Senec-Speicher liefert die Information der Abschaltung auch über die Schnittstelle.
So könnte auch die Erkennung des neuen Modus auch autom. laufen.
Zusätzlich sollte es auch einen Schalter geben um die Netzabschaltung manuell setzten zu können (für Anwender die solch eine Information nicht autom. aus dem Speicher/Wechselrichter auslesen können. Ggf. auch gekoppelt mit Zeitangabe bis wann dieser wieder deaktiviert werden soll und einer Deaktivierung des neuen Modus um Mitternacht (da keine Sonne mehr scheint).

Auslesen der Information über Abschaltung der Netzeinspeisung des SENEC-Speichers:
JSON: Sektion „PV1“ Wert „POWER_RATIO“
(Angabe ist in Prozent)

premultiply, 2024-04-09:
Das ist so leider nicht direkt möglich, da in diesem Fall die Regelgröße fehlt.
Technisch wird hier temporär eine Nulleinspeisung realisiert. Die potenzielle aber abgeregelte PV-Erzeugung lässt sich leider nicht messen.

Die einzige Chance wäre von der Regelung für ausreichend Netzbezug zu sorgen, die die PV dann bis zu ihrem Limit ausgleichen kann. Diesen Arbeitspunkt kannst du bei evcc mittels `residualPower` verschieben (Netzbezug ist hier negativ).

chefinen, 2024-04-09:
Hallo premultiply,
Genau so ist es wie Sie im ersten Teil Beschrieben haben. residualPower könnt so tatsächlich funktionieren. Werde ich beim nächsten Abschalten testen.
Wobei ich residualPower bereits für das Laden meines Speichers (Wert 200, also positiv ) nutze Zudem ist diese Einstellung in der Konfiguration-Datei die auf meiner NAS liegt. 
Auch hier müsste immer manuell eingegriffen werden.

Lässt sich da nichts automatisches Programmieren?

Vielen Dank und viele Grüße

VolkerK62, 2024-04-09:
`residualpower`lässt sich per API ändern. Musst dir dann nur einen externen Automatismus bauen.