evcc-io/evcc #13363: Lademodus "Aus" vs. "evcc inaktiv"

Hooch76, 2024-04-09:
**Is your feature request related to a problem? Please describe.**

Hängt nicht mit einem Problem zusammen.
Der evcc-Modus "Aus" entspricht in meiner Konstellation (SMA Wechselrichter + SMA EV Charger) dem Zustand "Wallbox aus". evcc setzt in diesem Fall den entsprechenden Parameter in der Wallbox (Ereignis im Sunny Portal: "_Parameter "'Betriebsart des Ladevorgangs'" erfolgreich gesetzt. "Schnellladen" auf "Ladestopp"_"). 
Der Feature Request ist, dass man evcc über die Oberfläche auf aus / inaktiv schalten kann, so dass evcc aktiv keine Konfig-Änderungen an den Komponenten vornimmt. 

**Describe the solution you'd like**
Über die UI kann man evcc inaktiv schalten, so dass keine Änderungen an den integrierten Geräten vorgenommen werden.
Hintergrund: wenn ich den SMA EV Charger selbst zum Überschussladen verwenden möchte (= dessen eingebaute Funktion) muss ich den Drehschalter am EV Charger von "Schnellladen" auf "PV Überschussladen" umschalten. Das erkennt zwar evcc, setzt dann aber trotzdem auf Ladestopp. Somit ist der EV Charger aus und kann nicht für das SMA-eigene Überschussladen verwendet werden. 
Aktueller Workaround: den evcc-Dienst stoppen. 
Praktischer wäre es jedoch, wen man das über die evcc-UI oder API machen könnte oder evcc selbst diesen Zustand erkennt und übernimmt.
Jetzt könnte man natürlich argumentieren: "Nimm halt evcc zum Überschussladen". Eigentlich korrekt, aber der SMA EV Charger kann nicht per API zwischen 1p und 3p umgeschaltet werden ( https://github.com/evcc-io/evcc/discussions/3651 ) so dass Überschussladen mit evcc nur mit 3p und entsprechend viel Überschuss geht.

**Describe alternatives you've considered**
Alternative: evcc-Dienst stoppen.

**Additional context**

Zusätzlicher Modus "Inaktiv".
[image]

twingo667, 2024-04-09:
Würde auch die Performance bei mehreren Wallboxen erhöhen

andig, 2024-04-09:
Die Diskussion gabs schon mehrfach. Aktuell nicht vorgesehen/ keine Prio.

andig, 2024-04-09:
@Hooch76 warum willst du den EV Charger verwenden wenn du evcc hast? Was kann er besser das wir können sollten?

closed 2024-04-09 by andig

Hooch76, 2024-04-09:
> @Hooch76 warum willst du den EV Charger verwenden wenn du evcc hast? Was kann er besser das wir können sollten?

Mit evcc kann der EV Charger nur 3-phasig Überschussladen (d.h. erst ab ca 4,1kW), da sich der Drehschalter im Modus "Schnellladen" befinden muss. 

Der EV Charger (ohne evcc) kann aber auch von sich aus auf 1-phasig umschalten (im Öko-Modus), so dass ich schon ab ca 1,4kW überschussladen kann.  Sprich ich kann deutlich früher Überschuss nutzen. 
Im Öko-Modus funzt evcc nicht. Und im Schnelllademodus kann evcc nicht von 3p auf 1p umschalten, da das die API des EV Chargers nicht bietet.

Oder habe ich eine Option übersehen?

andig, 2024-04-10:
> Der EV Charger (ohne evcc) kann aber auch von sich aus auf 1-phasig umschalten (im Öko-Modus), so dass ich schon ab ca 1,4kW überschussladen kann. Sprich ich kann deutlich früher Überschuss nutzen.

Verstanden, das ist blöd. Aber: der richtige Fix hier wäre evcc die 1p3p Umschaltung zu erlauben zu der der Charger ja fähig ist. Vielleicht schaffst Du es ja als Kunde, SMA dazu zu bewegen Dir die entsprechende Doku zur Verfügung zu stellen- das bauen wir dan im Handumdrehen ein!

Hooch76, 2024-04-12:
Ich habe heute mal noch etwas rumprobiert, dabei war die Stellung des Schalters am EV Charger _immer_ "Öko" (also nicht Schnellladen).

evcc im Modus "PV". Auto angesteckt. Genug Überschuss für 1p Laden, aber nicht für 3p laden:

```
Apr 12 14:16:50 evccsrv evcc[1078423]: [lp-1  ] INFO 2024/04/12 14:16:50 car connected
Apr 12 14:16:50 evccsrv evcc[1078423]: [lp-1  ] DEBUG 2024/04/12 14:16:50 pv timer elapse
Apr 12 14:16:50 evccsrv evcc[1078423]: [lp-1  ] DEBUG 2024/04/12 14:16:50 pv timer inactive
Apr 12 14:16:50 evccsrv evcc[1078423]: [lp-1  ] DEBUG 2024/04/12 14:16:50 vehicle soc: 36%
Apr 12 14:16:50 evccsrv evcc[1078423]: [lp-1  ] DEBUG 2024/04/12 14:16:50 vehicle range: 183km
Apr 12 14:16:50 evccsrv evcc[1078423]: [lp-1  ] DEBUG 2024/04/12 14:16:50 pv charge current: 4.31A = 0A + 4.31A (-2976W @ 3p)
Apr 12 14:16:50 evccsrv evcc[1078423]: [lp-1  ] DEBUG 2024/04/12 14:16:50 pv enable timer reset
Apr 12 14:16:50 evccsrv evcc[1078423]: [lp-1  ] DEBUG 2024/04/12 14:16:50 pv timer inactive
Apr 12 14:17:19 evccsrv evcc[1078423]: [site  ] DEBUG 2024/04/12 14:17:19 ----
```

Es passiert nichts, d.h. Laden startet nicht.

Dann die Preisgrenze für's Schnellladen im PV-Modus so gesetzt, dass das Laden startet:

```
Apr 12 14:26:19 evccsrv evcc[1078423]: [site  ] DEBUG 2024/04/12 14:26:19 ----
Apr 12 14:26:20 evccsrv evcc[1078423]: [lp-1  ] DEBUG 2024/04/12 14:26:20 charge power: 0W
Apr 12 14:26:20 evccsrv evcc[1078423]: [site  ] DEBUG 2024/04/12 14:26:20 pv power: 3224W
Apr 12 14:26:20 evccsrv evcc[1078423]: [site  ] DEBUG 2024/04/12 14:26:20 battery soc: 100%
Apr 12 14:26:20 evccsrv evcc[1078423]: [site  ] DEBUG 2024/04/12 14:26:20 battery power: 0W
Apr 12 14:26:20 evccsrv evcc[1078423]: [site  ] DEBUG 2024/04/12 14:26:20 grid meter: -2814W
Apr 12 14:26:20 evccsrv evcc[1078423]: [site  ] DEBUG 2024/04/12 14:26:20 grid powers: [-1012 -962 -839]W
Apr 12 14:26:20 evccsrv evcc[1078423]: [site  ] DEBUG 2024/04/12 14:26:20 grid currents: [-4.4 -4.16 -3.86]A
Apr 12 14:26:20 evccsrv evcc[1078423]: [site  ] DEBUG 2024/04/12 14:26:20 site power: -2514W
Apr 12 14:26:20 evccsrv evcc[1078423]: [lp-1  ] DEBUG 2024/04/12 14:26:20 charge currents: [-0 -0 -0]A
Apr 12 14:26:20 evccsrv evcc[1078423]: [lp-1  ] DEBUG 2024/04/12 14:26:20 charger status: B
Apr 12 14:26:20 evccsrv evcc[1078423]: [lp-1  ] DEBUG 2024/04/12 14:26:20 next soc poll remaining time: 29s
Apr 12 14:26:20 evccsrv evcc[1078423]: [lp-1  ] DEBUG 2024/04/12 14:26:20 pv charge current: 3.64A = 0A + 3.64A (-2514W @ 3p)
Apr 12 14:26:34 evccsrv evcc[1078423]: [lp-1  ] DEBUG 2024/04/12 14:26:34 set smart cost limit: 0.14
Apr 12 14:26:49 evccsrv evcc[1078423]: [site  ] DEBUG 2024/04/12 14:26:49 ----
Apr 12 14:26:49 evccsrv evcc[1078423]: [lp-1  ] DEBUG 2024/04/12 14:26:49 charge power: 0W
Apr 12 14:26:49 evccsrv evcc[1078423]: [site  ] DEBUG 2024/04/12 14:26:49 pv power: 3391W
Apr 12 14:26:50 evccsrv evcc[1078423]: [site  ] DEBUG 2024/04/12 14:26:50 battery soc: 100%
Apr 12 14:26:50 evccsrv evcc[1078423]: [site  ] DEBUG 2024/04/12 14:26:50 battery power: 0W
Apr 12 14:26:50 evccsrv evcc[1078423]: [site  ] DEBUG 2024/04/12 14:26:50 grid meter: -3024W
Apr 12 14:26:50 evccsrv evcc[1078423]: [site  ] DEBUG 2024/04/12 14:26:50 grid powers: [-1069 -1019 -936]W
Apr 12 14:26:50 evccsrv evcc[1078423]: [site  ] DEBUG 2024/04/12 14:26:50 grid currents: [-4.64 -4.39 -4.3]A
Apr 12 14:26:50 evccsrv evcc[1078423]: [site  ] DEBUG 2024/04/12 14:26:50 site power: -2724W
Apr 12 14:26:50 evccsrv evcc[1078423]: [lp-1  ] DEBUG 2024/04/12 14:26:50 charge currents: [-0 -0 -0]A
Apr 12 14:26:50 evccsrv evcc[1078423]: [lp-1  ] DEBUG 2024/04/12 14:26:50 charger status: B
Apr 12 14:26:50 evccsrv evcc[1078423]: [lp-1  ] DEBUG 2024/04/12 14:26:50 next soc poll remaining time: 0s
Apr 12 14:26:50 evccsrv evcc[1078423]: [lp-1  ] DEBUG 2024/04/12 14:26:50 max charge current: 14A
Apr 12 14:26:51 evccsrv evcc[1078423]: [lp-1  ] DEBUG 2024/04/12 14:26:51 pv timer elapse
Apr 12 14:26:51 evccsrv evcc[1078423]: [lp-1  ] DEBUG 2024/04/12 14:26:51 pv timer inactive
Apr 12 14:26:51 evccsrv evcc[1078423]: [lp-1  ] ERROR 2024/04/12 14:26:51 charger enable: switch position not on fast charging - SMA's own optimized charging was activated
Apr 12 14:27:19 evccsrv evcc[1078423]: [site  ] DEBUG 2024/04/12 14:27:19 ----
Apr 12 14:27:19 evccsrv evcc[1078423]: [lp-1  ] DEBUG 2024/04/12 14:27:19 charge power: 16W
Apr 12 14:27:19 evccsrv evcc[1078423]: [site  ] DEBUG 2024/04/12 14:27:19 pv power: 3529W
Apr 12 14:27:20 evccsrv evcc[1078423]: [site  ] DEBUG 2024/04/12 14:27:20 battery soc: 100%
Apr 12 14:27:20 evccsrv evcc[1078423]: [site  ] DEBUG 2024/04/12 14:27:20 battery power: 0W
Apr 12 14:27:20 evccsrv evcc[1078423]: [site  ] DEBUG 2024/04/12 14:27:20 grid meter: -3118W
Apr 12 14:27:20 evccsrv evcc[1078423]: [site  ] DEBUG 2024/04/12 14:27:20 grid powers: [-1065 -1064 -988]W
Apr 12 14:27:20 evccsrv evcc[1078423]: [site  ] DEBUG 2024/04/12 14:27:20 grid currents: [-4.73 -4.59 -4.51]A
Apr 12 14:27:20 evccsrv evcc[1078423]: [site  ] DEBUG 2024/04/12 14:27:20 site power: -2818W
Apr 12 14:27:20 evccsrv evcc[1078423]: [lp-1  ] DEBUG 2024/04/12 14:27:20 charge currents: [6.21 -0 -0]A
Apr 12 14:27:20 evccsrv evcc[1078423]: [lp-1  ] DEBUG 2024/04/12 14:27:20 charger status: C
Apr 12 14:27:20 evccsrv evcc[1078423]: [lp-1  ] INFO 2024/04/12 14:27:20 start charging ->
Apr 12 14:27:20 evccsrv evcc[1078423]: [telegram] DEBUG 2024/04/12 14:27:20 sending to ***
[... cut ...]
In der GUI sieht man das auch (es ist zwar 3p konfiguriert, aber evcc erkennt, dass die Wallbox (von sich aus) 1p lädt und zeigt entsprechend die max. Ladeleistung für 1p an):

[image]

@andig Ich weiss nur nicht, was man mit den Fakten jetzt anstellen kann. Zumal ich noch beobachten muss, was passiert wenn mehr Überschuss vorhanden ist, dass die Wallbox auf 3p umschaltet. Wenn also evcc die Ladung automatisch starten würde, wenn evcc im `PV` Modus ist und `SMA's own optimized charging was activated` erkennt wird, dann wäre das dem Ziel schon ein Stück näher.

andig, 2024-04-12:
Da lässt sich nix mit anfangen da weiter unklar ist wie wir die WB bzgl. 1p3p steuern könnten.