evcc-io/evcc #13366: api/state: v0.125.0 vs. v0.121.1 - "batteryConfigured" entfernt? ersetzt durch "batteryDischargeControl"?

jfraundo251158, 2024-04-09:
### Describe the bug

Liebe Kollegen, meine WatchApp - die auf der API aufsetzt - ist nicht mehr kompatibel mit mindestens der v0.125.0. Wurde "batteryConfigured" entfernt und/oder ersetzt durch "batteryDischargeControl"?

Die WatchApp wertet nachfolgende Elemente aus - gab es da weitere Anpassungen seit der v0.121.1 ?
"batteryConfigured": true,               --- show battery icon/value if configured
"batteryPower": 1234,                      --- batterieValue ! not sent when batteryConfigured == false
"batterySoc": 73,                              --- batterieLoad  ! not sent when batteryConfigured == false
"gridConfigured": true,
"gridPower": 8.9
"homePower": 373.9000000000001
"chargePower": 3260
"chargedEnergy": 5689.099999999999
"connected": true
"mode": "pv"

Vielen Dank für schnelles feedback!

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
.
```

### Log details

```text
.
```

### What type of operating system are you running?

Linux

mdkeil, 2024-04-09:
batteryConfigured wurde mit 0.125.0 aus der public API entfernt. #12498

jfraundo251158, 2024-04-09:
Gut zu wissen - Danke! Wie soll sich aus anderem Element ableiten, ob eine Batterie vorhanden ist oder nicht?

closed 2024-04-09 by andig

mdkeil, 2024-04-09:
Ich würde vermuten, dass es den Battery-Block ohne Batterie nicht gibt.. sonst halt auf batterySoc testen.

[image]

andig, 2024-04-09:
len(battery)>0