evcc-io/evcc #13386: Mercedes: Fahrzeug wird nicht mehr erkannt / ist nicht erreichbar

niniob, 2024-04-10:
### Describe the bug

Seit dem Upgrade auf 0.125.0 wird unser Mercedes nicht mehr von evcc erkannt. Nach dem Verbinden mit der Wallbox erscheint in evcc zunächst "Gastfahrzeug" (Verbunden). Wenn man das Fahrzeug dann manuell ändert, bekommt man einen roten Balken und eine durchgestrichene Wolke mit dem Tooltip "Fahrzeug war nicht erreichbar. Versuche evcc neu zu starten". Neustart ändert aber leider nichts. 

Mit `sudo evcc vehicle` werden die aktuellen Daten aber ohne Fehler angezeigt:
```
[main  ] INFO 2024/04/10 23:49:29 evcc 0.125.0
[main  ] INFO 2024/04/10 23:49:29 using config file: /etc/evcc.yaml
[db    ] INFO 2024/04/10 23:49:29 using sqlite database: /root/.evcc/evcc.db
[mercedes] DEBUG 2024/04/10 23:49:29 identity.NewIdentity - token not valid - Add expiry
[mercedes] DEBUG 2024/04/10 23:49:29 identity.NewIdentity - token not valid - database token check started
Soc:           74%
Capacity:      90.0kWh
Charge status: B
Range:         248km
Odometer:      54237km
Finish time:   2024-04-11 00:00:00 +0200 CEST
Position:      0,0
Features:      []
```

**ABER:** Wenn evcc zunächst gestoppt wird (`sudo systemctl stop evcc`) und über `sudo evcc --log debug` wieder gestartet wird, dann funktioniert es. 

### Steps to reproduce

1. Fahrzeug mit Wallbox verbinden
2. In evcc Fahrzeug manuell von "Gastfahrzeug" auf "EQV 300" ändern
3. Neben Fahrzeugnamen erscheint durchgestrichene Wolke --> funktioniert nicht
4. `sudo systemctl stop evcc`
5. `sudo evcc --log debug`  --> funktioniert

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: ...

interval: 15s # control cycle interval

sponsortoken: ...

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: false

meters:
- type: template
  template: solax 
  id: 2  
  host: 192.168.107.176  
  port: 502  
  usage: grid  
  modbus: tcpip  
  name: grid1
- type: template
  template: solax 
  id: 2  
  host: 192.168.107.176  
  port: 502  
  usage: pv  
  modbus: tcpip  
  name: pv2
- type: template
  template: solax 
  id: 2  
  host: 192.168.107.176  
  port: 502  
  usage: battery  
  modbus: tcpip  
  name: battery3

chargers:
- type: template
  template: innogy-ebox 
  id: 1  
  host: 192.168.107.220  
  port: 5555  
  modbus: tcpip  
  name: wallbox4

loadpoints:
- title: Garage
  charger: wallbox4
  mode: pv

site:
  title: evcc
  meters:
    grid: grid1
    pv:
    - pv2
    battery:
    - battery3

vehicles:
  - name: EQV
    title: EQV 300
    type: mercedes
    account: ... # Mercedes Me Nutzer-Id (email)
    region: EMEA # MB me Region (EMEA, APAC, NORAM)
    vin: ... # Erforderlich, wenn mehr als ein Fahrzeug im Account registriert
    capacity: 90 # Akkukapazität in kWh (optional)
    tokens:
      access: ...
      refresh: ...

modbusproxy:
  - port: 5555
    uri: 192.168.107.220:5555
```

### Log details

```text
[main  ] INFO 2024/04/10 23:33:57 evcc 0.125.0
[main  ] INFO 2024/04/10 23:33:57 using config file: /etc/evcc.yaml
[main  ] INFO 2024/04/10 23:33:57 starting ui and api at :7070
[db    ] INFO 2024/04/10 23:33:58 using sqlite database: /root/.evcc/evcc.db
[proxy-5555] DEBUG 2024/04/10 23:33:58 modbus proxy for 192.168.107.220:5555 listening at :5555
[mercedes] DEBUG 2024/04/10 23:33:58 identity.NewIdentity - token not valid - Add expiry
[mercedes] DEBUG 2024/04/10 23:33:58 identity.NewIdentity - token not valid - database token check started
[site  ] WARN 2024/04/10 23:33:58 battery configured but residualPower is missing or <= 0 (add residualPower: 100 to site), see https://docs.evcc.io/en/docs/reference/configuration/site#residualpower
[site  ] INFO 2024/04/10 23:33:58 site config:
[site  ] INFO 2024/04/10 23:33:58   meters:      grid ✓ pv ✓ battery ✓
[site  ] INFO 2024/04/10 23:33:58     grid:      power ✓ energy ✓ currents ✗
[site  ] INFO 2024/04/10 23:33:58     pv 1:      power ✓ energy ✓ currents ✗
[site  ] INFO 2024/04/10 23:33:58     battery 1: power ✓ energy ✗ currents ✗ soc ✓ capacity ✗
[site  ] INFO 2024/04/10 23:33:58   vehicles:
[site  ] INFO 2024/04/10 23:33:58     vehicle 1: range ✓ finish ✓ status ✓ climate ✗ wakeup ✗
[lp-1  ] INFO 2024/04/10 23:33:58 loadpoint 1:
[lp-1  ] INFO 2024/04/10 23:33:58   mode:        pv
[lp-1  ] INFO 2024/04/10 23:33:58   charger:     power ✓ energy ✗ currents ✓ phases ✗ wakeup ✗
[lp-1  ] INFO 2024/04/10 23:33:58   meters:      charge ✓
[lp-1  ] INFO 2024/04/10 23:33:58     charge:    power ✓ energy ✗ currents ✓
[lp-1  ] DEBUG 2024/04/10 23:33:58 phase timer inactive
[lp-1  ] DEBUG 2024/04/10 23:33:58 pv timer inactive
[site  ] WARN 2024/04/10 23:33:58 interval <30s can lead to unexpected behavior, see https://docs.evcc.io/docs/reference/configuration/interval
[site  ] DEBUG 2024/04/10 23:33:58 ----
[lp-1  ] DEBUG 2024/04/10 23:33:58 charge power: 0W
[site  ] DEBUG 2024/04/10 23:33:59 pv power: 0W
[site  ] DEBUG 2024/04/10 23:33:59 battery soc: 54%
[site  ] DEBUG 2024/04/10 23:33:59 battery power: 712W
[site  ] DEBUG 2024/04/10 23:33:59 grid meter: 128W
[site  ] DEBUG 2024/04/10 23:33:59 site power: 840W
[lp-1  ] DEBUG 2024/04/10 23:33:59 charge currents: [0.07 0.11 0.07]A
[lp-1  ] DEBUG 2024/04/10 23:33:59 charger status: B
[lp-1  ] INFO 2024/04/10 23:33:59 car connected
[lp-1  ] DEBUG 2024/04/10 23:33:59 vehicle api refresh
[lp-1  ] DEBUG 2024/04/10 23:33:59 pv timer elapse
[lp-1  ] DEBUG 2024/04/10 23:33:59 pv timer inactive
[main  ] DEBUG 2024/04/10 23:33:59 vehicle status: B (EQV 300)
[lp-1  ] INFO 2024/04/10 23:33:59 vehicle updated: unknown -> EQV 300
[lp-1  ] DEBUG 2024/04/10 23:33:59 vehicle status: B
[lp-1  ] DEBUG 2024/04/10 23:33:59 vehicle soc: 74%
[lp-1  ] DEBUG 2024/04/10 23:33:59 vehicle range: 248km
[lp-1  ] DEBUG 2024/04/10 23:33:59 pv charge current: 0A = 0A + -1.22A (840W @ 3p)
[lp-1  ] DEBUG 2024/04/10 23:33:59 pv enable timer reset
[lp-1  ] DEBUG 2024/04/10 23:33:59 pv timer inactive
[site  ] DEBUG 2024/04/10 23:34:13 ----
[lp-1  ] DEBUG 2024/04/10 23:34:13 charge power: 0W
[site  ] DEBUG 2024/04/10 23:34:14 pv power: 0W
[site  ] DEBUG 2024/04/10 23:34:14 battery soc: 54%
[site  ] DEBUG 2024/04/10 23:34:14 battery power: 837W
[site  ] DEBUG 2024/04/10 23:34:14 grid meter: -0W
[site  ] DEBUG 2024/04/10 23:34:14 site power: 837W
[lp-1  ] DEBUG 2024/04/10 23:34:14 vehicle odometer: 54237km
[lp-1  ] DEBUG 2024/04/10 23:34:14 charge currents: [0.07 0.07 0.11]A
[lp-1  ] DEBUG 2024/04/10 23:34:14 charger status: B
[lp-1  ] DEBUG 2024/04/10 23:34:14 pv charge current: 0A = 0A + -1.21A (837W @ 3p)
[site  ] DEBUG 2024/04/10 23:34:28 ----
[lp-1  ] DEBUG 2024/04/10 23:34:28 charge power: 0W
[site  ] DEBUG 2024/04/10 23:34:29 pv power: 0W
[site  ] DEBUG 2024/04/10 23:34:29 battery soc: 54%
[site  ] DEBUG 2024/04/10 23:34:29 battery power: 795W
[site  ] DEBUG 2024/04/10 23:34:29 grid meter: -0W
[site  ] DEBUG 2024/04/10 23:34:29 site power: 795W
[lp-1  ] DEBUG 2024/04/10 23:34:29 charge currents: [0.07 0.11 0.11]A
[lp-1  ] DEBUG 2024/04/10 23:34:29 charger status: B
[lp-1  ] DEBUG 2024/04/10 23:34:29 pv charge current: 0A = 0A + -1.15A (795W @ 3p)
[site  ] DEBUG 2024/04/10 23:34:43 ----
[lp-1  ] DEBUG 2024/04/10 23:34:43 charge power: 0W
[site  ] DEBUG 2024/04/10 23:34:44 pv power: 0W
[site  ] DEBUG 2024/04/10 23:34:44 battery soc: 54%
[site  ] DEBUG 2024/04/10 23:34:44 battery power: 796W
[site  ] DEBUG 2024/04/10 23:34:44 grid meter: 44W
[site  ] DEBUG 2024/04/10 23:34:44 site power: 840W
[lp-1  ] DEBUG 2024/04/10 23:34:44 charge currents: [0.07 0.11 0.07]A
[lp-1  ] DEBUG 2024/04/10 23:34:44 charger status: B
[lp-1  ] DEBUG 2024/04/10 23:34:44 pv charge current: 0A = 0A + -1.22A (840W @ 3p)
[site  ] DEBUG 2024/04/10 23:34:58 ----
[lp-1  ] DEBUG 2024/04/10 23:34:58 charge power: 0W
[site  ] DEBUG 2024/04/10 23:34:59 pv power: 0W
[site  ] DEBUG 2024/04/10 23:34:59 battery soc: 54%
[site  ] DEBUG 2024/04/10 23:34:59 battery power: 795W
[site  ] DEBUG 2024/04/10 23:34:59 grid meter: -0W
[site  ] DEBUG 2024/04/10 23:34:59 site power: 795W
[lp-1  ] DEBUG 2024/04/10 23:34:59 charge currents: [0.07 0.11 0.07]A
[lp-1  ] DEBUG 2024/04/10 23:34:59 charger status: B
[lp-1  ] DEBUG 2024/04/10 23:34:59 pv charge current: 0A = 0A + -1.15A (795W @ 3p)
```

### What type of operating system are you running?

Linux

### Version

0.125.0

andig, 2024-04-11:
/c @ReneNulschDE

ReneNulschDE, 2024-04-11:
Kannst Du bitte noch den Log output eines start mit "sudo systemctl start evcc" hinzufügen. Ich vermute, deine Datenbank bei start als Service ist eine andere als mit sudo evcc. In der /root/.evcc/evcc.db ist der korrekte Token und in der anderen ein alter/falscher/keiner...

Mich interessiert folgende Zeile:
`[db    ] INFO 2024/04/10 23:33:58 using sqlite database: /root/.evcc/evcc.db`

niniob, 2024-04-11:
> Kannst Du bitte noch den Log output eines start mit "sudo systemctl start evcc" hinzufügen. Ich vermute, deine Datenbank bei start als Service ist eine andere als mit sudo evcc. In der /root/.evcc/evcc.db ist der korrekte Token und in der anderen ein alter/falscher/keiner...
> 
> Mich interessiert folgende Zeile: `[db ] INFO 2024/04/10 23:33:58 using sqlite database: /root/.evcc/evcc.db`
> 
> Wenn es dort eine andere DB ist, kannst du in der evcc.yaml einen festen pfad zur `/root/.evcc/evcc.db` eintragen.

Ok, alles klar ... aber wie erzeuge ich den Log ouput beim Start mit "sudo systemctl start evcc"?

ReneNulschDE, 2024-04-11:
`sudo journalctl -u evcc --since "yesterday"`

niniob, 2024-04-11:
Vielen Dank! Du hast Recht, es werden wohl unterschiedliche Datenbanken genutzt.

`Apr 11 11:33:03 evcc evcc[1873]: [db    ] INFO 2024/04/11 11:33:03 using sqlite database: /var/lib/evcc/evcc.db`

Woher kommt es das unterschiedliche Datenbanken verwendet werden bzw. was ist "richtig"? Sollte ich `/root/.evcc/evcc.db` oder `/var/lib/evcc/evcc.db` verwenden?

ReneNulschDE, 2024-04-11:
Ich bin kein evcc spezie, kenne mich nur mit der MB-vehicle integration aus. Meine Vermutung ist, dass die Datenbank im Pfad des aktuellen Benutzers angelegt wird und dieser Pfad ist bei sudo-Nutzung und systemd-service unterschiedlich. Daher kannst Du den pfad in der evcc.yaml definieren.

Aktuell hast du in der `/root/.evcc/evcc.db` den richtigen Token zum Zugriff auf die MB-API.

```
# database configuration for persisting charge sessions and settings
database:
   type: sqlite
  dsn: /root/.evcc/evcc.db
```

@andig : Hast etwas hinzuzufügen?

andig, 2024-04-11:
Die DB über cli (mit sudo) und Service ist laut Logs oben identisch:

> using sqlite database: /root/.evcc/evcc.db

> using sqlite database: /root/.evcc/evcc.db

Damit sollte da auch das gleiche Token verwendet werden und das identisch funktionieren.

andig, 2024-04-11:
Ah, ich sehe schon. Das Log aus dem Issue ist nicht das Log vom Service, sagt also nichts über die verwendete DB aus...

ReneNulschDE, 2024-04-12:
Danke, schliesse das issue bitte.

closed 2024-04-12 by niniob

MrIngenieur, 2024-04-16:
> Danke, schliesse das issue bitte. 

Ist die Lösung dann jetzt, dass jeder der vor der 0.125.0 einen aktiven Mercedes Token hatte, seine Datenbank wegwerfen muss?
Der aktuelle Assistent für die Tokengenerierung bzw. vermutliche eher das Skript/die Routine für das Auslesen der evcc.yaml arbeitet hier ja dem Anschein nach nicht korrekt, wenn der Token aus der YAML in der falschen Datenbank landet. (Vielleicht auch ein Linux/Rechteproblem, wenn der systemd-Aufruf anders arbeitet als der sudo Aufruf.) 

Ich fände es toll, wenn man diesem Fehler auf den Grund gehen könnte und das auch für bestehende Installationen wieder reparieren kann. 
Wenn ich das richtig verstehe, dann müsste ja beim YAML auslesen "nur" darauf geachtet werden, dass die Token in den korrekten evcc.db-Pfad kopiert werden und dieser Pfad auch per systemd-Aufruf verwendet wird, korrekt? 
Nicht jeder möchte seine Ladehistorie einfach so wegwerfen und auch nicht jeder ist so bewandert, dass er einfach irgendwelche Datenbank-Dateien rauslöscht. 

Daher wäre das Issue aus meiner Sicht noch nicht wirklich geschlossen. 
@andig @ReneNulschDE wie seht ihr das?

ReneNulschDE, 2024-04-16:
Meiner Ansicht nach: Du musst Dich nur entscheiden eine Datenbank zu nehmen und nicht zwei... Wie schon mehrfach geschrieben, lautet meine Empfehlung einen db-pfad in der Config zu setzen. 

Die Tokens werden in der Datenbank erstellt, die im aktuellen Kontext, die vom evcc system vorgegeben ist. Meine Komponente nutzt die Standard-Setting-APIs von evcc.

Man kann diskutieren ob der aktuelle Weg von evcc optimal ist wie der default-db pfad ermittelt wird. Aktuell ist es der Homepath des aktuellen users + /evcc/evcc.db. Und dieser ist bei Service/sudo/user Aufruf halt unterschiedlich je nach Umgebung. Meiner Meinung wäre es besser die DB im Pfad der Config Datei zu managen, wenn kein Pfad in der config definiert ist - aber ich bin weder ein evcc-spezie noch user.

Die Datenbankdaten können mit dem Befehl evcc settings set mercedes.[account]-[region] "" gelöscht werden. (Sofern man die richtige evcc.yaml und damit evcc.db nutzt.

Ich selbst kann und werde hier nichts weiter machen. Das Problem hat ja nicht nur diese Integration sondern wurde auch von den Tesla usern schon angesprochen. Für mich ist dies etwas für das Core-Team. Oder etwas für Pro-User die mit eigenen PRs die Doku verbessern können.

Grüße
Rene

MrIngenieur, 2024-04-17:
Hi Rene, 
Danke dir fürs Feedback, genau das meinte ich auch mit "wie evcc.yaml" ausgelesen und in Datenbank übertragen wird bzw. Welche Datenbank bei welchem Aufruf genutzt wird. 
So wie ich das verstehe, liegt das nicht an der Mercedes-Integration, sondern an der Art und Weise wie EVCC hier funktioniert und mit seinen Datenbankzugriffen umgeht. 
--> Das sollte behoben werden, dann funktioniert Mercedes-Integration (und die jedes anderen Herstellers) auch wieder sauber. 
Daher meine Frage: Ist das Problem schon gelöst oder müsste da nicht EVCC in den Basisfunktionen nachbessern? @andig

andig, 2024-04-17:
Welches Problem?! Wenns eins gibt bitte Issue dazu.

MrIngenieur, 2024-04-18:
> Welches Problem?! Wenns eins gibt bitte Issue dazu. 

Done #13479