evcc-io/evcc #13390: KW Zähler in Ladevorgänge nicht korrekt

philg1234, 2024-04-11:
### Describe the bug

Hallo,

mir ist aufgefallen, dass die Geladen kWh nicht mit dem meines Chargers (goE) zusammenpasst.
Die Anfang und End Zählerstände, welche in evcc unter den Ladevorgängen angezeigt werden sind korrekt, allerdings ist die Rechnung nicht korrekt.

Beispiel:

[image]

Hier wird 33,8 angezeigt obwohl die Differenz = 42,2 ist.

Grüße
Phil

### Steps to reproduce

1. Ladevorgänge Differenz zwischen Anfang-Zählerstand und End-Zählerstand

### Configuration details

```yaml
xx
```

### Log details

```text
xxx
```

### What type of operating system are you running?

Linux

### Version

v0.124.10

mdkeil, 2024-04-11:
Ist das bei jeder Ladesession falsch?-- hab gerade mal stichpunktartig bei mir nachgeschaut.. scheint in meiner Konstellation zu stimmen.

philg1234, 2024-04-11:
Ich habe nicht jede kontrolliert aber Stichpunktartig 10 - 15 und alle fehlerhaft.

dmnkmnr, 2024-04-11:
Ich habe ein ähnliches/das gleiche Problem. Die in evcc angezeigten (errechneten) Werte stimmen nicht mit den tatsächlichen Werten (von der Wallbox gemonitort) überein. Evcc geht davon aus, dass mit 8,7 kW geladen werden kann, obwohl es auto-bedingt nur 2,785 kW sind. Somit stimmen auch die kWh in Summe nicht. Anbei aktuelle Screenshots, einmal von ivcc, einmal von meiner Wallbox.
[image]
[image]

VolkerK62, 2024-04-11:
@dmnkmnr das ist OT. Du hast ein anderes Problem. Mach dazu am besten eine Diskussion auf.

VolkerK62, 2024-04-11:
@mdkeil auf bei go-E?

Sowohl `chargedEnergy` als auch `totalEnergy` kommen von der Wallbox.

mdkeil, 2024-04-11:
Nein, mit einer anderen WB.. wusste auch nicht, dass die go-E auch die chargedEnergy direkt liefert. (wird der Wert dann auch direkt verwendet?).. Vielleicht berechnet die go-E die Sessions fehlerhaft, bzw. Nullt die zwischendurch (warum auch immer).. ggfs. sollte man da noch eine Plausibilitätsprüfung einbauen..

closed 2024-04-11 by andig

andig, 2024-04-11:
Mangels Logfile nicht angeschaut

philg1234, 2024-04-11:
Okay. 😅😅.

Heißt logfiles nachliefern oder neuen Issue eröffnen?

philg1234, 2024-04-12:
Nachtrag:

ich wollte soeben mit "evcc --log debug" die Logs sammeln, aber leider nimmt er die falsche Datenbank etc.
Kann ich bei evcc --log debug angeben welches yaml und Datenbank er benutzen soll?

philg1234, 2024-04-12:
Edit:

habe nun den Debug Modus gestartet und eine Testladung durchgeführt.

In diesem Test hat alles Problemlos funktioniert und die Rechnung war korrekt. Zuvor hatte ich noch ein Update auf die aktuelle Version durchgeführt + ein Update der go E Charger Station.

Ich werde es die nächsten Ladungen im Blick behalten und ansonsten ein neues Thema eröffnen.

philg1234, 2024-05-08:
Fehler ist wieder aufgetreten.

Ich werde einen neuen Issue hierfür mit Logfiles erstellen.