evcc-io/evcc #13411: UI: schlechte bis keine Bedienbarkeit des Menüs bei iPhone

TTXaser2, 2024-04-12:
### Describe the bug

Hallo, 

seit dem Update auf 0.125.0 hab ich Probleme mit der Bedienbarkeit des Menüs auf einem iPhone 12, iOS-Version 17.4.1 mit als Icon auf dem Homescreen abgelegtem Lesezeichen der EVCC-Seite. Ich denk der folgende Screenshot sagt alles.
[image]

Auch Fehler- und Warnmeldungen von EVCC sind sehr schlecht klickbar. Mit der zuvor verwendeten Version 0.124.9 hatte ich das Problem nicht.

### Steps to reproduce

1. Rufe in Safari die EVCC-Webseite auf. Tippe anschließend auf das Teilen-Icon (viereckiges Kästchen mit Pfeil nach oben). Wähle "Zum Homebildschirm" und gebe dem Bookmark einen Namen. 
2. Schliesse Safari
3. Rufe die EVCC-Webseite über das im Homebildschirm angelegte Icon auf
4. Siehe den rechts ganz nach oben gerutschten, nahezu unbedienbaren Menü-Button

### Configuration details

```yaml
Keine Config, da UI-Problem
```

### Log details

```text
Kein Log, da UI-Problem
```

### What type of operating system are you running?

Linux

### Version

Version: `0.125.0`

naltatis, 2024-04-12:
Ja, eigentlich sollte das in Problem gefixt sein. @TTXaser2 probier mal, wie von @StefanSchoof vorgeschlagen, den Bookmark neu anzulegen. Das scheint ein Bug beim letzten iOS Update zu sein, gegen das wir nichts machen können.

An-Wi, 2024-04-13:
Ich nutze auch iPhone 12 mit iOS 17.4.1 und kann das Problem überhaupt nicht bestätigen oder reproduzieren:
Weder das "alte" Lesezeichen, das ich vor einiger Zeit angelegt hatte, noch ein soeben testweise neu angelegtes Lesezeichen via evcc-Version 0.125.0 mit Safari Browser zeigen bei mir das beschriebene Problem.
@TTXaser2 Hast du die "Steps to reproduce" wirklich nochmal neu ausgeführt?

Der einzige Unterschied, den ich sonst sehe, falls die Info "operating system" korrekt ist: evcc läuft bei mir auf einem Windows System

GrimmiMeloni, 2024-04-13:
Ich hatte das Problem auch. Neuanlegen wie hier beschrieben hat es auch für mich gelöst. Danke.

naltatis, 2024-04-13:
Ok, ich geh davon aus, dass das das beschrieben Problem ist.
@TTXaser2 falls das Neuanlagen nicht hilft gib Bescheid. Dann mach ich den issue wieder auf.

closed 2024-04-13 by naltatis

TTXaser2, 2024-04-18:
Sorry für die späte Reaktion, war die letzten Tage elektronisch abgekoppelt.

Der Vollständigkeit halber: Vielen Dank für die vielen Hilfestellungen, Löschen und Neuanlegen hat das Problem gelöst, ganz wie von @StefanSchoof oben beschrieben.

naltatis, 2024-04-18:
@TTXaser2 danke für die Rückmeldung!