evcc-io/evcc #13416: EVCC im Browser als Vollbild

StevieC121176, 2024-04-13:
Bei mir läuft EVCC auf meinem Kühlschrankdisplay. Könnte man in EVCC einen Vollbildmodus „erzwingen“ sodaß das komplette Display genutzt wird? So könnten die Loadpoints übereinander angezeigt werden , was eine bessere Optik ergeben würde. Möglich scheint das zu sein, ich hab mal ein vergleichsbild einer anderen Anwendung angehängt, die auf dem Kühlschrank im Vollbild läuft. Browser ist der Kühlschrank eigene. (Betriebssystem Android)

evcc aktuell

[image]

Vollbild

[image]

naltatis, 2024-04-13:
Das zweite ist auch einfache eine Webseite die du im Browser aufrufst? Oder ist das ne App?

StevieC121176, 2024-04-13:
> Das zweite ist auch einfache eine Webseite die du im Browser aufrufst? Oder ist das ne App?

das ist auch eine normale Webseite. Ähnlich wie EVCC auf dem RaspberryPi installiert und aufgerufen durch http://ip-Adresse:3000/

naltatis, 2024-04-13:
Thema Loadpoints übereinander:
Aktuell machen wir das "nur" von drei breite des Screens abhängig. Ich sehe aber den Punkt, dass das bei sehr hochformatigen Screens untereinander schöner sein könnte. 

Hast du die Auflösung von deinem Screen oder besser noch die viewport Größe da? Kannst mal diese Seite dort aufrufen:

https://whatismyviewport.com

StevieC121176, 2024-04-13:
Ja die Seite funktioniert und liefert folgendes:

[image]

StevieC121176, 2024-04-13:
Ich hab noch folgendes zum Display gefunden, Auflösung 1080x1920 FullHD , 103,5 DPI bei 26,5cmx47cm

naltatis, 2024-04-13:
Vollbild: Ist die Solectrus Seite bei dir sofort Fullscreen oder gehst du über die Funktion im Menü der Webapp manuell in Vollbild?

naltatis, 2024-04-13:
Zur Auflösung: Interessant, dass der Browser Viewport (982px) nicht die volle Bildschirmbreite (1080px) bekommt. Ab 992px (also genau in dem Gap) beginnt bei uns der [Large-Breakpoint](https://getbootstrap.com/docs/5.2/layout/breakpoints/#available-breakpoints) bei dem die Ladepunkt ein Grid werden und nicht mehr left-right slidebar sind.

StevieC121176, 2024-04-13:
> Vollbild: Ist die Solectrus Seite bei dir sofort Fullscreen oder gehst du über die Funktion im Menü der Webapp manuell in Vollbild?

Der Fullscreen wird über einen Menüpunkt der Webapp aufgerufen

[Fullscreen request](https://developer.mozilla.org/en-US/docs/Web/API/Element/requestFullscreen)

StevieC121176, 2024-04-13:
> Zur Auflösung: Interessant, dass der Browser Viewport (982px) nicht die volle Bildschirmbreite (1080px) bekommt. Ab 992px (also genau in dem Gap) beginnt bei uns der [...]

Ich denke das liegt an der Menüstruktur des Kühlschranks. Da gibt es das Hauptfenster mit den Menüpunkten oben und alles andere wie Browser , Wetter usw läuft in einem etwas kleineren Fenster. Allerdings wird durch den Fullscreen request dann der gesamte Bildschirm nutzbar (1080px x 1920px)

naltatis, 2024-04-13:
> Der Fullscreen wird über einen Menüpunkt der Webapp aufgerufen

Ok, dann ergibt das Sinn. Dieser Part fehlte mir. Hatte schon nach unterschieden im Manifest und den Meta-Tags gesucht :D

Über die Browser-UI kannst du Fullscreen nicht triggern?

StevieC121176, 2024-04-13:
Nein, über die Browser UI geht das nicht. Da kann man nur den Verlauf anzeigen und die Favoriten verwalten.

mucki12, 2024-04-15:
@StevieC121176 
Dürfte ich mal fragen welcher Kühlschrank das überhaupt ist? Ein Samsung Family Hub kann es ja wahrscheinlich nicht sein, oder hat Samsung mittlerweile das Betriebssystem ausgewechselt?

StevieC121176, 2024-04-15:
> @StevieC121176 Dürfte ich mal fragen welcher Kühlschrank das überhaupt ist? 

Haier Cube90 Series 9

closed 2024-04-18 by andig