evcc-io/evcc #13417: VW: Fahrzeug nicht erreichbar

urspb, 2024-04-13:
### Describe the bug

Seit dem Upgrade auf 0.125.0 wird in der Web-Oberfläche der SOC nicht mehr angezeigt, sondern eine durchgestrichene Wolke:
[image]

### Steps to reproduce

1. Fahrzeug mit Wallbox verbinden
2. Neben Fahrzeugnamen erscheint durchgestrichene Wolke --> funktioniert nicht
3. Mehrfacher Restart von evcc hilft nicht

### Configuration details

```yaml
Konfiguration (/home/urs/evcc.yaml):

# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

# trial token, valid until 2024-05-03
sponsortoken: *****

log: debug
levels:
  cache: error
  lp-1: trace
  vw: trace

# unique installation id
plant: *****

interval: 20s # control cycle interval

meters:
- type: template
  template: sma-hybrid
  id: 3
  host: 192.168.1.190
  port: 502
  usage: pv
  modbus: tcpip
  name: pv1
- type: template
  template: sma-hybrid
  id: 3
  host: 192.168.1.190
  port: 502
  usage: battery
  modbus: tcpip
  name: battery2
- type: template
  template: sma-home-manager
  usage: grid
  host: 192.168.1.195
  name: grid3

vehicles:
- type: template
  template: vw
  title: ID.4
  user: *****
  password: *****
  vin: *****
  capacity: 77
  timeout: 35s
  name: IDA

chargers:
- name: Easee
  type: template
  template: easee
  user: *****
  password: *****
  charger: EHYP4BHD
  timeout: 20s # optional
#  authorize: # Steuert ob evcc die Authentifizierung am Charger vornimmt. Vorteil ist ein kontrollierter Ladestart. Nicht kompatibel mit RFID Identifikation von Fahrzeugen. (optional)

influx:
  url: http://localhost:8086
  database: evcc_log
  token: *****
  org: urspb

tariffs:
  currency: EUR # (default EUR)
  grid:
    # static grid price
    type: fixed
    price: 0.22 # [currency]/kWh

  feedin:
    # rate for feeding excess (pv) energy to the grid
    type: fixed
    price: 0.082 # [currency]/kWh

site:
  title: WeWeg12d
  residualPower: 100
  meters:
    grid: grid3
    pv:
    - pv1
    battery:
    - battery2

loadpoints:
- title: Carport # display name for UI
  charger: Easee # Wallbox Referenz
  vehicle: IDA # Referenz auf Standardfahrzeug
  mode: pv # charge mode (off, now, minpv, pv)

hems:
  type: sma
  AllowControl: false
```

### Log details

```text
Started evcc.service - evcc.
[main  ] INFO 2024/04/13 11:54:34 evcc 0.125.0
[main  ] INFO 2024/04/13 11:54:34 using config file: /etc/evcc.yaml
[main  ] INFO 2024/04/13 11:54:34 starting ui and api at :7070
[db    ] INFO 2024/04/13 11:54:34 using sqlite database: /var/lib/evcc/evcc.db
[main  ] ERROR 2024/04/13 11:54:37 creating vehicle IDA failed: cannot create vehicle type 'template': cannot create vehicle type 'vw': missing state
[lp-1  ] DEBUG 2024/04/13 11:54:37 charge total import: 516.483kWh
[semp  ] WARN 2024/04/13 11:54:37 SEMP_BASE_URL unspecified, using http://192.168.1.25:7070 instead
[site  ] INFO 2024/04/13 11:54:37 site config:
[site  ] INFO 2024/04/13 11:54:37   meters:      grid ✓ pv ✓ battery ✓
[site  ] INFO 2024/04/13 11:54:37     grid:      power ✓ energy ✓ currents ✓
[site  ] INFO 2024/04/13 11:54:37     pv 1:      power ✓ energy ✓ currents ✗
[site  ] INFO 2024/04/13 11:54:37     battery 1: power ✓ energy ✓ currents ✗ soc ✓ capacity ✗
[site  ] INFO 2024/04/13 11:54:37   vehicles:
[site  ] INFO 2024/04/13 11:54:37     vehicle 1: range ✗ finish ✗ status ✗ climate ✗ wakeup ✗
[lp-1  ] INFO 2024/04/13 11:54:37 loadpoint 1:
[lp-1  ] INFO 2024/04/13 11:54:37   mode:        pv
[lp-1  ] INFO 2024/04/13 11:54:37   charger:     power ✓ energy ✓ currents ✓ phases ✓ wakeup ✗
[lp-1  ] INFO 2024/04/13 11:54:37   meters:      charge ✓
[lp-1  ] INFO 2024/04/13 11:54:37     charge:    power ✓ energy ✓ currents ✓
[site  ] DEBUG 2024/04/13 11:54:37 set buffer soc: 50
[site  ] DEBUG 2024/04/13 11:54:37 set buffer start soc: 60
[site  ] DEBUG 2024/04/13 11:54:37 set priority soc: 20
[site  ] DEBUG 2024/04/13 11:54:37 set battery discharge control: false
[lp-1  ] DEBUG 2024/04/13 11:54:37 phase timer inactive
[lp-1  ] DEBUG 2024/04/13 11:54:37 pv timer inactive
[lp-1  ] INFO 2024/04/13 11:54:37 vehicle updated: unknown -> ID.4
[site  ] WARN 2024/04/13 11:54:37 interval <30s can lead to unexpected behavior, see https://docs.evcc.io/docs/reference/configuration/interval
[site  ] DEBUG 2024/04/13 11:54:37 ----
[lp-1  ] DEBUG 2024/04/13 11:54:37 charge power: 0W
[site  ] DEBUG 2024/04/13 11:54:38 pv power: 4786W
[site  ] DEBUG 2024/04/13 11:54:38 battery soc: 30%
[site  ] DEBUG 2024/04/13 11:54:38 battery power: -4184W
[site  ] DEBUG 2024/04/13 11:54:38 grid meter: 48W
[site  ] DEBUG 2024/04/13 11:54:38 grid powers: [-92 35 105]W
[site  ] DEBUG 2024/04/13 11:54:38 grid currents: [-0.786 0.474 0.823]A
[site  ] DEBUG 2024/04/13 11:54:38 site power: -4036W
[lp-1  ] DEBUG 2024/04/13 11:54:38 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2024/04/13 11:54:38 charge total import: 516.483kWh
[lp-1  ] DEBUG 2024/04/13 11:54:38 charger status: A
[lp-1  ] INFO 2024/04/13 11:54:38 car disconnected
[lp-1  ] DEBUG 2024/04/13 11:54:38 set session soc limit: 0
[lp-1  ] DEBUG 2024/04/13 11:54:38 set session energy limit: 0
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.125.0

andig, 2024-04-13:
Probier mal obs wieder neue Terms&Conditions gibt. Ansonst

    evcc vehicle --log trace

urspb, 2024-04-13:
Soweit ich weiß, keine neuen Nutzungsbedingungen...
Hier der Output von `evcc vehicle --log trace`:
```
[main  ] INFO 2024/04/13 13:25:21 evcc 0.125.0
[main  ] INFO 2024/04/13 13:25:21 using config file: /home/urs/evcc.yaml
[db    ] INFO 2024/04/13 13:25:22 using sqlite database: /home/urs/.evcc/evcc.db
[db    ] TRACE 2024/04/13 13:25:22 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="settings" -1 <nil>
[db    ] TRACE 2024/04/13 13:25:22 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "settings" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2024/04/13 13:25:22 SELECT * FROM `settings` LIMIT 1 -1 <nil>
[db    ] TRACE 2024/04/13 13:25:22 SELECT * FROM `settings` 0 <nil>
[db    ] TRACE 2024/04/13 13:25:22 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="devices" -1 <nil>
[db    ] TRACE 2024/04/13 13:25:22 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="device_details" -1 <nil>
[db    ] TRACE 2024/04/13 13:25:22 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="configs" -1 <nil>
[db    ] TRACE 2024/04/13 13:25:22 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "configs" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2024/04/13 13:25:22 SELECT * FROM `configs` LIMIT 1 -1 <nil>
[db    ] TRACE 2024/04/13 13:25:22 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="config_details" -1 <nil>
[db    ] TRACE 2024/04/13 13:25:22 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "config_details" AND sql IS NOT NULL order by type = "table" desc 2 <nil>
[db    ] TRACE 2024/04/13 13:25:22 SELECT * FROM `config_details` LIMIT 1 -1 <nil>
[db    ] TRACE 2024/04/13 13:25:22 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%CONSTRAINT ""fk_configs_details"" %" OR sql LIKE "%CONSTRAINT fk_configs_details %" OR sql LIKE "%CONSTRAINT `fk_configs_details`%" OR sql LIKE "%CONSTRAINT [fk_configs_details]%" OR sql LIKE "%CONSTRAINT 	fk_configs_details	%") -1 <nil>
[db    ] TRACE 2024/04/13 13:25:22 SELECT count(*) FROM sqlite_master WHERE type = "index" AND tbl_name = "config_details" AND name = "idx_unique" -1 <nil>
[db    ] TRACE 2024/04/13 13:25:22 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%CONSTRAINT ""fk_devices_details"" %" OR sql LIKE "%CONSTRAINT fk_devices_details %" OR sql LIKE "%CONSTRAINT `fk_devices_details`%" OR sql LIKE "%CONSTRAINT [fk_devices_details]%" OR sql LIKE "%CONSTRAINT 	fk_devices_details	%") -1 <nil>
[db    ] TRACE 2024/04/13 13:25:22 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%""device_id"" %" OR sql LIKE "%device_id %" OR sql LIKE "%`device_id`%" OR sql LIKE "%[device_id]%" OR sql LIKE "%	device_id	%") -1 <nil>
[db    ] TRACE 2024/04/13 13:25:22 SELECT * FROM `configs` WHERE `configs`.`class` = 3 ORDER BY id 0 <nil>
[id    ] TRACE 2024/04/13 13:25:22 GET https://emea.bff.cariad.digital/user-login/v1/authorize?client_id=a24fba63-34b3-4d43-b181-942111e6bda8%40apps_vw-dilab_com&code_challenge=EFYyQKienq5FCnKxD0vDTxkptfsREynXEsxWRctWD-M&code_challenge_method=S256&redirect_uri=weconnect%3A%2F%2Fauthenticated&response_type=code+id_token+token&scope=openid+profile+badge+cars+vin&nonce=LVbAUEwTsEJZIBghXULOETafGMIjkiGcQYpPITyWgeg&state=0b523b62-2542-4246-8fdc-97029ac84682
[id    ] TRACE 2024/04/13 13:25:22 GET https://identity.vwgroup.io/oidc/v1/authorize?client_id=a24fba63-34b3-4d43-b181-942111e6bda8%40apps_vw-dilab_com&scope=openid+profile+badge+cars+dealers+vin&response_type=code+id_token+token&redirect_uri=weconnect%3A%2F%2Fauthenticated&state=97d17c37b3b9bfbce5ca52c39a9f693887cb6e66faeb76b4d4b4e7cc77b3bf25&nonce=LVbAUEwTsEJZIBghXULOETafGMIjkiGcQYpPITyWgeg
[id    ] TRACE 2024/04/13 13:25:22 GET https://identity.vwgroup.io/signin-service/v1/signin/a24fba63-34b3-4d43-b181-942111e6bda8@apps_vw-dilab_com?relayState=d6bdffd2ff8e5115e50bec75fee47b325551bce4
[id    ] TRACE 2024/04/13 13:25:22 <!DOCTYPE html>
<html lang="en" xmlns="http://www.w3.org/1999/html">
<head>
    <meta charset="utf-8"/>
    <meta name="viewport"
          content="width=device-width, height=device-height, initial-scale=1, maximum-scale=1, user-scalable=no"/>
    <meta name="identitykit" content="loginIdentifier"/>

<script type="text/javascript">(window.NREUM||(NREUM={})).init={privacy:{cookies_enabled:true},ajax:{deny_list:["bam.eu01.nr-data.net"]},distributed_tracing:{enabled:true}};(window.NREUM||(NREUM={})).loader_config={agentID:"535875755",accountID:"2547829",trustKey:"2217462",xpid:"VgMDVl5RARAEUlFQBQQOXlY=",licenseKey:"NRJS-ecd5929e8859410bbd9",applicationID:"460225982"};;/*! For license information please see nr-loader-spa-1.256.0.min.js.LICENSE.txt */
(()=>{var e,t,r={234:(e,t,r)=>{"use strict";r.d(t,{P_:()=>m,Mt:()=>b,C5:()=>s,DL:()=>E,OP:()=>N,lF:()=>D,Yu:()=>x,Dg:()=>v,CX:()=>c,GE:()=>w,sU:()=>O});var n=r(8632),i=r(9567);const o={beacon:n.ce.beacon,errorBeacon:n.ce.errorBeacon,licenseKey:void 0,applicationID:void 0,sa:void 0,queueTime:void 0,applicationTime:void 0,ttGuid:void 0,user:void 0,account:void 0,product:void 0,extra:void 0,jsAttributes:{},userAttributes:void 0,atts:void 0,transactionName:void 0,tNamePlain:void 0},a={};function s(e){if(!e)throw new Error("All info objects require an agent identifier!");if(!a[e])throw new Error("Info for  [... cut ...] >{let[t,r]=e;Object.keys(i).includes(t)&&(this.sharedContext[t]=r)}))}catch(e){(0,n.Z)("An error occurred while setting SharedContext",e)}}}},8e3:(e,t,r)=>{"use strict";r.d(t,{LP:
[main  ] ERROR 2024/04/13 13:25:22 creating vehicle IDA failed: cannot create vehicle type 'template': cannot create vehicle type 'vw': missing state
.Soc:     vehicle not available: cannot create vehicle type 'template': cannot create vehicle type 'vw': missing state
Capacity: 77.0kWh
Features: [Offline Retryable]
```

StefanSchoof, 2024-04-13:
`https://identity.vwgroup.io/signin-service/v1/consent/marketing` klingt aber genau danach. Hast du versucht einmal in Browser die Seite zu öffnen?

StefanSchoof, 2024-04-13:
Weil noch niemand raus gefunden hat, was die App anders macht als die VW Site um einen gültigen Tokens zu bekommen.

urspb, 2024-04-13:
> Weil noch niemand raus gefunden hat, was die App anders macht als die VW Site um einen gültigen Tokens zu bekommen.

Danke für die Info und ein "Hoch soll'n sie leben" auf VW und seine offene API-Dokumentation.

closed 2024-04-13 by urspb

andig, 2024-04-13:
> Deinem Hinweis habe ich mich unter https://www.volkswagen.de/de.html angemeldet und wurde aufgefordert, eine Einwilligung in personalisierte Werbung zu geben.

Lesson learned: Fehler erstmal nicht bei evcc suchen und den Vorschlägen folgen 👍🏻