evcc-io/evcc #13420: Batterieeinstellungen - Verhindere Entladung im Schnell-Modus und bei geplantem Laden

Dragonkit, 2024-04-13:
### Describe the bug

Seit dem Update auf 0.125.0 geht die Funktion bei der Tesla Powerwall 2 nicht mehr.
Es kommt der Fehler battery mode: 500 Internal Server Error

### Steps to reproduce

Das kann man ganz einfach testen in dem ich den Schwellwert der Kosten runter setze dann kommt der Fehler sobald er auf hohe Ladeleistung geht und aus dem Speicher zieht.

### Configuration details

```yaml
# meter definitions
# name can be freely chosen and is used as reference when assigning meters to site and loadpoints
# for documentation see https://docs.evcc.io/docs/devices/meters
meters:
  - name: my_grid
    type: template
    template: tesla-powerwall
    usage: grid
    host: xxx # IP-Adresse oder Hostname
    password: xxx # Passwort des Benutzers "Kunde". Default sind die letzten 5 Zeichen des auf dem Tesla Gateway genannten Passworts.
    refreshToken: xxxx # Siehe https://docs.evcc.io/docs/devices/meters#tesla-powerwall (optional)
    siteId: # optionale Product ID dieser Energy Site, zum Übersteuern der automatischen Erkennung (optional) 
  - name: my_pv
    type: template
    template: tesla-powerwall
    usage: pv
    host: xxx  # IP-Adresse oder Hostname
    password: xxx # Passwort des Benutzers "Kunde". Default sind die letzten 5 Zeichen des auf dem Tesla Gateway genannten Passworts.
    refreshToken: xxx # Siehe https://docs.evcc.io/docs/devices/meters#tesla-powerwall (optional)
    siteId: # optionale Product ID dieser Energy Site, zum Übersteuern der automatischen Erkennung (optional) 
  - name: my_battery
    type: template
    template: tesla-powerwall
    usage: battery
    host: xxx  # IP-Adresse oder Hostname
    password: xxx # Passwort des Benutzers "Kunde". Default sind die letzten 5 Zeichen des auf dem Tesla Gateway genannten Passworts.
    refreshToken: xxx  # Siehe https://docs.evcc.io/docs/devices/meters#tesla-powerwall (optional)
    siteId: # optionale Product ID dieser Energy Site, zum Übersteuern der automatischen Erkennung (optional) 

# charger definitions
# name can be freely chosen and is used as reference when assigning charger to vehicle
# for documentation see https://docs.evcc.io/docs/devices/chargers
chargers:
  - name: Wallbox_Garage
    type: template
    template: twc3
    host: xxx  # IP-Adresse oder Hostname 
  - name: Wallbox_Gartenhaus
    type: template
    template: twc3
    host: xxx  # IP-Adresse oder Hostname 

# vehicle definitions
# name can be freely chosen and is used as reference when assigning vehicle to loadpoint
# for documentation see https://docs.evcc.io/docs/devices/vehicles
vehicles:
  - name: Tesla_Model_3
    type: template
    template: tesla
    title: Tesla Model 3 # Wird in der Benutzeroberfläche angezeigt (optional)
    accessToken: xxx  # Siehe https://docs.evcc.io/docs/devices/vehicles#tesla
    refreshToken: xxx  # Siehe https://docs.evcc.io/docs/devices/vehicles#tesla
    vin: xxx # Erforderlich, wenn mehrere Fahrzeuge des Herstellers vorhanden sind (optional)
    capacity: xxx # Akkukapazität in kWh (optional)
    control: true # Notwendig für Tesla Wall Connector. Der Virtual Key für evcc muss eingerichtet sein. Siehe https://tesla.evcc.io (optional) 
  - name: Tesla_Model_Y
    type: template
    template: tesla
    title: Tesla Model Y # Wird in der Benutzeroberfläche angezeigt (optional)
    accessToken: xxx  # Siehe https://docs.evcc.io/docs/devices/vehicles#tesla
    refreshToken: xxx  # Siehe https://docs.evcc.io/docs/devices/vehicles#tesla
    vin: xxx # Erforderlich, wenn mehrere Fahrzeuge des Herstellers vorhanden sind (optional)
    capacity: 79 # Akkukapazität in kWh (optional)
    control: true # Notwendig für Tesla Wall Connector. Der Virtual Key für evcc muss eingerichtet sein. Siehe https://tesla.evcc.io (optional)
```

### Log details

```text
[site  ] DEBUG 2024/04/13 13:11:22 ----
[lp-1  ] DEBUG 2024/04/13 13:11:22 charge power: 0W
[lp-2  ] DEBUG 2024/04/13 13:11:22 charge power: 1500W
[site  ] DEBUG 2024/04/13 13:11:22 pv power: 6496W
[site  ] DEBUG 2024/04/13 13:11:22 battery soc: 100%
[site  ] DEBUG 2024/04/13 13:11:22 battery power: -10W
[site  ] DEBUG 2024/04/13 13:11:22 grid meter: -735W
[site  ] DEBUG 2024/04/13 13:11:22 site power: -645W
[lp-2  ] DEBUG 2024/04/13 13:11:22 charge voltages: [239 238 238]V
[lp-2  ] DEBUG 2024/04/13 13:11:22 detected connected phases: 3p
[lp-2  ] DEBUG 2024/04/13 13:11:22 charge currents: [2.1 2.1 2.1]A
[lp-2  ] DEBUG 2024/04/13 13:11:22 detected active phases: 3p
[lp-2  ] DEBUG 2024/04/13 13:11:22 charger status: C
[lp-2  ] DEBUG 2024/04/13 13:11:22 soc estimated: 72.15% (vehicle: 72.00%)
[lp-2  ] DEBUG 2024/04/13 13:11:22 vehicle soc: 72%
[lp-2  ] DEBUG 2024/04/13 13:11:22 vehicle soc limit: 100%
[lp-2  ] DEBUG 2024/04/13 13:11:22 vehicle range: 325km
[lp-2  ] DEBUG 2024/04/13 13:11:22 max charge current: 14A
[lp-2  ] DEBUG 2024/04/13 13:11:22 pv timer elapse
[lp-2  ] DEBUG 2024/04/13 13:11:22 pv timer inactive
[site  ] ERROR 2024/04/13 13:11:23 battery mode: 500 Internal Server Error
[lp-2  ] DEBUG 2024/04/13 13:11:35 set smart cost limit: 0.11
[site  ] DEBUG 2024/04/13 13:11:37 ----
[lp-1  ] DEBUG 2024/04/13 13:11:37 charge power: 0W
[lp-2  ] DEBUG 2024/04/13 13:11:37 charge power: 9960W
[site  ] DEBUG 2024/04/13 13:11:37 pv power: 6587W
[site  ] DEBUG 2024/04/13 13:11:37 battery soc: 100%
[site  ] DEBUG 2024/04/13 13:11:37 battery power: 4530W
[site  ] DEBUG 2024/04/13 13:11:37 grid meter: 3379W
[site  ] DEBUG 2024/04/13 13:11:37 site power: 8009W
[lp-1  ] DEBUG 2024/04/13 13:11:37 charge voltages: [2.6 0 0]V
[lp-1  ] DEBUG 2024/04/13 13:11:37 charge currents: [0.4 0.1 0.5]A
[lp-1  ] DEBUG 2024/04/13 13:11:37 charger status: A
[lp-2  ] DEBUG 2024/04/13 13:11:38 set charge mode: pv
[site  ] DEBUG 2024/04/13 13:11:38 ----
[lp-1  ] DEBUG 2024/04/13 13:11:38 charge power: 0W
[lp-2  ] DEBUG 2024/04/13 13:11:38 charge power: 9988W
[site  ] DEBUG 2024/04/13 13:11:39 pv power: 6591W
[site  ] DEBUG 2024/04/13 13:11:39 battery soc: 100%
[site  ] DEBUG 2024/04/13 13:11:39 battery power: 4540W
[site  ] DEBUG 2024/04/13 13:11:39 grid meter: 3365W
[site  ] DEBUG 2024/04/13 13:11:39 site power: 8005W
[lp-2  ] DEBUG 2024/04/13 13:11:39 charge voltages: [238 235 234]V
[lp-2  ] DEBUG 2024/04/13 13:11:39 detected connected phases: 3p
[lp-2  ] DEBUG 2024/04/13 13:11:39 charge currents: [14.1 14.1 14.2]A
[lp-2  ] DEBUG 2024/04/13 13:11:39 detected active phases: 3p
[lp-2  ] DEBUG 2024/04/13 13:11:39 charger status: C
[lp-2  ] DEBUG 2024/04/13 13:11:39 soc estimated: 72.19% (vehicle: 72.00%)
[lp-2  ] DEBUG 2024/04/13 13:11:39 vehicle soc: 72%
[lp-2  ] DEBUG 2024/04/13 13:11:39 vehicle soc limit: 100%
[lp-2  ] DEBUG 2024/04/13 13:11:39 vehicle range: 325km
[lp-2  ] DEBUG 2024/04/13 13:11:39 pv charge current: 2.4A = 14A + -11.6A (8005W @ 3p)
[lp-2  ] DEBUG 2024/04/13 13:11:39 pv timer reset
[lp-2  ] DEBUG 2024/04/13 13:11:39 pv timer inactive
[lp-2  ] DEBUG 2024/04/13 13:11:39 max charge current: 2A
[lp-2  ] DEBUG 2024/04/13 13:11:42 set charge mode: minpv
[site  ] DEBUG 2024/04/13 13:11:42 ----
[lp-1  ] DEBUG 2024/04/13 13:11:42 charge power: 0W
[lp-2  ] DEBUG 2024/04/13 13:11:42 charge power: 3634W
[site  ] DEBUG 2024/04/13 13:11:42 pv power: 6594W
[site  ] DEBUG 2024/04/13 13:11:42 battery soc: 100%
[site  ] DEBUG 2024/04/13 13:11:42 battery power: 1390W
[site  ] DEBUG 2024/04/13 13:11:42 grid meter: -49W
[site  ] DEBUG 2024/04/13 13:11:42 site power: 1441W
[lp-2  ] DEBUG 2024/04/13 13:11:42 charge voltages: [238 238 237]V
[lp-2  ] DEBUG 2024/04/13 13:11:42 detected connected phases: 3p
[lp-2  ] DEBUG 2024/04/13 13:11:42 charge currents: [5.1 5.1 5.1]A
[lp-2  ] DEBUG 2024/04/13 13:11:42 detected active phases: 3p
[lp-2  ] DEBUG 2024/04/13 13:11:42 charger status: C
[lp-2  ] DEBUG 2024/04/13 13:11:42 soc estimated: 72.20% (vehicle: 72.00%)
[lp-2  ] DEBUG 2024/04/13 13:11:42 vehicle soc: 72%
[lp-2  ] DEBUG 2024/04/13 13:11:42 vehicle soc limit: 100%
[lp-2  ] DEBUG 2024/04/13 13:11:42 vehicle range: 325km
[lp-2  ] WARN 2024/04/13 13:11:42 charger logic error: current mismatch (got 5A, expected 2A)
```

### What type of operating system are you running?

Windows

andig, 2024-04-13:
Bitte trace log. In 125 hat sich nichts geändert.

Dragonkit, 2024-04-14:
Kann ich gerne heute Mittag mal machen wenn der Strom wieder günstig ist wenn mir jemand sagt wie ich das bekomme. Habe EVCC auf dem Home Assistant Green laufen und da bisher nur die Logs gefunden die ich auch oben im Text habe.

So ist das eingestellt.

# log settings
log: info
levels:
  site: debug
  lp-1: debug
  lp-2: debug
  cache: error
  db: error

Dragonkit, 2024-04-14:
Mehr als das zeigt er mir leider nicht an.

[lp-1  ] INFO 2024/04/14 13:03:13 start charging ->
[lp-1  ] DEBUG 2024/04/14 13:03:13 wake-up timer: stop
[lp-1  ] DEBUG 2024/04/14 13:03:13 soc estimated: 96.02% (vehicle: 96.00%)
[lp-1  ] DEBUG 2024/04/14 13:03:13 vehicle soc: 96%
[lp-1  ] DEBUG 2024/04/14 13:03:13 vehicle soc limit: 100%
[lp-1  ] DEBUG 2024/04/14 13:03:13 vehicle range: 411km
[site  ] ERROR 2024/04/14 13:03:14 battery mode: 500 Internal Server Error
[site  ] DEBUG 2024/04/14 13:03:28 ----
[lp-1  ] DEBUG 2024/04/14 13:03:28 charge power: 11440W
[lp-2  ] DEBUG 2024/04/14 13:03:28 charge power: 0W
[site  ] DEBUG 2024/04/14 13:03:28 pv power: 7252W
[site  ] DEBUG 2024/04/14 13:03:28 battery soc: 100%
[site  ] DEBUG 2024/04/14 13:03:28 battery power: 4550W
[site  ] DEBUG 2024/04/14 13:03:28 grid meter: 3408W
[site  ] DEBUG 2024/04/14 13:03:28 site power: 8058W
[lp-2  ] DEBUG 2024/04/14 13:03:28 charge voltages: [2 0 4.6]V
[lp-2  ] DEBUG 2024/04/14 13:03:28 charge currents: [0.2 0 0.2]A
[lp-2  ] DEBUG 2024/04/14 13:03:28 charger status: B
[site  ] ERROR 2024/04/14 13:03:28 battery mode: 500 Internal Server Error
[site  ] DEBUG 2024/04/14 13:03:43 ----
[lp-1  ] DEBUG 2024/04/14 13:03:43 charge power: 11321W
[lp-2  ] DEBUG 2024/04/14 13:03:43 charge power: 0W
[site  ] DEBUG 2024/04/14 13:03:43 pv power: 7143W
[site  ] DEBUG 2024/04/14 13:03:43 battery soc: 100%
[site  ] DEBUG 2024/04/14 13:03:43 battery power: 4550W
[site  ] DEBUG 2024/04/14 13:03:43 grid meter: 3425W
[site  ] DEBUG 2024/04/14 13:03:43 site power: 8075W
[lp-1  ] DEBUG 2024/04/14 13:03:43 charge voltages: [236 235 234]V
[lp-1  ] DEBUG 2024/04/14 13:03:43 detected connected phases: 3p
[lp-1  ] DEBUG 2024/04/14 13:03:43 charge currents: [16 16.1 16]A
[lp-1  ] DEBUG 2024/04/14 13:03:43 detected active phases: 3p
[lp-1  ] DEBUG 2024/04/14 13:03:43 charger status: C
[lp-1  ] DEBUG 2024/04/14 13:03:43 soc estimated: 96.16% (vehicle: 96.00%)
[lp-1  ] DEBUG 2024/04/14 13:03:43 vehicle soc: 96%
[lp-1  ] DEBUG 2024/04/14 13:03:43 vehicle soc limit: 100%
[lp-1  ] DEBUG 2024/04/14 13:03:43 vehicle range: 411km
[site  ] ERROR 2024/04/14 13:03:43 battery mode: 500 Internal Server Error
[lp-1  ] DEBUG 2024/04/14 13:03:56 set charge mode: off
[lp-1  ] DEBUG 2024/04/14 13:03:56 pv timer reset
[lp-1  ] DEBUG 2024/04/14 13:03:56 pv timer inactive
[site  ] DEBUG 2024/04/14 13:03:56 ----
[lp-1  ] DEBUG 2024/04/14 13:03:56 charge power: 11318W
[lp-2  ] DEBUG 2024/04/14 13:03:56 charge power: 0W
[site  ] DEBUG 2024/04/14 13:03:56 pv power: 7365W
[site  ] DEBUG 2024/04/14 13:03:56 battery soc: 100%
[site  ] DEBUG 2024/04/14 13:03:56 battery power: 4540W
[site  ] DEBUG 2024/04/14 13:03:56 grid meter: 3298W
[site  ] DEBUG 2024/04/14 13:03:56 site power: 7938W
[lp-1  ] DEBUG 2024/04/14 13:03:56 charge voltages: [237 235 235]V
[lp-1  ] DEBUG 2024/04/14 13:03:56 detected connected phases: 3p
[lp-1  ] DEBUG 2024/04/14 13:03:56 charge currents: [16 16 16]A
[lp-1  ] DEBUG 2024/04/14 13:03:56 detected active phases: 3p
[lp-1  ] DEBUG 2024/04/14 13:03:56 charger status: C
[lp-1  ] DEBUG 2024/04/14 13:03:56 soc estimated: 96.22% (vehicle: 96.00%)
[lp-1  ] DEBUG 2024/04/14 13:03:56 vehicle soc: 96%
[lp-1  ] DEBUG 2024/04/14 13:03:56 vehicle soc limit: 100%
[lp-1  ] DEBUG 2024/04/14 13:03:56 vehicle range: 411km
[lp-1  ] DEBUG 2024/04/14 13:03:56 charger disable
[lp-1  ] DEBUG 2024/04/14 13:03:56 wake-up timer: stop
[site  ] ERROR 2024/04/14 13:03:57 battery mode: 500 Internal Server Error
[site  ] DEBUG 2024/04/14 13:03:58 ----
[lp-1  ] DEBUG 2024/04/14 13:03:58 charge power: 0W
[lp-2  ] DEBUG 2024/04/14 13:03:58 charge power: 0W
[site  ] DEBUG 2024/04/14 13:03:58 pv power: 7631W
[site  ] DEBUG 2024/04/14 13:03:58 battery soc: 100%
[site  ] DEBUG 2024/04/14 13:03:58 battery power: -10W
[site  ] DEBUG 2024/04/14 13:03:58 grid meter: -4124W
[site  ] DEBUG 2024/04/14 13:03:58 site power: -4034W
[lp-2  ] DEBUG 2024/04/14 13:03:58 charge voltages: [2 0 4.6]V
[lp-2  ] DEBUG 2024/04/14 13:03:58 charge currents: [0.2 0 0.2]A
[lp-2  ] DEBUG 2024/04/14 13:03:58 charger status: B
[site  ] ERROR 2024/04/14 13:03:58 battery mode: 500 Internal Server Error
[site  ] DEBUG 2024/04/14 13:04:13 ----
[lp-1  ] DEBUG 2024/04/14 13:04:13 charge power: 0W
[lp-2  ] DEBUG 2024/04/14 13:04:13 charge power: 0W
[site  ] DEBUG 2024/04/14 13:04:13 pv power: 7496W
[site  ] DEBUG 2024/04/14 13:04:13 battery soc: 100%
[site  ] DEBUG 2024/04/14 13:04:13 battery power: 10W
[site  ] DEBUG 2024/04/14 13:04:13 grid meter: -3888W
[site  ] DEBUG 2024/04/14 13:04:13 site power: -3778W
[lp-1  ] DEBUG 2024/04/14 13:04:13 charge voltages: [2.6 0 0]V
[lp-1  ] DEBUG 2024/04/14 13:04:13 charge currents: [0.4 0.1 0.5]A
[lp-1  ] DEBUG 2024/04/14 13:04:13 charger status: B
[lp-1  ] INFO 2024/04/14 13:04:13 stop charging <-

andig, 2024-04-14:
powerwall: trace

Dragonkit, 2024-04-15:
Reicht das so aus ? Hab es nur kurz gestartet und dann wieder gestoppt. Ich hab ein paar Daten mit ##### ersetzt. 

[powerwall] TRACE 2024/04/15 13:34:37 GET https://owner-api.teslamotors.com/api/1/energy_sites/***/site_status
[powerwall] TRACE 2024/04/15 13:34:38 {"response":{"resource_type":"battery","site_name":"#####","gateway_id":"#####","percentage_charged":100,"battery_type":"ac_powerwall","backup_capable":true,"battery_power":-20,"go_off_grid_test_banner_enabled":null,"storm_mode_enabled":true,"powerwall_onboarding_settings_set":true,"powerwall_tesla_electric_interested_in":null,"vpp_tour_enabled":null,"sync_grid_alert_enabled":true,"breaker_alert_enabled":true}}
[powerwall] TRACE 2024/04/15 13:34:38 POST https://owner-api.teslamotors.com/api/1/energy_sites/***/backup
[powerwall] TRACE 2024/04/15 13:34:38 {"backup_reserve_percent":101}
--
{"response":null,"error":"upstream internal error","error_description":""}
[site  ] ERROR 2024/04/15 13:34:38 battery mode: 500 Internal Server Error
[site  ] DEBUG 2024/04/15 13:35:07 ----
[lp-1  ] DEBUG 2024/04/15 13:35:07 charge power: 0W
[lp-2  ] DEBUG 2024/04/15 13:35:07 charge power: 10102W
[powerwall] TRACE 2024/04/15 13:35:07 GET https://#####/api/meters/aggregates
[powerwall] TRACE 2024/04/15 13:35:07 {"site":{"last_communication_time":"2024-04-15T13:35:07.829967+02:00","instant_power":12,"instant_reactive_power":-4441,"instant_apparent_power":4441.016212535144,"frequency":0,"energy_exported":2067745.810760498,"energy_imported":8187880.138793945,"instant_average_voltage":421.4079614815078,"instant_average_current":22.0675,"i_a_current":0,"i_b_current":0,"i_c_current":0,"last_phase_voltage_communication_time":"0001-01-01T00:00:00Z","last_phase_power_communication_time":"0001-01-01T00:00:00Z","last_phase_energy_communication_time":"0001-01-01T00:00:00Z","timeout":1500000000,"num_meters_aggregated":1,"instant_total_current":22.0675},"battery":{"last_communication_time":"2024-04-15T13:35:07.889614+02:00","instant_power":1690,"instant_reactive_power":-20,"instant_apparent_power":1690.118339052032,"frequency":49.974000000000004,"energy_exported":2975716,"energy_imported":3335263,"instant_average_voltage":244.3,"instant_average_current":-36.7,"i_a_current":0,"i_b_current":0,"i_c_current":0,"last_phase_voltage_communication_time":"0001-01-01T00:00:00Z","last_phase_power_communication_time":"0001-01-01T00:00:00Z","last_phase_energy_communication_time":"0001-01-01T00:00:00Z","timeout":1500000000,"num_meters_aggregated":1,"instant_total_current":-36.7},"load":{"last_communication_time":"2024-04-15T13:35:07.829967+02:00","instant_power":10783.25,"instant_reactive_power":-518.75,"instant_apparent_power":10795.72054681854,"frequency":0,"energy_exported":0,"energy_imported":12794374.778423011,"instant_average_voltage":421.4079614815078,"instant_average_current":25.588624291981226,"i_a_current":0,"i_b_current":0,"i_c_current":0,"last_phase_voltage_communication_time":"0001-01-01T00:00:00Z","last_phase_power_communication_time":"0001-01-01T00:00:00Z","last_phase_energy_communication_time":"0001-01-01T00:00:00Z","timeout":1500000000,"instant_total_current":25.588624291981226},"solar":{"last_communication_time":"2024-04-15T13:35:07.83042+02:00","instant_power":9061,"instant_reactive_power":3961,"instant_apparent_power":9888.945444282723,"frequency":0,"energy_exported":7049121.655024141,"energy_imported":15334.204634577036,"instant_average_voltage":421.42528198958354,"instant_average_current":40.897999999999996,"i_a_current":0,"i_b_current":0,"i_c_current":0,"last_phase_voltage_communication_time":"0001-01-01T00:00:00Z","last_phase_power_communication_time":"0001-01-01T00:00:00Z","last_phase_energy_communication_time":"0001-01-01T00:00:00Z","timeout":1500000000,"num_meters_aggregated":1,"instant_total_current":40.897999999999996}}

andig, 2024-04-15:
Bitte formatieren

Dragonkit, 2024-04-15:
Ich hoffe es passt jetzt so. das kopiert es nicht besser rüber.

andig, 2024-04-16:
Das fällt ja auf:

```
[powerwall] TRACE 2024/04/15 13:34:38 {"backup_reserve_percent":101}
{"response":null,"error":"upstream internal error","error_description":""}
```

Vielleicht verschluckt er sich an den 101. Vmtl. ist der Soc  > 100% und dann kommt es dazu. Ich probiere mal was.

andig, 2024-04-16:
Closed in https://github.com/evcc-io/evcc/commit/07008826482f578ff21cf17d33e3594957cb176a

closed 2024-04-16 by andig