evcc-io/evcc #13423: "Anladen" im PV Modus

andig, 2024-04-14:
Siehe https://github.com/evcc-io/evcc/discussions/11710

andig, 2024-04-17:
Eines statt erkanntes? 👍🏻

VolkerK62, 2024-04-17:
Dann wird zwar bei mehreren Fahrzeugen immer angeladen, auch wenn das angesteckte das gar nicht braucht. Aber das ist dann der "Preis" den man dafür zahlen muss, dass es immer funktioniert.

Wobei ja wohl nicht nur manche Autos, sondern auch manche Wallboxen das brauchen https://github.com/evcc-io/evcc/discussions/11710#discussioncomment-9142051
Also evtl auch noch als feature am charger?
Ist ja wohl nur bei RFID nötig. Wenn die betroffenen Wallboxen auch die entsprechende Einstellung (RFID oder plug&charge) liefern, könnte man das sogar ins Template einbauen.

GrimmiMeloni, 2024-05-09:
Macht es Sinn hier wieder aufzumachen? Der #13534 ist zwar gemerged und zu, aber der letzte commit war ein revert. Effektiv ist dieses Thema also noch ungeklärt, oder?

kvray, 2024-05-10:
Keba P30x zusammen mit BMW iX1 benötigt ebenfalls ein "welcomecharge". Hier liegt es daran, dass die Authetifizierung mit der RFID Karte nach knapp 30 sekunden abläuft, wenn nicht geladen wird. Aktuell löse ich das extern. Die Funktion "welcomecharge" wäre also eine gute Lösung.

roberix, 2024-05-27:
Hallo.

Wir leiden mit allen unsere Keba's unter diesem Verhalten. Das Anladen Feature würde uns sehr helfen.

Vielen Dank

LG

Rob

Oggerschummer, 2024-05-27:
I can confirm the Audi Q4 (Model 2023) has the same behavior - with no solar power available in PV mode the car tries to start loading but after some try and error the sync seems to stop with a technical error (red light). Even if enough solar power becomes available it won't retry on its own. 
Only a short switch to Min+PV and back helps when starting to charge. Charger is a Alfen Single Line Pro, firmware 6.5 with plug&charge enabled. No such issue without evcc on the same charger, works fine at other AC chargers with load balancing.
Impression is that a minimum voltage/power needs to be present when connecting the car (rough guess by a dummy).
What I see in parallel is that it takes evcc quite some time to start charging from PV- could be sort of a timeout as well.
Any parameter that can influence that?

So yes, a _welcomeload_ of a e.g. 10 seconds would potentially solve it when initially starting to load.
No Prio1 for sure, I can handle it. But the WAF within the family is a bit low ;-)

Regards
Thomas

PS: Versehentlich Englisch geantwortet aufgrund "Macht der Gewohnheit"

Magoo2, 2024-05-30:
Ich denke auch der Bedarf ist da, bei Keba gibt es richtig Probleme.
Details auch hier: [#13122]Ich denke auch der Bedarf ist da, bei Keba gibt es richtig Probleme.
Bitte unbedingt das "anladen" Feature implementieren.

TobiasHuber1980, 2024-06-21:
> Hallo.
> 
> Wir leiden mit allen unsere Keba's unter diesem Verhalten. Das Anladen Feature würde uns sehr helfen.
> 
> Vielen Dank
> 
> LG
> 
> Rob

Dito - ich hänge mich mal hier dran....

Hofyyy, 2024-07-12:
Ich klemme mich hier auch mal rein. Das Feature würde auch mein Usecase verbessern, wenn ich mal den Stecker falsch eingesteckt habe und einen Ladeplan für morgen früh drin habe. In dem Fall merke ich nicht das der Stecker final verriegelt hat (BMWi3) und dadurch die Ladung nicht startet.

andig, 2024-07-14:
Blöderweise kann ich nicht mehr feststellen, welchen Fehler die Implementierung beim letzten mal verursacht hatte. Gäbe es jemanden, der das testen würde? Docker kann ich gerne bereit stellen.

andig, 2024-07-14:
PR ist nochmal angepasst und könnte getestet werden.

Hofyyy, 2024-07-14:
ich hab urlaub. das bekomme ich die tage hin. bräuchte einen docker für i5. der rasperry ist raus.

andig, 2024-07-14:
Ich denke wir sollten es NICHT von der Fahreugerkennung abhängig machen sondern _immer_ aktivieren. Der aktuelle Code funktioniert sonst nicht da PV Mode und connect nicht gleichzeitig erfolgen.

Hofyyy, 2024-07-14:
Ok kannst du mir das noch ein wenig erklären? Also ich habe jetzt auf dem Loadpoint als default PV eingestellt. in den autos keinen extra mode bei erkennung. Damit würde er dann wieviele Minuten einfach laden?

Magoo2, 2024-07-14:
> Ok kannst du mir das noch ein wenig erklären? Also ich habe jetzt auf dem Loadpoint als default PV eingestellt. in den autos keinen extra mode bei erkennung. Damit würde er dann wieviele Minuten einfach laden?

20 Sekunden reichen, dann hat die Wallbox das Auto korrekt erkannt und alles läuft wie geplant.

Hofyyy, 2024-07-14:
Ich will garnicht diskutieren was gut ist. Ich möchte gerne wissen wie es sich verhalten soll, bevor ich hier das ausprobieren anfange. Also nach dem anstecken, wird ohne autoerkennung pauschal für 20 sekunden geladen und die autoerkennung darf es auch nicht verändern?

andig, 2024-07-14:
Docker `testing`. Immer einschalten wenn es ein "verfügbares" Fahrzeug mit entsprechendem Feature gibt. Ausschalten im nächsten Zyklus (Sofortladen) oder nach PV Timer (PV Modus).

brand1011, 2024-07-16:
> Ich denke wir sollten es NICHT von der Fahreugerkennung abhängig machen sondern _immer_ aktivieren. Der aktuelle Code funktioniert sonst nicht da PV Mode und connect nicht gleichzeitig erfolgen.
Das ganze grundsätzlich zu aktivieren halte ich für falsch. Mein Auto hat dieses Problem nicht und kommt wunderbar damit klar, dass die box nicht "anlädt". Auch ein anstecken im Modus "off" und umschalten auf "now" am nächsten Tag macht keine Probleme. Oder habe ich Deine Aussage falsch verstanden?

GrimmiMeloni, 2024-07-16:
Sehe ich so wie @brand1011 - es sollte vom konfigurierten Fahrzeugtyp abhängig sein, oder konfigurierbar sein falls es auf die Kombination von Charger / Fahrzeug ankommt.

TobiasHuber1980, 2024-07-17:
In meinen Augen ist es abhängig von der Kombination aus wallbox und Fahrzeug.

VolkerK62, 2024-07-17:
>In meinen Augen ist es abhängig von der Kombination aus wallbox und Fahrzeug.

ist das so?  
Soweit ich es verfolgt habe, gibt es Wallboxen, die das brauchen und es gibt Fahrzeuge, die das brauchen.
Es ist nicht von einer bestimmten Kombination abhängig.

Magoo2, 2024-07-17:
> > In meinen Augen ist es abhängig von der Kombination aus wallbox und Fahrzeug.
> 
> ist das so? Soweit ich es verfolgt habe, gibt es Wallboxen, die das brauchen und es gibt Fahrzeuge, die das brauchen. Es ist nicht von einer bestimmten Kombination abhängig.

Ich bin bei dir, es gibt keine Kombination, sondern es ist von den Einzelkomponenten abhängig.

andig, 2024-07-17:
> Oder habe ich Deine Aussage falsch verstanden?

Zur Klarheit:

The `welcomeCharge` feature will start charging immediately for a short time when a vehicle connects, irrespective of loadpoint mode. This ensures that any vehicle is "enabled" to continue charging at a later time when mode allows. Charging will only start if the connected vehicle- which is generally not known at time it connects (vehicle detection has not yet succeeded) potentially needs power at connection time. This is the case if a) any vehicle with this feature is NOT yet connected or b) the loadpointrequires it. Configuration is done by assigning the `welcomeCharge` feature to vehicles and/or loadpoints.

b) ist noch nicht implementiert.

TobiasHuber1980, 2024-07-17:
> >In meinen Augen ist es abhängig von der Kombination aus wallbox und Fahrzeug.
> 
> 
> 
> ist das so?  
> 
> Soweit ich es verfolgt habe, gibt es Wallboxen, die das brauchen und es gibt Fahrzeuge, die das brauchen.
> 
> Es ist nicht von einer bestimmten Kombination abhängig.

Nein, nicht eine bestimmte! Kombination. Es gibt Fahrzeuge die die Funktion unabhängig von der WB benötigen, jedoch aber auch wallboxen (keba) unabhängig vom Fahrzeug. 
Unterm Strich meinen wir beide das gleiche 😀

Hofyyy, 2024-07-17:
Wenn es Wallboxen und Fahrzeuge gibt wo man es braucht, ist es für den Normalsterblichen nicht leichter das verhalten in die wallbox zu probgrammieren? Also ein Loadpoint verhält sich immer gleich, und würde das nicht auch die Implementierung verreinfachen?

andig, 2024-07-17:
Ich verstehe die Diskussion nicht. Welches Problem soll hier gelöst werden? Können wir das bitte abkürzen- wenn es keine konkreten Änderungsbedarfe gibt bleibt es so.

TobiasHuber1980, 2024-07-17:
@andig 
nun ja es gibt Wallboxen (z.B. Keba), die das anladen-feature benötigen (unabhängig vom Fahrzeug)

und

es gibt Fahrzeuge, die das anladen-feature benötigen (unabhängig von der Wallbox). 

also einfach zwei verschiedene Verursacher mit der gleichen Wirkung.

brand1011, 2024-07-17:
Ich denke das hat @andig verstanden und so wie ich seine Antwort auf meine Frage verstanden habe, soll es einen optionalen Feature Parameter geben, den man entweder dem Fahrzeug oder dem Ladepunkt hinzufügen kann. Damit wären ja beide hier diskutierten Fälle abgedeckt. 

> Configuration is done by assigning the welcomeCharge feature to vehicles and/or loadpoints (use features: [...] similar to coarseCurrent).

Ich denke das ist eine gute Lösung.

Hofyyy, 2024-07-17:
> Ich denke das hat @andig verstanden und so wie ich seine Antwort auf meine Frage verstanden habe, soll es einen optionalen Parameter (welcomeCharge) geben, den man entweder dem Fahrzeug oder dem Ladepunkt hinzufügen kann. Damit wären ja beide hier genannten Fälle abgedeckt.
> 
> > Configuration is done by assigning the welcomeCharge feature to vehicles and/or loadpoints (use features: [...] similar to coarseCurrent).
> 
> Ich denke das ist eine gute Lösung. Man könnte höchstens darüber diskutieren, ob man das Feature, bei Wallboxen die es zwingend benötigen (hier Keba), direkt in deren Wallbox Modul programmiert wie @Hofyyy vorgeschlagen hat.

@brand1011 
Da ich meine Fahrzeuge als Template drin habe und keine Lust habe alles umzustellen. Wie kann ich das denn an den Loadpoint fest tackern? Ich scheine zu blöd ...

TobiasHuber1980, 2024-07-17:
> @brand1011
>
> Da ich meine Fahrzeuge als Template drin habe und keine Lust habe alles umzustellen. Wie kann ich das denn an den Loadpoint fest tackern? Ich scheine zu blöd ...

Loadpoint: siehe oben:
b) ist noch nicht implementiert.

andig, 2024-07-20:
Wissen wir denn, welche Fahrzeuge und Wallboxen betroffen sind? Ich kenne aktuell KEBA (alle? Modbus und UDP?) und BMW. Wenn es das wäre würde es genügen, das zu diesen beiden Templates hinzu zu fügen. Es wäre keine generelle Option am Ladepunkt notwendig.

roberix, 2024-07-20:
Hi. Bestätige Modbus und UDP. 

LG

rob

andig, 2024-07-20:
Da das UDP Template deprecated ist hab ich's nur bei Modbus umgesetzt- Test wäre Klasse.

VolkerK62, 2024-07-20:
Da sind wohl noch ein paar mehr Fahrzeuge
https://github.com/evcc-io/evcc/discussions/11710#discussioncomment-9147180

brand1011, 2024-07-20:
Hab einen BMW i4 und der ist von dem Problem nicht betroffen.

machtandrebel, 2024-07-21:
Der smart #1 pennt auch ein, wenn er nach nach dem Einstecken keinen Saft bekommt und wacht dann leider bei PVÜ nicht auf.

Magoo2, 2024-07-21:
Mercedes C Klasse ist auch von den Phänomen betroffen.

TobiasHuber1980, 2024-07-21:
https://github.com/evcc-io/evcc/discussions/11710#discussioncomment-8737679

VW IDs auch?

Tranquility76, 2024-07-21:
> https://github.com/evcc-io/evcc/discussions/11710#discussioncomment-8737679
> 
> VW IDs auch? 

Konnte ich bei einem ID.4 bisher nicht feststellen. Der legt brav immer wieder los, wenn EVCC die Wallbox freigibt.

closed 2024-07-21 by andig

svenr2015, 2024-07-22:
> https://github.com/evcc-io/evcc/discussions/11710#discussioncomment-8737679
> 
> VW IDs auch? 

Beim e-Golf scheint dies zuzutreffen. Wäre also auch bei VW gut zu ergänzen. Vielen Dank!

Hofyyy, 2024-07-23:
Meine subjektive Meinung ist, ich finde es sogar ein tolles "standard" Feature und verstehe nicht warum wir es nicht einfach generell zugänglich machen. Es fängt sogar den Fall eines nicht richtig verriegelten Ladekabels, welches ich in meinem BMWi3 schon 3x hatte. Was zur folge hatte, das morgens der Ladeplan nicht durchgeführt werden konnte. in meinem Fall würde Ich beide Wallboxen so einstellen, das Sie das neue Feature nutzen, unabhängig avon welche Autos im laufe der Zeit so vorbei kommen, oder per PV oder getimed geladen werden.

premultiply, 2024-07-23:
Warum? Ganz einfach: Weil sich dieses "Feature" über sämtliche Restriktionen des Lastmanagements hinwegsetzt und damit im Zweifelsfall Lasten (und Kosten, Stichwort Spitzenlasttarif) erzeugt die es nicht erzeugen dürfte. Das ist insbesondere in Ländern ausserhalb von Deutschland teilweise ein ersthaftes Problem.

e-fahrer-jochen, 2024-07-23:
Der Smart EQ braucht das kurze Anladen auch.

machtandrebel, 2024-08-11:
Gibt es für das "welcomecharge feature" auch eine Doku, damit man das einrichten kann?

[image]

andig, 2024-08-11:
Noch nicht: https://github.com/evcc-io/docs/issues/560