evcc-io/evcc #13429: enphase "authentication required" error

ronajon, 2024-04-14:
### Describe the bug

i'm experiencing issues with my Enphase Envoy IQ.
i've added the enphase as grid and pv, but the error i got is "Authorization Required"
the token is extracted from the enphase site as suggested, and working. i can login on the envoy-ip with the token

`meters:
  - name: my_grid
    type: template
    template: enphase
    usage: grid
    host: 192.168.110.80
    token: <token>
`

`
[http  ] TRACE 2024/04/14 16:35:41 GET http://192.168.110.80/production.json
[http  ] TRACE 2024/04/14 16:35:41 <html>
<head><title>301 Moved Permanently</title></head>
<body>
<center><h1>301 Moved Permanently</h1></center>
<hr><center>openresty/1.17.8.1</center>
</body>
</html>
[http  ] TRACE 2024/04/14 16:35:41 GET https://192.168.110.80/production.json
[http  ] TRACE 2024/04/14 16:35:42 <html>
<head><title>401 Authorization Required</title></head>
<body>
<center><h1>401 Authorization Required</h1></center>
<hr><center>Redirecting to <a id='link' href=''></a></center>
<script>
const redirect_url = window.location.protocol + '//' + window.location.host + '/home';
function redirect() { window.location.href = redirect_url; }
window.onload = function() {
    const link = document.getElementById('link');
    link.href = redirect_url;
    link.text = redirect_url;
    setTimeout(redirect, 500);
}
</script>
</body>
</html>
`

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

sponsortoken: <..token..>

log: trace
#levels:
#cache: error

# unique installation id
plant: ********

interval: 30s # control cycle interval

meters:
  - name: my_grid
    type: template
    template: enphase
    usage: grid
    host: 192.168.110.80
    token: <..token..>

chargers:
  - name: alfen
    type: template
    template: alfen
    # Modbus TCP
    modbus: tcpip
    id: 1
    host: 192.168.110.81 # Hostname
    port: 502 # Port 

vehicles:
  - name: tesla
    type: template
    template: tesla
    title: # Will be displayed in the user interface (optional)
    accessToken: <..token..>
    refreshToken: <..token..>
    capacity: 60 # Battery capacity in kWh (optional) 

site:
  title: Home
  meters:
    grid: my_grid

loadpoints:
  - title: wallcharger
    charger: alfen
    mode: off

tariffs:
  grid:
    # either
    type: tibber
    token: <..token..> # access token
    homeid: <..token..>
```

### Log details

```text
[http  ] TRACE 2024/04/14 16:47:14 GET http://192.168.110.80/production.json
[http  ] TRACE 2024/04/14 16:47:14 <html>
<head><title>301 Moved Permanently</title></head>
<body>
<center><h1>301 Moved Permanently</h1></center>
<hr><center>openresty/1.17.8.1</center>
</body>
</html>
[http  ] TRACE 2024/04/14 16:47:14 GET https://192.168.110.80/production.json
[http  ] TRACE 2024/04/14 16:47:14 <html>
<head><title>401 Authorization Required</title></head>
<body>
<center><h1>401 Authorization Required</h1></center>
<hr><center>Redirecting to <a id='link' href=''></a></center>
<script>
const redirect_url = window.location.protocol + '//' + window.location.host + '/home';
function redirect() { window.location.href = redirect_url; }
window.onload = function() {
    const link = document.getElementById('link');
    link.href = redirect_url;
    link.text = redirect_url;
    setTimeout(redirect, 500);
}
</script>
</body>
</html>
```

### What type of operating system are you running?

Linux

### Version

0.125.0

TobiasNeuschulz, 2024-04-16:
The Meters configuration for the grid looks fine for me. I'm missing the settings for pv and maybe battery.
Which Firmware are you using on your enphase system?

I have an enphase system running with evcc version 0.125.0

You can try my documentation:
https://github.com/TobiasNeuschulz/EVCC-Enphase/blob/main/Enphase-IQ-Battery.md

ronajon, 2024-04-17:
i deliberate left out the configuration for pv. i don't have a battery.

my evcc version is 0.125.0
enphase firmware i  D7.0.88 (5580b1) 

i've looked at your configuration, but i see nothing i'm missing or have configured different in mine. 

could it be possible to access the enphase grid info with a custom approach ?

TobiasNeuschulz, 2024-04-17:
I have Firmware D7.6.323
I think your firmware is around 2 years old. Maybe an update helps.

ronajon, 2024-04-17:
i tried it through a custom template, unfortunately that did not work either.
i'll contact enphase for a firmware upgrade

closed 2024-04-17 by andig