evcc-io/evcc #13430: Tesla Fleet API Token läuft immer noch ab

Raudi1, 2024-04-14:
### Describe the bug

Bei mir gibt es immer noch Probleme mit den Tesla Fleet API Token. Bei mir ist jetzt schon zum 2. Mal das Token nicht mehr gültig.

### Steps to reproduce

1. Token und Refresh Token auf tesla.evcc.io erzeugen und in die Config eintragen
2. Ein paar Tage (glaube ungefähr eine Woche) funktioniert alles
3. Token läuft ab

### Configuration details

```yaml
vehicles:
  - name: wario
    type: template
    template: tesla
    title: Wario
    accessToken: XXXXXXXXXXXXXXXXXXXXXXXXXXXX
    refreshToken: EU_XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
    vin: LRWXXXXXXXXXXX
    capacity: 60
    mincurrent: 1
    maxcurrent: 16
    control: true
```

### Log details

```text
using config file: /config/evcc.yaml
starting evcc: 'EVCC_DATABASE_DSN=/data/evcc.db evcc --config /config/evcc.yaml'
[main  ] INFO 2024/04/14 16:27:48 evcc 0.125.0
[main  ] INFO 2024/04/14 16:27:48 using config file: /config/evcc.yaml
[main  ] ERROR 2024/04/14 16:27:48 creating vehicle wario failed: cannot create vehicle type 'template': cannot create vehicle type 'tesla': token expired
```

### What type of operating system are you running?

HomeAssistant Add-on

### Version

0.125.0

andig, 2024-04-14:
Was soll ich sagen. Es braucht ein Log von Tesla und DB über den ganzen Vorgang

closed 2024-04-14 by andig

simonhard, 2024-04-14:
@andig , es scheint, als würde der accessToken nicht refresht. Ein frischer accessToken funktioniert. Er ist aber ja nur 8h gültig, danach bekomme ich den selben Fehler.
Wie speichert evcc einen neuen accessToken nach refresh? Beschreibt er die evcc.yaml? ggf. liegt es dann an mangelnden Schreibrechten in einem Docker-Volume.

siehe auch hier:
https://github.com/evcc-io/evcc/discussions/13190

Raudi1, 2024-04-14:
> Was soll ich sagen. Es braucht ein Log von Tesla und DB über den ganzen Vorgang. Level trace.

Werde ich aktivieren und das nächste Mal wenn es auftaucht dann nachreichen. Ich kann es aber leider nicht ad-hoc reproduzieren.

Stacykid, 2024-04-17:
Das mit dem abgelaufenen Token habe ich auch, zumindest ist zum 2. mal der Tesla nur als Gastfahrzeug erkannt worden.
Nach einem Refresh der Token geht alles wieder. Bei mir liegen da jeweils ca. 4 Wochen zwischen.