evcc-io/evcc #13435: Dynamischer Strompreis per MQTT

ich77, 2024-04-15:
Eröffne nach zwei Diskussionen ([13400](https://github.com/evcc-io/evcc/discussions/13400) / [13421](https://github.com/evcc-io/evcc/discussions/13421)) doch noch einmal einen Feature Request. Bezugnehmend auf den geschlossenen Request [13399](https://github.com/evcc-io/evcc/issues/13399)

> Wir haben von der [WSW](https://www.wsw-online.de/energie/) einen dynamischen Stromtarif, bei dem der Preis täglich um 15Uhr über 
> (einen MQTT Broker übertragen wird. Wäre es möglich, den auch in EVCC einzubinden? Weitere Infos gibt es [hier](https://talmarkt.wsw-online.de/developer)
> 
> Oder gibt es einen anderen Weg, die Preise in EVCC bekannt zu machen?
> 
> Vielen Dank

Der MQTT Stream liefert einen Preis, der sich jede Stunde verändert. Aktuell liefert der MQTT folgende Daten:
`[{"from":"2024-04-13T22:00:00Z","to":"2024-04-13T23:00:00Z","value":23.618,"mainUnit":"CT","refUnit":"KWH","status":"FINAL"},{"from":"2024-04-13T23:00:00Z","to":"2024-04-14T00:00:00Z","value":23.666,"mainUnit":"CT","refUnit":"KWH","status":"FINAL"},{"from":"2024-04-14T00:00:00Z","to":"2024-04-14T01:00:00Z","value":23.677,"mainUnit":"CT","refUnit":"KWH","status":"FINAL"},{"from":"2024-04-14T01:00:00Z","to":"2024-04-14T02:00:00Z","value":23.701,"mainUnit":"CT","refUnit":"KWH","status":"FINAL"},{"from":"2024-04-14T02:00:00Z","to":"2024-04-14T03:00:00Z","value":23.845,"mainUnit":"CT","refUnit":"KWH","status":"FINAL"},{"from":"2024-04-14T03:00:00Z","to":"2024-04-14T04:00:00Z","value":24.275,"mainUnit":"CT","refUnit":"KWH","status":"FINAL"},{"from":"2024-04-14T04:00:00Z","to":"2024-04-14T05:00:00Z","value":25.097,"mainUnit":"CT","refUnit":"KWH","status":"FINAL"},{"from":"2024-04-14T05:00:00Z","to":"2024-04-14T06:00:00Z","value":25.114,"mainUnit":"CT","refUnit":"KWH","status":"FINAL"},{"from":"2024-04-14T06:00:00Z","to":"2024-04-14T07:00:00Z","value":24.449,"mainUnit":"CT","refUnit":"KWH","status":"FINAL"},{"from":"2024-04-14T07:00:00Z","to":"2024-04-14T08:00:00Z","value":23.567,"mainUnit":"CT","refUnit":"KWH","status":"FINAL"},{"from":"2024-04-14T08:00:00Z","to":"2024-04-14T09:00:00Z","value":23.285,"mainUnit":"CT","refUnit":"KWH","status":"FINAL"},{"from":"2024-04-14T09:00:00Z","to":"2024-04-14T10:00:00Z","value":22.119,"mainUnit":"CT","refUnit":"KWH","status":"FINAL"},{"from":"2024-04-14T10:00:00Z","to":"2024-04-14T11:00:00Z","value":18.537,"mainUnit":"CT","refUnit":"KWH","status":"FINAL"},{"from":"2024-04-14T11:00:00Z","to":"2024-04-14T12:00:00Z","value":16.421,"mainUnit":"CT","refUnit":"KWH","status":"FINAL"},{"from":"2024-04-14T12:00:00Z","to":"2024-04-14T13:00:00Z","value":16.418,"mainUnit":"CT","refUnit":"KWH","status":"FINAL"},{"from":"2024-04-14T13:00:00Z","to":"2024-04-14T14:00:00Z","value":18.801,"mainUnit":"CT","refUnit":"KWH","status":"FINAL"},{"from":"2024-04-14T14:00:00Z","to":"2024-04-14T15:00:00Z","value":22.823,"mainUnit":"CT","refUnit":"KWH","status":"FINAL"},{"from":"2024-04-14T15:00:00Z","to":"2024-04-14T16:00:00Z","value":23.754,"mainUnit":"CT","refUnit":"KWH","status":"FINAL"},{"from":"2024-04-14T16:00:00Z","to":"2024-04-14T17:00:00Z","value":31.567,"mainUnit":"CT","refUnit":"KWH","status":"FINAL"},{"from":"2024-04-14T17:00:00Z","to":"2024-04-14T18:00:00Z","value":34.015,"mainUnit":"CT","refUnit":"KWH","status":"FINAL"},{"from":"2024-04-14T18:00:00Z","to":"2024-04-14T19:00:00Z","value":37.836,"mainUnit":"CT","refUnit":"KWH","status":"FINAL"},{"from":"2024-04-14T19:00:00Z","to":"2024-04-14T20:00:00Z","value":36.967,"mainUnit":"CT","refUnit":"KWH","status":"FINAL"},{"from":"2024-04-14T20:00:00Z","to":"2024-04-14T21:00:00Z","value":35.937,"mainUnit":"CT","refUnit":"KWH","status":"FINAL"},{"from":"2024-04-14T21:00:00Z","to":"2024-04-14T22:00:00Z","value":33.948,"mainUnit":"CT","refUnit":"KWH","status":"FINAL"},{"from":"2024-04-14T22:00:00Z","to":"2024-04-14T23:00:00Z","value":33.320,"mainUnit":"CT","refUnit":"KWH","status":"FINAL"},{"from":"2024-04-14T23:00:00Z","to":"2024-04-15T00:00:00Z","value":33.237,"mainUnit":"CT","refUnit":"KWH","status":"FINAL"},{"from":"2024-04-15T00:00:00Z","to":"2024-04-15T01:00:00Z","value":32.794,"mainUnit":"CT","refUnit":"KWH","status":"FINAL"},{"from":"2024-04-15T01:00:00Z","to":"2024-04-15T02:00:00Z","value":32.870,"mainUnit":"CT","refUnit":"KWH","status":"FINAL"},{"from":"2024-04-15T02:00:00Z","to":"2024-04-15T03:00:00Z","value":32.519,"mainUnit":"CT","refUnit":"KWH","status":"FINAL"},{"from":"2024-04-15T03:00:00Z","to":"2024-04-15T04:00:00Z","value":33.889,"mainUnit":"CT","refUnit":"KWH","status":"FINAL"},{"from":"2024-04-15T04:00:00Z","to":"2024-04-15T05:00:00Z","value":37.339,"mainUnit":"CT","refUnit":"KWH","status":"FINAL"},{"from":"2024-04-15T05:00:00Z","to":"2024-04-15T06:00:00Z","value":43.696,"mainUnit":"CT","refUnit":"KWH","status":"FINAL"},{"from":"2024-04-15T06:00:00Z","to":"2024-04-15T07:00:00Z","value":41.802,"mainUnit":"CT","refUnit":"KWH","status":"FINAL"},{"from":"2024-04-15T07:00:00Z","to":"2024-04-15T08:00:00Z","value":34.462,"mainUnit":"CT","refUnit":"KWH","status":"FINAL"},{"from":"2024-04-15T08:00:00Z","to":"2024-04-15T09:00:00Z","value":33.152,"mainUnit":"CT","refUnit":"KWH","status":"FINAL"},{"from":"2024-04-15T09:00:00Z","to":"2024-04-15T10:00:00Z","value":31.082,"mainUnit":"CT","refUnit":"KWH","status":"FINAL"},{"from":"2024-04-15T10:00:00Z","to":"2024-04-15T11:00:00Z","value":28.821,"mainUnit":"CT","refUnit":"KWH","status":"FINAL"},{"from":"2024-04-15T11:00:00Z","to":"2024-04-15T12:00:00Z","value":27.344,"mainUnit":"CT","refUnit":"KWH","status":"FINAL"},{"from":"2024-04-15T12:00:00Z","to":"2024-04-15T13:00:00Z","value":25.109,"mainUnit":"CT","refUnit":"KWH","status":"FINAL"},{"from":"2024-04-15T13:00:00Z","to":"2024-04-15T14:00:00Z","value":24.495,"mainUnit":"CT","refUnit":"KWH","status":"FINAL"},{"from":"2024-04-15T14:00:00Z","to":"2024-04-15T15:00:00Z","value":24.518,"mainUnit":"CT","refUnit":"KWH","status":"FINAL"},{"from":"2024-04-15T15:00:00Z","to":"2024-04-15T16:00:00Z","value":29.076,"mainUnit":"CT","refUnit":"KWH","status":"FINAL"},{"from":"2024-04-15T16:00:00Z","to":"2024-04-15T17:00:00Z","value":31.802,"mainUnit":"CT","refUnit":"KWH","status":"FINAL"},{"from":"2024-04-15T17:00:00Z","to":"2024-04-15T18:00:00Z","value":32.958,"mainUnit":"CT","refUnit":"KWH","status":"FINAL"},{"from":"2024-04-15T18:00:00Z","to":"2024-04-15T19:00:00Z","value":32.732,"mainUnit":"CT","refUnit":"KWH","status":"FINAL"},{"from":"2024-04-15T19:00:00Z","to":"2024-04-15T20:00:00Z","value":31.458,"mainUnit":"CT","refUnit":"KWH","status":"FINAL"},{"from":"2024-04-15T20:00:00Z","to":"2024-04-15T21:00:00Z","value":29.965,"mainUnit":"CT","refUnit":"KWH","status":"FINAL"},{"from":"2024-04-15T21:00:00Z","to":"2024-04-15T22:00:00Z","value":27.633,"mainUnit":"CT","refUnit":"KWH","status":"FINAL"},{"from":"2024-04-15T22:00:00Z","to":"2024-04-15T23:00:00Z","value":27.108,"mainUnit":"CT","refUnit":"KWH","status":"FORECAST"},{"from":"2024-04-15T23:00:00Z","to":"2024-04-16T00:00:00Z","value":26.830,"mainUnit":"CT","refUnit":"KWH","status":"FORECAST"},{"from":"2024-04-16T00:00:00Z","to":"2024-04-16T01:00:00Z","value":26.350,"mainUnit":"CT","refUnit":"KWH","status":"FORECAST"},{"from":"2024-04-16T01:00:00Z","to":"2024-04-16T02:00:00Z","value":25.039,"mainUnit":"CT","refUnit":"KWH","status":"FORECAST"},{"from":"2024-04-16T02:00:00Z","to":"2024-04-16T03:00:00Z","value":27.036,"mainUnit":"CT","refUnit":"KWH","status":"FORECAST"},{"from":"2024-04-16T03:00:00Z","to":"2024-04-16T04:00:00Z","value":28.249,"mainUnit":"CT","refUnit":"KWH","status":"FORECAST"},{"from":"2024-04-16T04:00:00Z","to":"2024-04-16T05:00:00Z","value":30.118,"mainUnit":"CT","refUnit":"KWH","status":"FORECAST"},{"from":"2024-04-16T05:00:00Z","to":"2024-04-16T06:00:00Z","value":31.015,"mainUnit":"CT","refUnit":"KWH","status":"FORECAST"},{"from":"2024-04-16T06:00:00Z","to":"2024-04-16T07:00:00Z","value":30.846,"mainUnit":"CT","refUnit":"KWH","status":"FORECAST"},{"from":"2024-04-16T07:00:00Z","to":"2024-04-16T08:00:00Z","value":29.667,"mainUnit":"CT","refUnit":"KWH","status":"FORECAST"},{"from":"2024-04-16T08:00:00Z","to":"2024-04-16T09:00:00Z","value":26.905,"mainUnit":"CT","refUnit":"KWH","status":"FORECAST"},{"from":"2024-04-16T09:00:00Z","to":"2024-04-16T10:00:00Z","value":24.835,"mainUnit":"CT","refUnit":"KWH","status":"FORECAST"},{"from":"2024-04-16T10:00:00Z","to":"2024-04-16T11:00:00Z","value":25.091,"mainUnit":"CT","refUnit":"KWH","status":"FORECAST"},{"from":"2024-04-16T11:00:00Z","to":"2024-04-16T12:00:00Z","value":24.519,"mainUnit":"CT","refUnit":"KWH","statu [... cut ...] T","refUnit":"KWH","status":"FORECAST"},{"from":"2024-04-16T19:00:00Z","to":"2024-04-16T20:00:00Z","value":31.498,"mainUnit":"CT","refUnit":"KWH","status":"FORECAST"},{"from":"2024-04-16T20:00:00Z","to":"2024-04-16T21:00:00Z","value":31.098,"mainUnit":"CT","refUnit":"KWH","status":"FORECAST"},{"from":"2024-04-16T21:00:00Z","to":"2024-04-16T22:00:00Z","value":30.732,"mainUnit":"CT","refUnit":"KWH","status":"FORECAST"}]`

Würde mich sehr freuen, wenn es dafür eine Lösung gibt.

Vielen Dank

Markus

andig, 2024-04-15:
Das Thema gabs schon x-fach. Bisher hatte niemand eine Lösung wie das generisch aussehen soll. Solange es dafür keinen Vorschlag gibt WONTFIX.

mdkeil, 2024-04-15:
Ich glaube, ihm geht es primär um die Implementierung der Abfrage der Preise vom seinem Stromanbieter WSW; Das Feature Request wurde unglücklich bezeichnet. :)

andig, 2024-04-15:
Wir haben ja schon einen `custom` Tarif der genau einen Wert dynamisch liefern kann. Hier wäre die Idee ja einen Forecast zu erzeugen, also X Werte. Dafür gabs bisher keine gute Idee.

Eine hätte ich: man könnte probieren, per JQ aus z.B. einem HTTP Request wie oben eine Tabelle von `api.Rates` zusammen zu bauen, also i.W. Price, From, To. Das kann gerne mal jemand probieren. Wenn es gelingt (JQ Query) dann würde ich für den Unterbau in evcc sorgen.

mdkeil, 2024-04-15:
klingt zumindest nach einem Anfang.. denn sicher werden spätestens ab 2025 eine Vielzahl von Stromanbietern dynamische Tarife anbieten (müssen).. bis dahin sollte es dann eine praktikable Schnittstelle geben.

andig, 2024-04-15:
Da wir jede Stunde neue Werte holen ist das Wurst.

mdkeil, 2024-04-15:
das via jq zu matchen ist noch relativ trivial--

`jq -r (.[] | [.from, .to, .value]) | @tsv`

```
2024-04-13T22:00:00Z	2024-04-13T23:00:00Z	23.618
2024-04-13T23:00:00Z	2024-04-14T00:00:00Z	23.666
2024-04-14T00:00:00Z	2024-04-14T01:00:00Z	23.677
2024-04-14T01:00:00Z	2024-04-14T02:00:00Z	23.701
2024-04-14T02:00:00Z	2024-04-14T03:00:00Z	23.845
2024-04-14T03:00:00Z	2024-04-14T04:00:00Z	24.275
2024-04-14T04:00:00Z	2024-04-14T05:00:00Z	25.097
2024-04-14T05:00:00Z	2024-04-14T06:00:00Z	25.114
2024-04-14T06:00:00Z	2024-04-14T07:00:00Z	24.449
2024-04-14T07:00:00Z	2024-04-14T08:00:00Z	23.567
2024-04-14T08:00:00Z	2024-04-14T09:00:00Z	23.285
2024-04-14T09:00:00Z	2024-04-14T10:00:00Z	22.119
2024-04-14T10:00:00Z	2024-04-14T11:00:00Z	18.537
2024-04-14T11:00:00Z	2024-04-14T12:00:00Z	16.421
2024-04-14T12:00:00Z	2024-04-14T13:00:00Z	16.418
2024-04-14T13:00:00Z	2024-04-14T14:00:00Z	18.801
2024-04-14T14:00:00Z	2024-04-14T15:00:00Z	22.823
2024-04-14T15:00:00Z	2024-04-14T16:00:00Z	23.754
2024-04-14T16:00:00Z	2024-04-14T17:00:00Z	31.567
2024-04-14T17:00:00Z	2024-04-14T18:00:00Z	34.015
2024-04-14T18:00:00Z	2024-04-14T19:00:00Z	37.836
2024-04-14T19:00:00Z	2024-04-14T20:00:00Z	36.967
2024-04-14T20:00:00Z	2024-04-14T21:00:00Z	35.937
2024-04-14T21:00:00Z	2024-04-14T22:00:00Z	33.948
2024-04-14T22:00:00Z	2024-04-14T23:00:00Z	33.32
2024-04-14T23:00:00Z	2024-04-15T00:00:00Z	33.237
2024-04-15T00:00:00Z	2024-04-15T01:00:00Z	32.794
2024-04-15T01:00:00Z	2024-04-15T02:00:00Z	32.87
2024-04-15T02:00:00Z	2024-04-15T03:00:00Z	32.519
2024-04-15T03:00:00Z	2024-04-15T04:00:00Z	33.889
2024-04-15T04:00:00Z	2024-04-15T05:00:00Z	37.339
2024-04-15T05:00:00Z	2024-04-15T06:00:00Z	43.696
2024-04-15T06:00:00Z	2024-04-15T07:00:00Z	41.802
2024-04-15T07:00:00Z	2024-04-15T08:00:00Z	34.462
2024-04-15T08:00:00Z	2024-04-15T09:00:00Z	33.152
2024-04-15T09:00:00Z	2024-04-15T10:00:00Z	31.082
2024-04-15T10:00:00Z	2024-04-15T11:00:00Z	28.821
2024-04-15T11:00:00Z	2024-04-15T12:00:00Z	27.344
2024-04-15T12:00:00Z	2024-04-15T13:00:00Z	25.109
2024-04-15T13:00:00Z	2024-04-15T14:00:00Z	24.495
2024-04-15T14:00:00Z	2024-04-15T15:00:00Z	24.518
2024-04-15T15:00:00Z	2024-04-15T16:00:00Z	29.076
2024-04-15T16:00:00Z	2024-04-15T17:00:00Z	31.802
2024-04-15T17:00:00Z	2024-04-15T18:00:00Z	32.958
2024-04-15T18:00:00Z	2024-04-15T19:00:00Z	32.732
2024-04-15T19:00:00Z	2024-04-15T20:00:00Z	31.458
2024-04-15T20:00:00Z	2024-04-15T21:00:00Z	29.965
2024-04-15T21:00:00Z	2024-04-15T22:00:00Z	27.633
2024-04-15T22:00:00Z	2024-04-15T23:00:00Z	27.108
2024-04-15T23:00:00Z	2024-04-16T00:00:00Z	26.83
2024-04-16T00:00:00Z	2024-04-16T01:00:00Z	26.35
2024-04-16T01:00:00Z	2024-04-16T02:00:00Z	25.039
2024-04-16T02:00:00Z	2024-04-16T03:00:00Z	27.036
2024-04-16T03:00:00Z	2024-04-16T04:00:00Z	28.249
2024-04-16T04:00:00Z	2024-04-16T05:00:00Z	30.118
2024-04-16T05:00:00Z	2024-04-16T06:00:00Z	31.015
2024-04-16T06:00:00Z	2024-04-16T07:00:00Z	30.846
2024-04-16T07:00:00Z	2024-04-16T08:00:00Z	29.667
2024-04-16T08:00:00Z	2024-04-16T09:00:00Z	26.905
2024-04-16T09:00:00Z	2024-04-16T10:00:00Z	24.835
2024-04-16T10:00:00Z	2024-04-16T11:00:00Z	25.091
2024-04-16T11:00:00Z	2024-04-16T12:00:00Z	24.519
2024-04-16T12:00:00Z	2024-04-16T13:00:00Z	24.115
2024-04-16T13:00:00Z	2024-04-16T14:00:00Z	24.218
2024-04-16T14:00:00Z	2024-04-16T15:00:00Z	26.569
2024-04-16T15:00:00Z	2024-04-16T16:00:00Z	28.697
2024-04-16T16:00:00Z	2024-04-16T17:00:00Z	30.982
2024-04-16T17:00:00Z	2024-04-16T18:00:00Z	35.276
2024-04-16T18:00:00Z	2024-04-16T19:00:00Z	35.205
2024-04-16T19:00:00Z	2024-04-16T20:00:00Z	31.498
2024-04-16T20:00:00Z	2024-04-16T21:00:00Z	31.098
2024-04-16T21:00:00Z	2024-04-16T22:00:00Z	30.732
```
wie man das in ein entsprechendes go-object/array/table piped, da habe ich keine Ahnung ;)

andig, 2024-04-15:
Es braucht json mit den erwähnten Feldern.

mdkeil, 2024-04-15:
`[.[] | {from: .from, to: .to, price: .value}]` ?

```
[
  {
    "from": "2024-04-13T22:00:00Z",
    "to": "2024-04-13T23:00:00Z",
    "price": 23.618
  },
  {
    "from": "2024-04-13T23:00:00Z",
    "to": "2024-04-14T00:00:00Z",
    "price": 23.666
  },
  {
    "from": "2024-04-14T00:00:00Z",
    "to": "2024-04-14T01:00:00Z",
    "price": 23.677
  },
  {
    "from": "2024-04-14T01:00:00Z",
    "to": "2024-04-14T02:00:00Z",
    "price": 23.701
  },
  {
    "from": "2024-04-14T02:00:00Z",
    "to": "2024-04-14T03:00:00Z",
    "price": 23.845
  },
  {
    "from": "2024-04-14T03:00:00Z",
    "to": "2024-04-14T04:00:00Z",
    "price": 24.275
  },
  {
    "from": "2024-04-14T04:00:00Z",
    "to": "2024-04-14T05:00:00Z",
    "price": 25.097
  },
  {
    "from": "2024-04-14T05:00:00Z",
    "to": "2024-04-14T06:00:00Z",
    "price": 25.114
  },
  {
    "from": "2024-04-14T06:00:00Z",
    "to": "2024-04-14T07:00:00Z",
    "price": 24.449
  },
  {
    "from": "2024-04-14T07:00:00Z",
    "to": "2024-04-14T08:00:00Z",
    "price": 23.567
  },
  {
    "from": "2024-04-14T08:00:00Z",
    "to": "2024-04-14T09:00:00Z",
    "price": 23.285
  },
  {
    "from": "2024-04-14T09:00:00Z",
    "to": "2024-04-14T10:00:00Z",
    "price": 22.119
  },
  {
    "from": "2024-04-14T10:00:00Z",
    "to": "2024-04-14T11:00:00Z",
    "price": 18.537
  },
  {
[... 271 lines cut ...]
    "to": "2024-04-16T18:00:00Z",
    "price": 35.276
  },
  {
    "from": "2024-04-16T18:00:00Z",
    "to": "2024-04-16T19:00:00Z",
    "price": 35.205
  },
  {
    "from": "2024-04-16T19:00:00Z",
    "to": "2024-04-16T20:00:00Z",
    "price": 31.498
  },
  {
    "from": "2024-04-16T20:00:00Z",
    "to": "2024-04-16T21:00:00Z",
    "price": 31.098
  },
  {
    "from": "2024-04-16T21:00:00Z",
    "to": "2024-04-16T22:00:00Z",
    "price": 30.732
  }
]
```

mdkeil, 2024-04-15:
Die Frage ist, ob man nicht z.B. den Anteil "to" weglässt da es eigentlich nicht notwendig ist.. tibber z.B. liefert ja auch nur  "startsAt".. bzgl. des Zeitstempels-- die Konvertierung passiert im Backend?-- dieser Anbieter liefert die Zeiten in UTC, Tibber z.B. in local.

andig, 2024-04-15:
Es braucht rfc3339 für Time. Im Zweifel mittels jq herstellen. To können wir optional machen. Falls nicht vorhanden wird es auf den Folge-TS gesetzt. Das letzte Intervall würde dann nicht genutzt.

andig, 2024-04-15:
Kann man diese API zu Testzwecken abfragen?

andig, 2024-04-15:
Wäre Klasse wenn das jemand mit API-Zugang testen könnte. Wahrscheinlich wird JQ nicht für alle Fälle ausreichen, hier wäre es aber ein Match. Dann fehlt nur noch ein Template-PR für den spezifischen Tarif.

mdkeil, 2024-04-15:
Ich kann es morgen auch mal testen.. habe mir via Node-Red meine Tibber Abfrage passend umgebaut und via mqtt gepublished.

mdkeil, 2024-04-15:
wie sieht denn die Syntax in der yaml aus?

```
tariffs:
  grid:
    type: forecast
    forecast:
      source: mqtt
      topic: nodered/tibber
```

andig, 2024-04-15:
LGTM:

```yaml
tariffs:
  grid:
    type: custom
    forecast:
      source: mqtt
      broker: mq.wsw-online.de:8884
      user: talmarkt-flex
      password: wsw-flex
      topic: talmarktflex/EnergyPricesByGridNo/9900705000001
      jq: '[ .[] | {"start": .from, "end": .to, "price": (.value/100)} ] | tostring'
```

Jetzt fehlt nur noch ein PR für den Tarif, dann kann Beides rein.

closed 2024-04-17 by andig

mdkeil, 2024-04-18:
Auch für die Doku wichtig.. wie muss denn genau api.Rates übergeben werden..?

Ich übergebe testweise folgendes und es funktioniert nicht:

```
[
  {
    "start": "2024-04-18T22:00:00.000Z",
    "end": "2024-04-18T23:00:00.000Z",
    "price": 0.2431
  },
  {
    "start": "2024-04-18T23:00:00.000Z",
    "end": "2024-04-19T00:00:00.000Z",
    "price": 0.2236
  },
  {
    "start": "2024-04-19T00:00:00.000Z",
    "end": "2024-04-19T01:00:00.000Z",
    "price": 0.2114
  },
  {
    "start": "2024-04-19T01:00:00.000Z",
    "end": "2024-04-19T02:00:00.000Z",
    "price": 0.2101
  },
  {
    "start": "2024-04-19T02:00:00.000Z",
    "end": "2024-04-19T03:00:00.000Z",
    "price": 0.2083
  },
  {
    "start": "2024-04-19T03:00:00.000Z",
    "end": "2024-04-19T04:00:00.000Z",
    "price": 0.2097
  },
  {
    "start": "2024-04-19T04:00:00.000Z",
    "end": "2024-04-19T05:00:00.000Z",
    "price": 0.2228
  },
  {
    "start": "2024-04-19T05:00:00.000Z",
    "end": "2024-04-19T06:00:00.000Z",
    "price": 0.2334
  },
  {
    "start": "2024-04-19T06:00:00.000Z",
    "end": "2024-04-19T07:00:00.000Z",
    "price": 0.2417
  },
  {
    "start": "2024-04-19T07:00:00.000Z",
    "end": "2024-04-19T08:00:00.000Z",
    "price": 0.2325
  },
  {
    "start": "2024-04-19T08:00:00.000Z",
    "end": "2024-04-19T09:00:00.000Z",
    "price": 0.2229
  },
  {
    "start": "2024-04-19T09:00:00.000Z",
    "end": "2024-04-19T10:00:00.000Z",
    "price": 0.218
  },
  {
    "start": "2024-04-19T10:00:00.000Z",
    "end": "2024-04-19T11:00:00.000Z",
    "price": 0.2118
  },
  {
    "start": "2024-04-19T11:00:00.000Z",
    "end": "2024-04-19T12:00:00.000Z",
    "price": 0.2072
  },
  {
    "start": "2024-04-19T12:00:00.000Z",
    "end": "2024-04-19T13:00:00.000Z",
    "price": 0.2047
  },
  {
    "start": "2024-04-19T13:00:00.000Z",
    "end": "2024-04-19T14:00:00.000Z",
    "price": 0.2043
  },
  {
    "start": "2024-04-19T14:00:00.000Z",
    "end": "2024-04-19T15:00:00.000Z",
    "price": 0.2118
  },
  {
    "start": "2024-04-19T15:00:00.000Z",
    "end": "2024-04-19T16:00:00.000Z",
    "price": 0.2212
  },
  {
    "start": "2024-04-19T16:00:00.000Z",
    "end": "2024-04-19T17:00:00.000Z",
    "price": 0.2268
  },
  {
    "start": "2024-04-19T17:00:00.000Z",
    "end": "2024-04-19T18:00:00.000Z",
    "price": 0.229
  },
  {
    "start": "2024-04-19T18:00:00.000Z",
    "end": "2024-04-19T19:00:00.000Z",
    "price": 0.2301
  },
  {
    "start": "2024-04-19T19:00:00.000Z",
    "end": "2024-04-19T20:00:00.000Z",
    "price": 0.23
  },
  {
    "start": "2024-04-19T20:00:00.000Z",
    "end": "2024-04-19T21:00:00.000Z",
    "price": 0.2289
  }
]
```

andig, 2024-04-18:
RFC3339     = "2006-01-02T15:04:05Z07:00"

ohne ms

ich77, 2024-04-26:
Vielen Dank! Es funktioniert!