evcc-io/evcc #13439: panic rct bibliothek

TobiasHuber1980, 2024-04-15:
[server] ERROR 2024/04/15 11:12:08 http: panic serving 79.253.13.184:2946: runtime error: index out of range [0] with length 0 goroutine 3103 [running]: net/http.(*conn).serve.func1() net/http/server.go:1898 +0xb0 panic({0x2736160?, 0x4000f200f0?}) runtime/panic.go:770 +0x124 bytes.(*Buffer).WriteByte(0x40012b1068?, 0x0?) bytes/buffer.go:291 +0x94 <http://github.com/mlnoga/rct.(*DatagramBuilder).WriteByteUnescapedNoCRC(...)|github.com/mlnoga/rct.(*DatagramBuilder).WriteByteUnescapedNoCRC(...)> <http://github.com/mlnoga/rct@v0.1.2-0.20230731074838-03eacb926f99/build.go:39|github.com/mlnoga/rct@v0.1.2-0.20230731074838-03eacb926f99/build.go:39> <http://github.com/mlnoga/rct.(*DatagramBuilder).Build(0x4000bb4420|github.com/mlnoga/rct.(*DatagramBuilder).Build(0x4000bb4420>, 0x40012b10d8) <http://github.com/mlnoga/rct@v0.1.2-0.20230731074838-03eacb926f99/build.go:52|github.com/mlnoga/rct@v0.1.2-0.20230731074838-03eacb926f99/build.go:52> +0x4c <http://github.com/mlnoga/rct.(*Connection).Query(0x40005b3240|github.com/mlnoga/rct.(*Connection).Query(0x40005b3240>, 0xb5317b78) <http://github.com/mlnoga/rct@v0.1.2-0.20230731074838-03eacb926f99/connection.go:112|github.com/mlnoga/rct@v0.1.2-0.20230731074838-03eacb926f99/connection.go:112> +0x6c <http://github.com/mlnoga/rct.(*Connection).QueryFloat32(0x40012b1158|github.com/mlnoga/rct.(*Connection).QueryFloat32(0x40012b1158>?, 0xd8bd38?) <http://github.com/mlnoga/rct@v0.1.2-0.20230731074838-03eacb926f99/connection.go:132|github.com/mlnoga/rct@v0.1.2-0.20230731074838-03eacb926f99/connection.go:132> +0x1c <http://github.com/evcc-io/evcc/meter.(*RCT).queryFloat.func1()|github.com/evcc-io/evcc/meter.(*RCT).queryFloat.func1()> <http://github.com/evcc-io/evcc/meter/rct.go:174|github.com/evcc-io/evcc/meter/rct.go:174> +0x28 <http://github.com/cenkalti/backoff/v4.doRetryNotify[...](0x40012b1310|github.com/cenkalti/backoff/v4.doRetryNotify[...](0x40012b1310>?, {0x3509568, 0x40004c8480}, 0x0, {0x0, 0x0?}) <http://github.com/cenkalti/backoff/v4@v4.3.0/retry.go:88|github.com/cenkalti/backoff/v4@v4.3.0/retry.go:88> +0xe0 <http://github.com/cenkalti/backoff/v4.RetryNotifyWithData[...](...)|github.com/cenkalti/backoff/v4.RetryNotifyWithData[...](...)> <http://github.com/cenkalti/backoff/v4@v4.3.0/retry.go:54|github.com/cenkalti/backoff/v4@v4.3.0/retry.go:54> <http://github.com/cenkalti/backoff/v4.RetryWithData[...](0x7fbdb7b108|github.com/cenkalti/backoff/v4.RetryWithData[...](0x7fbdb7b108>?, {0x3509568?, 0x40004c8480?}) <http://github.com/cenkalti/backoff/v4@v4.3.0/retry.go:43|github.com/cenkalti/backoff/v4@v4.3.0/retry.go:43> +0x38 <http://github.com/evcc-io/evcc/meter.(*RCT).queryFloat(0x4000f361a0|github.com/evcc-io/evcc/meter.(*RCT).queryFloat(0x4000f361a0>, 0xb5317b78) <http://github.com/evcc-io/evcc/meter/rct.go:173|github.com/evcc-io/evcc/meter/rct.go:173> +0xb0 <http://github.com/evcc-io/evcc/meter.(*RCT).CurrentPower(0x4000f361a0)|github.com/evcc-io/evcc/meter.(*RCT).CurrentPower(0x4000f361a0)> <http://github.com/evcc-io/evcc/meter/rct.go:116|github.com/evcc-io/evcc/meter/rct.go:116> +0x48 <http://github.com/evcc-io/evcc/server.testInstance({0x2591020|github.com/evcc-io/evcc/server.testInstance({0x2591020>, 0x4000f361a0}) <http://github.com/evcc-io/evcc/server/http_config_helper.go:125|github.com/evcc-io/evcc/server/http_config_helper.go:125> +0x64 <http://github.com/evcc-io/evcc/server.deviceStatusHandler({0x35fc370|github.com/evcc-io/evcc/server.deviceStatusHandler({0x35fc370>, 0x40004c8240}, 0x40?) <http://github.com/evcc-io/evcc/server/http_config_device_handler.go:176|github.com/evcc-io/evcc/server/http_config_device_handler.go:176> +0x13c net/http.HandlerFunc.ServeHTTP(0x400179e360?, {0x35fc370?, 0x40004c8240?}, 0x5?) net/http/server.go:2166 +0x38 <http://github.com/evcc-io/evcc/server.(*HTTPd).RegisterSiteHandlers.ensureAuthHandler.func8.1({0x35fc370|github.com/evcc-io/evcc/server.(*HTTPd).RegisterSiteHandlers.ensureAuthHandler.func8.1({0x35fc370>, 0x40004c8240}, 0x400179e360) <http://github.com/evcc-io/evcc/server/http_auth.go:135|github.com/evcc-io/evcc/server/http_auth.go:135> +0x94 net/http.HandlerFunc.ServeHTTP(0x40009ac480?, {0x35fc370?, 0x40004c8240?}, 0x6d150?) net/http/server.go:2166 +0x38 <http://github.com/gorilla/handlers.(*cors).ServeHTTP(0x400143e120|github.com/gorilla/handlers.(*cors).ServeHTTP(0x400143e120>, {0x35fc370, 0x40004c8240}, 0x400179e360) <http://github.com/gorilla/handlers@v1.5.2/cors.go:54|github.com/gorilla/handlers@v1.5.2/cors.go:54> +0x2d4 <http://github.com/gorilla/handlers.CompressHandler.CompressHandlerLevel.func1({0x35f8098|github.com/gorilla/handlers.CompressHandler.CompressHandlerLevel.func1({0x35f8098>, 0x400157e540}, 0x400179e360) <http://github.com/gorilla/handlers@v1.5.2/compress.go:141|github.com/gorilla/handlers@v1.5.2/compress.go:141> +0x4e8 net/http.HandlerFunc.ServeHTTP(0x26afaa0?, {0x35f8098?, 0x400157e540?}, 0xc?) net/http/server.go:2166 +0x38 <http://github.com/evcc-io/evcc/server.jsonHandler.func1({0x35f8098|github.com/evcc-io/evcc/server.jsonHandler.func1({0x35f8098>, 0x400157e540}, 0x400179e360) <http://github.com/evcc-io/evcc/server/http_site_handler.go:72|github.com/evcc-io/evcc/server/http_site_handler.go:72> +0xf0 net/http.HandlerFunc.ServeHTTP(0x400179e240?, {0x35f8098?, 0x400157e540?}, 0x22488?) net/http/server.go:2166 +0x38 <http://github.com/gorilla/mux.(*Router).ServeHTTP(0x400061da40|github.com/gorilla/mux.(*Router).ServeHTTP(0x400061da40>, {0x35f8098, 0x400157e540}, 0x400179e000) <http://github.com/gorilla/mux@v1.8.1/mux.go:212|github.com/gorilla/mux@v1.8.1/mux.go:212> +0x194 net/http.serverHandler.ServeHTTP({0x4000d31980?}, {0x35f8098?, 0x400157e540?}, 0x6?) net/http/server.go:3137 +0xbc net/http.(*conn).serve(0x400103fc20, {0x35ff308, 0x40017cfce0}) net/http/server.go:2039 +0x508 created by net/http.(*Server).Serve in goroutine 1 net/http/server.go:3285 +0x3f0

[Slack Message](https://evccgroup.slack.com/archives/D05KCKH0QJX/p1713175023584209)

TobiasHuber1980, 2024-04-15:
und 
```
[site ] ERROR 2024/04/15 11:12:50 pv 1 power: invalid response to read of 23494E56414C4944: Cmd Response (05) Id Total grid power [W] (91617C58) Data [191 144 35 201]
[site ] ERROR 2024/04/15 11:12:58 battery 1 soc: invalid response to read of 42617474657279207374617465206F6620636861726765: Cmd Response (05) Id Total grid power [W] (91617C58) Data [64 42 95 72]
[site ] ERROR 2024/04/15 11:12:59 grid meter: invalid response to read of 546F74616C206772696420706F776572205B575D: Cmd Response (05) Id #INVALID (7924ABD9) Data [48 48 54 53 66 52 56 50 51 50 52 54 0 0 0 0]
[site ] ERROR 2024/04/15 11:13:10 pv 1 power: invalid response to read of 23494E56414C4944: Cmd Response (05) Id Total energy grid [Wh] (A59C8428) Data [0 0 0 0]
[site ] ERROR 2024/04/15 11:13:26 battery 1 power: invalid response to read of 4261747465727920706F776572205B575D: Cmd Response (05) Id Total grid power [W] (91617C58) Data [191 57 109 39]
[site ] ERROR 2024/04/15 11:14:07 pv 1 power: invalid response to read of 536F6C61722067656E657261746F72204120706F776572205B575D: Cmd Response (05) Id Battery state of charge (959930BF) Data [63 64 221 5]
[site ] ERROR 2024/04/15 11:14:08 battery 1 power: invalid response to read of 4261747465727920706F776572205B575D: Cmd Response (05) Id Battery state of charge (959930BF) Data [63 64 222 41]
[site ] ERROR 2024/04/15 11:14:39 pv 1 power: invalid response to read of 536F6C61722067656E657261746F72204120706F776572205B575D: Cmd Response (05) Id #INVALID (DB11855B) Data [68 144 162 61]
[site ] ERROR 2024/04/15 11:15:11 pv 1 power: invalid response to read of 23494E56414C4944: Cmd Response (05) Id Battery power [W] (400F015B) Data [196 154 158 148]
[site ] ERROR 2024/04/15 11:15:34 grid meter: invalid response to read of 546F74616C206772696420706F776572205B575D: Cmd Response (05) Id #INVALID (7924ABD9) Data [48 48 54 53 66 52 56 50 51 50 52 54 0 0 0 0]
[site ] ERROR 2024/04/15 11:16:09 pv 1 power: invalid response to read of 536F6C61722067656E657261746F72204120706F776572205B575D: Cmd Response (05) Id #INVALID (DB11855B) Data [68 142 39 114]
[site ] ERROR 2024/04/15 11:16:37 pv 1 power: invalid response to read of 536F6C61722067656E657261746F72204120706F776572205B575D: Cmd Response (05) Id Battery state of charge (959930BF) Data [63 65 203 214]
[site ] ERROR 2024/04/15 11:16:48 grid meter: invalid response to read of 546F74616C206772696420706F776572205B575D: Cmd Response (05) Id #INVALID (DB11855B) Data [68 139 32 156]
[site ] ERROR 2024/04/15 11:17:08 battery 1 power: invalid response to read of 4261747465727920706F776572205B575D: Cmd Response (05) Id Total energy batt in [Wh] (5570401B) Data [73 136 144 105]
[site ] ERROR 2024/04/15 11:17:38 pv 1 power: invalid response to read of 536F6C61722067656E657261746F72204120706F776572205B575D: Cmd Response (05) Id Total energy grid [Wh] (A59C8428) Data [0 0 0 0]
[site ] ERROR 2024/04/15 11:17:46 grid meter: invalid response to read of 546F74616C206772696420706F776572205B575D: Cmd Response (05) Id Battery power [W] (400F015B) Data [196 88 51 44]
[site ] ERROR 2024/04/15 11:18:09 pv 1 power: invalid response to read of 536F6C61722067656E657261746F72204120706F776572205B575D: Cmd Response (05) Id Battery state of charge (959930BF) Data [63 66 74 216]
[site ] ERROR 2024/04/15 11:18:37 pv 1 power: invalid response to read of 536F6C61722067656E657261746F72204120706F776572205B575D: Cmd Response (05) Id Battery state of charge (959930BF) Data [63 66 107 142]
[site ] ERROR 2024/04/15 11:18:38 battery 1 power: invalid response to read of 4261747465727920706F776572205B575D: Cmd Response (05) Id Battery state of charge (959930BF) Data [63 66 108 177]
[site ] ERROR 2024/04/15 11:19:37 pv 1 power: invalid response to read of 536F6C61722067656E657261746F72204120706F776572205B575D: Cmd Response (05) Id Battery power [W] (400F015B) Data [196 30 154 49]
[site ] ERROR 2024/04/15 11:20:10 battery 1 power: invalid response to read of 4261747465727920706F776572205B575D: Cmd Response (05) Id #INVALID (DB11855B) Data [68 64 210 228]
[site ] ERROR 2024/04/15 11:20:42 pv 1 power: invalid response to read of 536F6C61722067656E657261746F72204220706F776572205B575D: Cmd Response (05) Id Battery power [W] (400F015B) Data [196 17 61 229]
[site ] ERROR 2024/04/15 11:20:45 grid energy: invalid response to read of 546F74616C20656E657267792067726964205B57685D: Cmd Response (05) Id Total energy batt in [Wh] (5570401B) Data [73 136 145 209]
[site ] ERROR 2024/04/15 11:21:08 pv 1 power: invalid response to read of 536F6C61722067656E657261746F72204120706F776572205B575D: Cmd Response (05) Id #INVALID (DB11855B) Data [68 63 104 60]
[site ] ERROR 2024/04/15 11:21:09 battery 1 power: invalid response to read of 4261747465727920706F776572205B575D: Cmd Response (05) Id Total grid power [W] (91617C58) Data [63 28 25 234]
[site ] ERROR 2024/04/15 11:21:37 pv 1 power: invalid response to read of 536F6C61722067656E657261746F72204120706F776572205B575D: Cmd Response (05) Id Battery state of charge (959930BF) Data [63 67 26 242]
[site ] ERROR 2024/04/15 11:21:41 battery 1 power: invalid response to read of 4261747465727920706F776572205B575D: Cmd Response (05) Id #INVALID (DB11855B) Data [68 57 27 86]
[site ] ERROR 2024/04/15 11:22:09 pv 1 power: invalid response to read of 536F6C61722067656E657261746F72204120706F776572205B575D: Cmd Response (05) Id Battery power [W] (400F015B) Data [196 22 142 39]
[... cut ...]
[site ] ERROR 2024/04/15 12:12:37 pv 1 power: invalid response to read of 536F6C61722067656E657261746F72204120706F776572205B575D: Cmd Response (05) Id Battery power [W] (400F015B) Data [196 78 227 74]
[site ] ERROR 2024/04/15 12:12:55 grid meter: invalid response to read of 546F74616C206772696420706F776572205B575D: Cmd Response (05) Id Battery state of charge (959930BF) Data [63 79 55 216]
[site ] ERROR 2024/04/15 13:21:37 pv 1 power: parsing failed in state 0
[site ] ERROR 2024/04/15 13:21:39 battery 1 power: invalid response to read of 4261747465727920706F776572205B575D: Cmd Response (05) Id #INVALID (7924ABD9) Data [48 48 54 53 66 52 56 50 51 50 52 54 0 0 0 0]
```

andig, 2024-04-15:
Was hat das mit dem Log zu tun? Wieso ist der Fehler nicht im Log?

TobiasHuber1980, 2024-04-15:
> Was hat das mit dem Log zu tun? Wieso ist der Fehler nicht im Log?

edit: jetzt alles in einem log im issue.

andig, 2024-04-15:
Sieht aus als käme der Aufruf aus dem UI, ist also concurrent zum Backend. Anscheinend ist die RCT Bibliothek nicht concurrency-safe. Das fällt schon direkt beim Connection Cache auf und setzt sich über gleichzeitige Zugriffe auf den Builder fort. Bei Compile mit `-race` sollte das Logging entsprechende Fehler zeigen. Hier wäre es notwendig, alle Zugriffe zu serialisieren.

@mlnoga  was meinst Du? Gibts einen besseren Weg?

closed 2024-04-19 by andig