evcc-io/evcc #13444: UI: verzögerte Log-Ausgabe

VolkerK62, 2024-04-15:
### Describe the bug

Wenn ich im UI das Log aufrufe, dann ist der letzte Eintrag immer ca 36 Minuten alt

Aktuelle Zeit 16:34

[image]

Lasse ich dann "Alle Einträge anzeigen", sind die letzten Einträge aktuell, allerdings eintspricht der timestamp des letzten Eintrages dem des Logbeginn

[image]

### Steps to reproduce

.

### Configuration details

```yaml
.
```

### Log details

```text
.
```

### What type of operating system are you running?

Linux

### Version

0.125.0 (39db91c9)

VolkerK62, 2024-04-15:
Nach einem Neustart von evcc wird alles richtig angezeigt.
Ist evtl ein Problem, das erst nach einiger Zeit auftritt.

andig, 2024-04-15:
> Die Download Version ist vollständig.

@naltatis das klingt nach ui bug?

andig, 2024-04-15:
Closed in https://github.com/evcc-io/evcc/commit/535c94c2fe547195db1807def7450b4e68ec6915. Nicht die beste Performance, aber das können wir noch lösen wenn es ein Problem werden sollte.

closed 2024-04-15 by andig

naltatis, 2024-04-19:
> ABER, die letzte Zeile ist immer eine älteren.

Das ist merkwürdig. Riecht nach Datenproblem. @andig

andig, 2024-04-19:
Kann ich nicht nachvollziehen. Bei mir ist auch die letzte Zeile aktuell. Da müsstet ihr mal parallel in den Request schauen ob auch die Antwort schon falsche Reihenfolge hat?

VolkerK62, 2024-04-19:
Nach ein paar Minuten ist es wieder da.
UNd es hat Prinzip.
Wenn ich "alle Einträge anzeigen" aktiviere, dann ist der letze Eintrag in der Liste, der, der vor dem ersten Eintrag stehen würde

Letzter Eintrag:` [lp-5 ] DEBUG 2024/04/19 17:05:28 charge power: 0W`
Erster Eintrag: `[lp-6 ] DEBUG 2024/04/19 17:05:28 charge power: 0W`

andig, 2024-04-19:
Was aber meine Frage nicht beantwortet...

naltatis, 2024-04-19:
Ich kann den Fehler hier auch nachvollziehen. Das passiert, wenn der Ring-Buffer vollgelaufen ist und anfängt vorne zu löschen. Ich hab das bei mir beschleunigt mal mit der demo.yaml und `interval: .05s` provoziert. Da hat man den Effekt dann sehr schnell.

andig, 2024-04-20:
> Sorry, die Frage versteh ich nicht.

@VolkerK62 Du kannst in der Entwicklerkonsole vom Browser sehen, welche Daten da hin und her geschickt werden. Damit lässt sich prüfen, ob Backend oder UI den Fehler machen. Hier ists jetzt aber klar :)

Fixed in https://github.com/evcc-io/evcc/commit/71f7dd756f93668d132aaeefa1a83ade189c28e4

VolkerK62, 2024-04-20:
ok. Danke für die Erklärung.