evcc-io/evcc #13446: Kostenberechnung Fehlerhaft

cschnatz, 2024-04-15:
### Describe the bug

Die Kostenberechnung in der Ladeübersicht scheint Fehlerhaft zu sein. Sowohl Grafana als auch Homeassistant melden für heute (fast) keinen Strombezug aus dem Netz, dennoch werden kosten für den Ladevorgang in der Ladeübersicht ausgewiesen. 

[image]
[image]
[image]
[image]

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: e1...

interval: 30s # control cycle interval

meters:
- type: template
  template: rct-power 
  usage: grid  
  host: 10.XX.XX.12  
  name: grid1
- type: template
  template: rct-power 
  usage: pv  
  host: 10.XX.XX.12  
  name: pv2
- type: template
  template: rct-power 
  usage: battery  
  host: 10.XX.XX.12  
  name: battery3

vehicles:
- type: template
  template: vw 
  title: ID3  
  user: c...z@gmail.com  
  password: H...  
  vin: WV... 
  capacity: 58  
  timeout: 10s  
  name: ev4
- name: ev5
  type: template
  template: tesla
  title: Model Y 
  accessToken: ey...
  refreshToken: EU_cd...
  vin: XP...
  capacity: 79

[... 613 lines cut ...]
Apr 15 14:40:12 csc-evcc evcc[16251]: [site  ] DEBUG 2024/04/15 14:40:12 grid meter: -0W
Apr 15 14:40:42 csc-evcc evcc[16251]: [site  ] DEBUG 2024/04/15 14:40:42 grid meter: 1W
Apr 15 14:41:12 csc-evcc evcc[16251]: [site  ] DEBUG 2024/04/15 14:41:12 grid meter: -4W
Apr 15 14:42:12 csc-evcc evcc[16251]: [site  ] DEBUG 2024/04/15 14:42:12 grid meter: 26W
Apr 15 14:42:42 csc-evcc evcc[16251]: [site  ] DEBUG 2024/04/15 14:42:42 grid meter: -1W
Apr 15 14:43:12 csc-evcc evcc[16251]: [site  ] DEBUG 2024/04/15 14:43:12 grid meter: 2W
Apr 15 14:43:42 csc-evcc evcc[16251]: [site  ] DEBUG 2024/04/15 14:43:42 grid meter: -1W
Apr 15 14:44:12 csc-evcc evcc[16251]: [site  ] DEBUG 2024/04/15 14:44:12 grid meter: -1W
Apr 15 14:44:42 csc-evcc evcc[16251]: [site  ] DEBUG 2024/04/15 14:44:42 grid meter: 1W
Apr 15 14:45:12 csc-evcc evcc[16251]: [site  ] DEBUG 2024/04/15 14:45:12 grid meter: -1W
Apr 15 14:45:42 csc-evcc evcc[16251]: [site  ] DEBUG 2024/04/15 14:45:42 grid meter: 1W
Apr 15 14:46:12 csc-evcc evcc[16251]: [site  ] DEBUG 2024/04/15 14:46:12 grid meter: -1W
Apr 15 14:46:42 csc-evcc evcc[16251]: [site  ] DEBUG 2024/04/15 14:46:42 grid meter: -1W
Apr 15 14:47:12 csc-evcc evcc[16251]: [site  ] DEBUG 2024/04/15 14:47:12 grid meter: -1W
Apr 15 14:47:42 csc-evcc evcc[16251]: [site  ] DEBUG 2024/04/15 14:47:42 grid meter: -666W
Apr 15 14:48:12 csc-evcc evcc[16251]: [site  ] DEBUG 2024/04/15 14:48:12 grid meter: -138W
```

### What type of operating system are you running?

Linux

### Version

0.125.0

VolkerK62, 2024-04-15:
hmm, wenn man "Tesla", "Haus" und "Speicher" addiert und davon "PV-Erzeugung" abzieht, bleiben ca. 5 kWh übrig. Das sind 25% von 20kWh geladene Energie.
Somit passt der Wert von 75% Sonne und der Preis passt dann auch.

cschnatz, 2024-04-15:
@VolkerK62 ja das sehe ich auch, gleichzeitg steht bei Netzbezug 0 und das aus drei Datenquellen (evcc, homeassistant, RCT Cloud Portal).

Wird hier evtl. die Energie aus dem Speicher als kostenpflichtig berechnet?

cschnatz, 2024-04-15:
Die Summer aller Werte von "grid meter" ergibt -968W über den Logzeitraum. Da komme ich auch nicht auf 5kWh

VolkerK62, 2024-04-15:
>Wird hier evtl. die Energie aus dem Speicher als kostenpflichtig berechnet?

Nein, Speicher ist wie PV, aber kalkulatorisch wird zuerst das Haus versorgt und dann erst das Auto.
https://docs.evcc.io/docs/faq#ersparnisberechnung

StefanSchoof, 2024-04-15:
> keinen Strombezug aus dem Netz, dennoch werden kosten für den Ladevorgang in der Ladeübersicht ausgewiesen

Nicht bekomme Einspeisevergütung wird als Preis des Ladevorgangs ausgewiesenen.

VolkerK62, 2024-04-15:
Deine Wallbox hat keinen Zähler, richtig?
Bin mir nicht ganz sicher, aber das könnte durch die "geschätzte" Lademenge ebenfalls zu einer Unschärfe führen.

cschnatz, 2024-04-15:
Angeblich hat die Elli Connect Pro ein Zähler aber dieser wird von evcc nicht erkannt. Wobei ich mir auch vorstellen kann, dass das an der wunderbaren Software auf den Elli Boxen liegt. Die zweite bekommt nichtmal mehr eine WLAN Verbindung hin...

Auffällig ist auch, dass der Hausstrom auf 0 geht sobald ein Ladevorgang startet...

StefanSchoof, 2024-04-15:
Wenn du die Pro hast, solltest du auch das pro Template nehmen: https://docs.evcc.io/docs/devices/chargers/#charger-pro-beta-1

cschnatz, 2024-04-15:
@naltatis hier als Screenshot im Anhang. 

[image]

naltatis, 2024-04-15:
Oh, das wackelt aber ordentlich dafür dass eigentlich kein Netzbezug vorhanden war.

Kannst du noch mal log mit den anderen Leistungswerten (pv-, battery-, chargepower) posten?

StefanSchoof, 2024-04-15:
Fehlende mA Steuerung der VW Software wie unter https://github.com/evcc-io/evcc/discussions/10487#discussioncomment-8831479?

Lösung wäre dann `features: ["coarsecurrent"]`.

cschnatz, 2024-04-15:
@naltatis das ist ein sehr langer Log ;) 

[evcc.log](https://github.com/evcc-io/evcc/files/14983447/evcc.log)

naltatis, 2024-04-17:
Ok, da sind merkwürdige Einträge drin.

```
Apr 15 10:09:11 csc-evcc evcc[16251]: [lp-1  ] DEBUG 2024/04/15 10:09:11 charge power: 0W
Apr 15 10:09:12 csc-evcc evcc[16251]: [site  ] DEBUG 2024/04/15 10:09:12 pv power: 3950W
Apr 15 10:09:12 csc-evcc evcc[16251]: [site  ] DEBUG 2024/04/15 10:09:12 battery soc: 74%
Apr 15 10:09:12 csc-evcc evcc[16251]: [site  ] DEBUG 2024/04/15 10:09:12 battery power: -3523W
Apr 15 10:09:12 csc-evcc evcc[16251]: [site  ] DEBUG 2024/04/15 10:09:12 grid meter: 0W
Apr 15 10:09:12 csc-evcc evcc[16251]: [site  ] DEBUG 2024/04/15 10:09:12 site power: -3422W
Apr 15 10:09:12 csc-evcc evcc[16251]: [lp-1  ] DEBUG 2024/04/15 10:09:12 charger status: B
Apr 15 10:09:12 csc-evcc evcc[16251]: [lp-1  ] DEBUG 2024/04/15 10:09:12 pv charge current: 4.96A = 0A + 4.96A (-3422W @ 3p)
```

Hier wird, warum auch immer, -3kW Hausverbrauch errechnet 🤔

Generell passt die Bilanz (Energiequellen vs. Verbraucher) an ganz vielen Stellen nicht. Mir ist noch nicht klar, woran das liegt.

[image]

Hier die Daten aus dem Log in CSV konvertiert:
[evcc.csv](https://github.com/evcc-io/evcc/files/15008160/evcc.csv)

\\cc @MarkusGH @premultiply @andig

naltatis, 2024-04-17:
Hier noch mal eine Aktualisiertung.

[image]

Man sieht, dass die Charge Power größer ist, als die Summe der Hauszähler. Schaut man sich den Anteil des roten Bereichs an kann das auch ungefähr die 25% fehlenden Eigenanteil den du beschreibst erklären.

```
Apr 15 16:14:11 csc-evcc evcc[16251]: [site  ] DEBUG 2024/04/15 16:14:11 ----
Apr 15 16:14:11 csc-evcc evcc[16251]: [lp-1  ] DEBUG 2024/04/15 16:14:11 charge power: 4140W
Apr 15 16:14:12 csc-evcc evcc[16251]: [site  ] DEBUG 2024/04/15 16:14:12 pv power: 206W
Apr 15 16:14:12 csc-evcc evcc[16251]: [site  ] DEBUG 2024/04/15 16:14:12 battery soc: 91%
Apr 15 16:14:12 csc-evcc evcc[16251]: [site  ] DEBUG 2024/04/15 16:14:12 battery power: 3019W
Apr 15 16:14:12 csc-evcc evcc[16251]: [site  ] DEBUG 2024/04/15 16:14:12 grid meter: 1W
Apr 15 16:14:12 csc-evcc evcc[16251]: [site  ] DEBUG 2024/04/15 16:14:12 site power: 3120W
Apr 15 16:14:12 csc-evcc evcc[16251]: [lp-1  ] DEBUG 2024/04/15 16:14:12 charger status: C
Apr 15 16:14:12 csc-evcc evcc[16251]: [lp-1  ] DEBUG 2024/04/15 16:14:12 soc estimated: 71.23% (vehicle: 70.00%)
Apr 15 16:14:12 csc-evcc evcc[16251]: [lp-1  ] DEBUG 2024/04/15 16:14:12 vehicle soc: 71%
Apr 15 16:14:12 csc-evcc evcc[16251]: [lp-1  ] DEBUG 2024/04/15 16:14:12 vehicle soc limit: 80%
Apr 15 16:14:12 csc-evcc evcc[16251]: [lp-1  ] DEBUG 2024/04/15 16:14:12 vehicle range: 227km
Apr 15 16:14:12 csc-evcc evcc[16251]: [lp-1  ] DEBUG 2024/04/15 16:14:12 pv charge current: 1.48A = 6A + -4.52A (3120W @ 3p)
```

cschnatz, 2024-04-17:
@naltatis ich kann dir hier leider gerade nicht ganz folgen. Wie kann der Charger mehr ziehen als pv+bat+grid liefern?

cschnatz, 2024-04-17:
There has to be something wrong with the charger power calcuation. It says its running at 11KW but the tesla is reporting only 5KW.

[image]

[image]

[image]

```
Apr 17 10:26:17 csc-evcc evcc[50335]: [site  ] DEBUG 2024/04/17 10:26:17 ----
Apr 17 10:26:17 csc-evcc evcc[50335]: [lp-1  ] DEBUG 2024/04/17 10:26:17 charge power: 0W
Apr 17 10:26:17 csc-evcc evcc[50335]: [lp-2  ] DEBUG 2024/04/17 10:26:17 charge power: 11040W
Apr 17 10:26:17 csc-evcc evcc[50335]: [site  ] DEBUG 2024/04/17 10:26:17 pv power: 8260W
Apr 17 10:26:17 csc-evcc evcc[50335]: [site  ] DEBUG 2024/04/17 10:26:17 battery soc: 90%
Apr 17 10:26:17 csc-evcc evcc[50335]: [site  ] DEBUG 2024/04/17 10:26:17 battery power: -2048W
Apr 17 10:26:17 csc-evcc evcc[50335]: [site  ] DEBUG 2024/04/17 10:26:17 grid meter: 2W
Apr 17 10:26:18 csc-evcc evcc[50335]: [site  ] DEBUG 2024/04/17 10:26:18 site power: -1946W
Apr 17 10:26:18 csc-evcc evcc[50335]: [lp-2  ] DEBUG 2024/04/17 10:26:18 charger status: C
Apr 17 10:26:18 csc-evcc evcc[50335]: [lp-2  ] DEBUG 2024/04/17 10:26:18 vehicle status: C
Apr 17 10:26:18 csc-evcc evcc[50335]: [lp-2  ] DEBUG 2024/04/17 10:26:18 vehicle soc: 81%
Apr 17 10:26:18 csc-evcc evcc[50335]: [lp-2  ] DEBUG 2024/04/17 10:26:18 vehicle soc limit: 90%
Apr 17 10:26:18 csc-evcc evcc[50335]: [lp-2  ] DEBUG 2024/04/17 10:26:18 vehicle range: 418km
```

VolkerK62, 2024-04-17:
>Eine mögliche Lösung wäre hier vmtl. dass wir, wenn wir die Charge Power schätzen müssen, den Wert nicht größer werden lassen als auf der Input Seite verfügbar ist.

Da solche Fälle immer wieder auftauchen, wäre das mMn eine sehr gute Idee.

naltatis, 2024-04-17:
> Da solche Fälle immer wieder auftauchen, wäre das mMn eine sehr gute Idee.

Ist nur leider nicht trivial. Bei mehreren Ladepunkten müsste man dann irgendwie untereinander ausgleichen :/

Als einfacheren Fix könnte man auch erstmal "nur" eine Prüfung in der Eigenanteilsrechnung einbauen (das war ja der Aufhänger des Issues), aber das löst, wie du schon gesagt hast, ja das Problem nicht wirklich. Dann ist zwar der Sonnenanteil korrekt(er), aber die abgeschätzte Energiemenge und die Summen und Preise darauf sind immer noch Käse.

andig, 2024-04-17:
Voraussetzung für korrekte Ergebnisse sind korrekte Messwerte. Das setzt "Messwerte" voraus.