evcc-io/evcc #13450: Senec Batterie blockieren bei Auto laden

KneterPeter, 2024-04-15:
Der Speicher (Senec v3 hybrid duo) ist voll und reicht für die Nacht.
Tibber ist billig zum Abend wegen Strum am Sonntag. 
Möchte Auto schnell aus dem Netz laden
Dann wird der Akku entladen.

Es gibt die Funktion, den Speicher nur für das Haus zu nutzen und nicht für das Auto freizugeben. Aber leider nur in der Senec App. 
[image]

In HomeAssistant in der Senec Integration ist der Switch auch zu finden. 
 [image]

Kann man die Funktion in evcc einbinden?

andig, 2024-04-15:
> Aber leider nur in der Senec App. Kann man die Funktion in evcc einbinden?

Wenns keine Schnittstelle gibt nein.

closed 2024-04-15 by andig

KneterPeter, 2024-04-16:
Schnittstelle gefunden. 

[image]

Https://IPdesSenecs//vars.html

Ich tippe Mal drauf, dass Senec dann den Wechselrichter drosselt um nur die Strommenge fürs Haus freizugeben und die 11kw fürs Auto aus dem Netz kommen. 

Ich werde die Tage mal versuchen herauszufinden, wie das klappt. Der V3 hybrid Duo hat 2 Wechselrichter, die beide per LAN angesteuert werden können. 

Das sollte doch Reverse engeniering mäßig machbar sein, oder?

premultiply, 2024-04-17:
Gern. PR welcome.

KneterPeter, 2024-04-19:
So, ich habe etwas herum experimentiert. 
Über irgendwelche Änderungen in der Vars.html geht das nicht. 

Ich habe jetzt Hardwaremäßig etwas umgebaut. Ich habe den ETH to rs485 und den Senec Speicher gleichzeitig an der Wallbox verbunden. 
Allerdings den Senecspeicher nur mit A+ und B-. 
Ground habe ich nicht verbunden. 

Bis jetzt scheint es alles zu klappen. 

Der Hausbedarf wird über die Batterie gedeckt und das Auto über Netzstrom. 

Anbei ein paar Screenshots. Einziger Nachteil. Es kommen natürlich ne Menge Fehlermeldungen im evcc aber bis jetzt läuft es so, wie ich es wollte. 

[image]
.
[image]

.
[image]

premultiply, 2024-04-19:
Mehrere Master am Bus sind bei Modbus nicht zulässig.

Marvin-Janssen, 2025-02-03:
Hallo zusammen, ich habe einen Senec Home V3 und eine ABL eMh1. Beides in evcc integriert, aber nicht die WallBox an dem Senec Speicher. Wenn ich den Akku beim Schnell Laden „Sperre“ wird leider der Akku immer voll mit genutzt. Dies sollte ja eigentlich in der Speicher Einstellung unterbunden werden. Evcc Version 0.133.0 muss ich an dem Senec Speicher noch irgendetwas einstellen, das dies funktioniert?

[image]

KneterPeter, 2025-02-04:
Hi, meines Wissens nach, können die Senecspeicher das nicht. 
Das funktioniert nur über die eigene App von Senec.
Leider.

Marvin-Janssen, 2025-10-16:
Also wenn ich jetzt nicht total daneben liege, ist der EnFluRi noch vor meiner Hauptsicherung. Also direkt hinter dem Zähler. Ich glaube das wäre nicht wirklich zulässig. Außerdem wäre dann auch kein Solaroptimiertes laden mehr möglich. Also ich kann mich auch täuschen und lasse mich gerne eines besseren belehren. Meiner Meinung nach müsste Senec einfach einen entsprechenden Datenpunkt zur Verfügung stellen, der den Speicher sperrt.

KneterPeter, 2025-10-16:
Einfach die Hausbatterie auch nachts laden lassen.

premultiply, 2025-10-17:
> Was wäre, wenn man die Wallbox einfach nicht über den EnFluRi des Senec-Speichers laufen lässt?

Das ist eine ganz blöde Idee und würde das gesamte Energiemanagement kaputt machen.

Es kann grundsätzlich nur funktionieren wenn der Zähler an der Wurzel sitzt und alle Energieflüsse vom und zum Netz aller dahinter liegenden Verbraucher erfasst.

salana2015, 2026-09-27:
Moin zusammen,
ich hab mir vor 3 Jahren folgende Lösung gebaut:
Ab Frühjahr, wenn genug Sonne für Sence Speicher und Auto da ist, bezieht die Wallbox hinter dem EnFluRi ihren Strom und ich hab die Batterie im evcc integriert (eigentlich nur, damit in kurzen Wolken-Schattenphasen bis zur Regelung der Strom aus dem Speicher gezogen wird)
Ab November, wenn die Sonner bei mir für Haus und Auto nicht mehr reicht, hab ich einen 4 Phasen Umschalter + FI im Schaltschrank und die Wallbox wird umgeschaltet, dass der Strom direkt vor dem EnFluRi abgezapft wird. Somit kennt der Senec den Verbraucher nicht und entlädt nicht den Akku, sondern kennt nur die restlichen Hausverbraucher.
Im Winter lade ich rein CO2 gesteuert über evcc, da die PV Leistung nicht einmal fürs Haus reicht (Abschattung durch Nordhang Lage).

Ich hab auch schon darüber nachgedacht das Kabel zwischen EnFluRi und dem Senec zu unterbrechen und einen entsprechenden Modbus Proxy mit Modifikation (ziehe Wallbox Verbrauch - gemessen über einen 3phasen Shelly) vom gemessenen EnFluRi Strom einzubauen, aber bisher nicht die Motivation dafür gehabt...

Freue mich, wenn jemand Ideen zur Verbesserung hat.
Ideal wäre doch, wenn man jede Wallbox über ModBus an den Senec anschließen könnte und nicht nur die kompatiblen und somit die Daten im Senec hätte.