evcc-io/evcc #13455: BMW/Mini SoC funktioniert nicht mehr

NoPlanetB, 2024-04-16:
### Describe the bug

Mini-SoC kann in openWB v2 (aktueller Master) über evcc nicht abgerufen werden und wird als 0% angezeigt. Laut openWB-Forum hat BMW am 11.04. eine Änderung an der Authentifizierung vorgenommen. Anmerkung: auch das native Mini-SoC-Modul von openWB funktioniert seit dem 11.04. nicht mehr. In Tronity wurde das Problem bereits behoben. 

### Steps to reproduce

Für die Anzeige 0%: openWB Hauptseite aufrufen, SoC für Mini inspizieren
Für die Fehlermeldung: openWB Status aufrufen, Fahrzeug Mini aufklappen

### Configuration details

```yaml
n/a wg. openWB
```

### Log details

```text
Fehlermeldung in openWB:
<class 'grpc.RpcError'> (<_InactiveRpcError of RPC that terminated with: status = StatusCode.UNKNOWN details = "cannot create vehicle type 'mini': Client authentication failed (e.g., login failure, unknown client, no client authentication included or unsupported authentication method): unexpected status: 401 (Unauthorized)" debug_error_string = "UNKNOWN:Error received from peer {grpc_message:"cannot create vehicle type \'mini\': Client authentication failed (e.g., login failure, unknown client, no client authentication included or unsupported authentication method): unexpected status: 401 (Unauthorized)", grpc_status:2, created_time:"2024-04-16T06:42:54.368796163+02:00"}" >,)
```

### What type of operating system are you running?

Linux

### Version

openWB 2.1.4-Alpha.1 (2024-04-12 15:27:16 +0200 [bb4e49158])

andig, 2024-04-16:
Working for me:

```
gor main.go vehicle --log trace,db:error --log-headers -c i3.yaml --cloud
[main  ] INFO 2024/04/16 08:09:53 evcc 0.0.0
[main  ] INFO 2024/04/16 08:09:53 using config file: i3.yaml
Soc:         73%
Capacity:    0.0kWh
Identifiers: [F0:7F:0C:04:C9:1A]
Features:    [CoarseCurrent]
```

Could you check if your token is expired or contact info@evcc.io with your token?

NoPlanetB, 2024-04-16:
The token works fine with our other car, Ford Mustang Mach-E. 
But anyway, I am going to mail it to info@evcc.io.
Thanks

andig, 2024-04-17:
> unexpected status: 401 (Unauthorized)

Das Problem sind vmtl. die Zugangsdaten zum Fahrzeug, nicht das Token.

NoPlanetB, 2024-04-18:
Stimmt tatsächlich, danke und sorry für die Mühe. 

 

Zur Erklärung: ich hatte, um das auszuschließen, mit einem anderen
Mini-Konto getestet, bevor ich das Issue öffnete; das funktionierte aber
auch nicht. Irgendwie wurden wohl durch meine Umstellung auf openWB v2 diese
beiden Mini-Konten gesperrt. Erfahrungsgemäß werden die Konten von Mini
unter manchem Umständen (k. A. welche) temporär für den Zugriff durch
Nicht-Mini-Clients gesperrt und nach geraumer Zeit wieder automatisch
entsperrt. Da aber zeitgleich im openWB-Forum eine Umstellung der Anmeldung
bei Mini berichtet wurde und das native Modul seitdem nicht funktioniert,
nahm ich an, dies müsse auch evcc betreffen.

 

Danke nochmals

#13455)

 

	
You don't often get email from ***@***.***
<https://aka.ms/LearnAboutSenderIdentification> 

	

<https://eur03.safelinks.protection.outlook.com/?url=https%3A%2F%2Fgithub.co
m%2Fevcc-io%2Fevcc%2Fissues%2F13455%23issuecomment-2062448703&data=05%7C02%7
Chenning.weber%40sap.com%7C468f513f5a7b499a50db08dc5f2689ac%7C42f7676cf45542
3c82f6dc2d99791af7%7C0%7C0%7C638489866333236299%7CUnknown%7CTWFpbGZsb3d8eyJW
IjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C0%7C%7C%7C&
sdata=OItPl5%2FUim65e%2BPF712qHpT4bWXOq9ToXzmw1Wi4FJc%3D&reserved=0> , or
unsubscribe.
<https://github.com/notifications/beacon/AUMMSZRAK6GI4GDRALEFB43Y53TQJA5CNFS
M6AAAAABGITLQBGWGG33NNVSW45C7OR4XAZNMJFZXG5LFINXW23LFNZ2KUY3PNVWWK3TUL5UWJTT

NoPlanetB, 2024-04-18:
Seit ein paar Stunden tritt nun folgender neuer Fehler auf. Das liegt nun aber nicht an meinen Zugangsdaten - oder?
<class 'grpc.RpcError'> (<_InactiveRpcError of RPC that terminated with: status = StatusCode.UNKNOWN details = "Get "https://cocoapi.bmwgroup.com/eadrax-vcs/v4/vehicles/state?apptimezone=120&appDateTime=1713460851623": unexpected status: 400 (Bad Request)" debug_error_string = "UNKNOWN:Error received from peer {created_time:"2024-04-18T19:20:51.719482648+02:00", grpc_status:2, grpc_message:"Get \"https://cocoapi.bmwgroup.com/eadrax-vcs/v4/vehicles/state?apptimezone=120&appDateTime=1713460851623\": unexpected status: 400 (Bad Request)"}" >,)

andig, 2024-04-18:
Geht bei mir immer noch. Bitte komplette Zugangsdaten an info@evcc.io

andig, 2024-04-18:
Also working with your credentials:

```
gor main.go vehicle --log trace,db:error --log-headers -c cfg/mini.yaml --cloud
[main  ] INFO 2024/04/18 20:46:43 evcc 0.0.0
[main  ] INFO 2024/04/18 20:46:43 using config file: cfg/mini.yaml
Soc:      99%
Capacity: 0.0kWh
Features: []
```

It is also working when I call the cmd/soc command:

    cmd/soc mini --user ... --token ey...
    99

NoPlanetB, 2024-04-19:
OK, danke. Aber wie komme ich jetzt weiter?

NoPlanetB, 2024-04-19:
Habe mal meine alte standalone openWB auf 1.9.303 reaktiviert; dort funktioniert die Mini-SoC-Abfrage. Dann liegt es wohl am Set-up meiner Custom openWB auf v2.1.4 oder an mrinas' evcc-Integration in 2.1.4. Ich forsche mal in diese Richtung weiter. Trotzdem danke.

andig, 2024-04-19:
Ahh, danke für die Info. Evtl. ein Problem mit den Sonderzeichen im Passwort? Ich mache hier mal zu :)

closed 2024-04-19 by andig