evcc-io/evcc #13457: Volkswagen We Connect ID: vehicle soc: unknown engine type: electric

yewbacca, 2024-04-16:
### Describe the bug

Im Falle von PHEV (Passat GTE) wird bei der vehicle soc Abfrage ein im Vergleich zu "echten" BEV (ID) Fahrzeugen leicht unterschiedlicher vwConn.idData:data:rangeStatus geliefert, der offenbar nicht richtig ausgewertet werden kann und zu dem Fehler im Titel führt. 

Hier der Datensatz, der beim Passat GTE zurückkommt:
	"rangeStatus":{
		"carCapturedTimestamp":"2021-10-02T14:50:03Z",              // UTC/locale time abhängig ob am laden oder nicht
		"carType":"hybrid",                                         // hybrid|electric
		"primaryEngine":{
			"type":"gasoline",
			"currentSOC_pct":85,
			"remainingRange_km":440                                 // !!! nicht fest km, sondern Einheit aus Setup (mi/km)
		},
		"secondaryEngine":{                                         // only when carType=hybrid
			"type":"electric",
			"currentSOC_pct":99,
			"remainingRange_km":34                                  // !!! nicht fest km, sondern Einheit aus Setup (mi/km)
		},
		"totalRange_km":474                                         // !!! nicht fest km, sondern Einheit aus Setup (mi/km)
	},

Wie man sieht, ist beim PHEV die gesuchte Information im "secondaryEngine" Block enthalten, erkennbar über den 
		"carType":"hybrid", // hybrid|electric
am Anfang des Datensatzes...		
		
Alternativ sollte anstelle des rangeStatus Blocks auch der batteryStatus verwendbar sein, nach bisheriger Erkenntnis bei PHEV/BEV identisch aufgebaut:
    "batteryStatus": {
      "carCapturedTimestamp":"2021-02-13T10:38:00Z",                // UTC time
      "currentSOC_pct":99,
      "cruisingRangeElectric_km":34                                 // !!! nicht fest km, sondern Einheit aus Setup (mi/km)
    },
		

### Steps to reproduce

Konfiguration eines PHEV von Volkswagen mit "Volkswagen We Connect ID" type
Bei der Prüfung am Ende der UI Device Konfiguration kommt der Fehler,
bei der Konfiguration via sudo evcc configure ebenfalls bei der Prüfung am Ende.
Wird das Fahrzeug trotzdem in die Konfiguration aufgenommen, kommt der Fehler bei jeder Anwahl des Fahrzeuges

### Configuration details

```yaml
type: template
  template: vw
  title: Passat GTE
  icon: car
  user: xxx.dom.de
  password: 'xyz'
  vin: WVWZZZ...........
  capacity: 13
  phases: 1
  cache: 15m
  mode: pv
  minCurrent: 6
  maxCurrent: 16
  priority: 1
  timeout: 10s
  name: ev1
```

### Log details

```text
[lp-1  ] INFO 2024/04/16 12:58:28 vehicle updated: unknown -> Passat GTE
[lp-1  ] DEBUG 2024/04/16 12:58:28 set charge mode: pv
[site  ] DEBUG 2024/04/16 12:58:28 ----
[lp-1  ] DEBUG 2024/04/16 12:58:28 charge power: 0W
[site  ] DEBUG 2024/04/16 12:58:28 pv power: 549W
[site  ] DEBUG 2024/04/16 12:58:28 battery soc: 49%
[site  ] DEBUG 2024/04/16 12:58:28 battery power: -122W
[site  ] DEBUG 2024/04/16 12:58:28 grid meter: -0W
[site  ] DEBUG 2024/04/16 12:58:28 site power: -122W
[lp-1  ] DEBUG 2024/04/16 12:58:30 vehicle odometer: 59210km
[lp-1  ] DEBUG 2024/04/16 12:58:30 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2024/04/16 12:58:30 charger status: A
[lp-1  ] INFO 2024/04/16 12:58:30 car disconnected
[lp-1  ] DEBUG 2024/04/16 12:58:30 set charge mode: pv
[lp-1  ] DEBUG 2024/04/16 12:58:30 set session soc limit: 0
[lp-1  ] DEBUG 2024/04/16 12:58:30 set session energy limit: 0
[lp-1  ] ERROR 2024/04/16 12:58:30 vehicle soc: unknown engine type: electric
[site  ] DEBUG 2024/04/16 12:58:58 ----
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.125.0

andig, 2024-04-17:
Bitte trace log

yewbacca, 2024-04-19:
Bin auf Dienstreise gestern und heute, kann erst heute Abend Ostermorgen wieder auf EVCC zugreifen eh des trace logMit freundlichen GrüßenMartin LehnheiserGesendet mit der 1&1 Mail AppAm 19.04.24 um 09:27 schrieb andig
                
            
            
                
                    Von: "andig" ***@***.***>Datum: 19. April 2024An: "evcc-io/evcc" ***@***.***>Cc: "Mention" ***@***.***>,"yewbacca" ***@***.***>Betreff: Re: [evcc-io/evcc] Volkswagen We Connect ID: vehicle soc: unknown engine type: electric (Issue #13457)
                
                
ping @yewbacca

yewbacca, 2024-04-21:
Issue wird erstmal geschlossen, da (scheinbar nach einem Update der vw-connect api) der Fehler nicht mehr reproduzierbar ist und weder in den Online-Meldungen noch im Status auftritt. Sollte der Fehler reproduzierbar wiederkommen, erfolgt ein re-open issue mit passendem tracelog.

closed 2024-04-21 by yewbacca

VolkerK62, 2024-07-10:
hier ist ein aktueller Fall  https://github.com/evcc-io/evcc/discussions/14821