evcc-io/evcc #13463: Parameter 'batteryConfigured' and 'pvConfigured' disappeared from rest API

MikeTheTux, 2024-04-16:
### Describe the bug

openHAB evcc binding is evaluating the two parameters 'batteryConfigured' and 'pvConfigured' from rest-API.

The two parameters disappeared without a trace. Was the breaking API change done on intention?
'gridConfigured' is still present.

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
.
```

### Log details

```text
{
   "result":{
      "auth":{
         "vehicles":{
            
         }
      },
      "battery":[
         {
            "power":305.13,
            "energy":806.3646160000001,
            "soc":60.1,
            "capacity":7.68,
            "controllable":true
         }
      ],
      "batteryCapacity":7.68,
      "batteryDischargeControl":true,
      "batteryEnergy":806.3646160000001,
      "batteryMode":"normal",
      "batteryPower":305.13,
      "batterySoc":60.1,
      "bufferSoc":100,
      "bufferStartSoc":100,
      "currency":"EUR",
      "fatal":null,
      "greenShareHome":0.9862311873579613,
      "greenShareLoadpoints":0,
      "gridConfigured":true,
      "gridCurrents":[
         -0.5720000267028809,
         0.652999997138977,
         0.628000020980835
      ],
      "gridEnergy":1901.719,
      "gridPower":4.300000190734863,
      "homePower":312.30000019073486,
      "loadpoints":[
         {
            "chargeCurrent":0,
            "chargeCurrents":[
               0,
               0,
               0
            ],
            "chargeDuration":21000000000,
            "chargePower":0,
            "chargeRemainingDuration":0,
            "chargeRemainingEnergy":15506.399999999994,
            "chargeTotalImport":5332.7002,
            "chargedEnergy":3.6000000000058208,
            "chargerFeatureHeating":false,
            "chargerFeatureIntegratedDevice":false,
            "chargerIcon":null,
            "chargerPhases1p3p":false,
            "chargerPhysicalPhases":null,
            "charging":false,
            "connected":true,
            "connectedDuration":0,
            "disableThreshold":0,
            "effectiveLimitSoc":100,
            "effectiveMaxCurrent":32,
            "effectiveMinCurrent":6,
            "effectivePlanSoc":0,
            "effectivePlanTime":"0001-01-01T00:00:00Z",
            "effectivePriority":10,
            "enableThreshold":0,
            "enabled":false,
            "limitEnergy":0,
            "limitSoc":0,
            "maxCurrent":32,
            "minCurrent":6,
            "mode":"pv",
            "phaseAction":"inactive",
            "phaseRemaining":0,
            "phasesActive":2,
            "phasesConfigured":3,
            "phasesEnabled":3,
            "planEnergy":0,
            "planOverrun":0,
            "planProjectedStart":"0001-01-01T00:00:00Z",
            "planTime":"0001-01-01T00:00:00Z",
            "priority":10,
            "pvAction":"inactive",
            "pvRemaining":0,
            "sessionCo2PerKWh":null,
            "sessionEnergy":3.6000000000058208,
            "sessionPrice":0.0017524800000028335,
            "sessionPricePerKWh":0.4868,
            "sessionSolarPercentage":0,
            "smartCostActive":false,
            "title":"Wallbox",
            "vehicleClimaterActive":false,
            "vehicleDetectionActive":false,
            "vehicleLimitSoc":0,
            "vehicleName":"Q5",
            "vehicleOdometer":44395,
            "vehicleRange":0,
            "vehicleSoc":1.0229787234042924
         },
         {
            "chargeCurrent":0,
            "chargeDuration":92360000000000,
            "chargePower":0,
            "chargedEnergy":6410.496488163158,
            "chargerFeatureHeating":true,
            "chargerFeatureIntegratedDevice":true,
            "chargerIcon":"waterheater",
            "chargerPhases1p3p":false,
            "chargerPhysicalPhases":1,
            "charging":false,
            "connected":true,
            "connectedDuration":0,
            "disableThreshold":0,
            "effectiveLimitSoc":40,
            "effectiveMaxCurrent":3,
            "effectiveMinCurrent":2,
            "effectivePlanSoc":0,
            "effectivePlanTime":"0001-01-01T00:00:00Z",
            "effectivePriority":10,
            "enableThreshold":0,
            "enabled":false,
            "limitEnergy":0,
            "limitSoc":40,
            "maxCurrent":3,
            "minCurrent":2,
            "mode":"pv",
            "phaseAction":"inactive",
            "phaseRemaining":0,
            "phasesActive":1,
            "phasesConfigured":1,
            "phasesEnabled":1,
            "planEnergy":0,
            "planOverrun":0,
            "planProjectedStart":"0001-01-01T00:00:00Z",
            "planTime":"0001-01-01T00:00:00Z",
            "priority":10,
            "pvAction":"inactive",
            "pvRemaining":0,
            "sessionCo2PerKWh":null,
            "sessionEnergy":6410.496488163158,
            "sessionPrice":0.6097946665645497,
            "sessionPricePerKWh":0.09512440537024273,
            "sessionSolarPercentage":96.75780499746966,
            "smartCostActive":false,
            "title":"Heatpump",
            "vehicleClimaterActive":null,
            "vehicleDetectionActive":false,
            "vehicleLimitSoc":0,
            "vehicleName":"heatpump",
            "vehicleOdometer":0,
            "vehicleRange":0,
            "vehicleSoc":60
         }
      ],
      "prioritySoc":100,
      "pv":[
         {
            "power":2.87
         }
      ],
      "pvEnergy":0,
      "pvPower":2.87,
      "residualPower":100,
      "siteTitle":"home",
      "smartCostType":"pricestatic",
      "sponsor":"...",
      "sponsorTokenExpires":0,
      "statistics":{
         "30d":{
            "avgCo2":0,
            "avgPrice":0.14893148173909884,
            "chargedKWh":208.50557401394784,
            "solarPercentage":83.4655430486416
         },
         "365d":{
            "avgCo2":0,
            "avgPrice":0.26975229340091245,
            "chargedKWh":575.1409692579458,
            "solarPercentage":53.61850459463629
         },
         "total":{
            "avgCo2":0,
            "avgPrice":0.26975229340091245,
            "chargedKWh":575.1409692579458,
            "solarPercentage":53.61850459463629
         }
      },
      "tariffFeedIn":0.082,
      "tariffGrid":0.4868,
      "tariffPriceHome":0.08757361535749725,
      "tariffPriceLoadpoints":0.4868,
      "vehicles":{
         "Q5":{
            "title":"Audi Q5",
            "icon":"car",
            "capacity":14.1,
            "limitSoc":100
         },
         "heatpump":{
            "title":"Heatpump",
            "icon":"waterheater"
         }
      },
      "version":"0.125.0"
   }
}
```

### What type of operating system are you running?

Docker container

### Version

0.125.0

naltatis, 2024-04-17:
Yes, they were deleted intentionally. You can get the same info by checking if `pv` and `battery` has entries.

As of now, `/api/state` structure may be subject to change from release to release. This might change once we've released 1.0.

closed 2024-04-17 by naltatis

MikeTheTux, 2024-04-17:
@naltatis, thanks for confirming my observation and my intended change within the openHAB evcc binding.

Is a similar change foreseen for `gridConfigured`?

naltatis, 2024-04-17:
@MikeTheTux yes, very likely. Honestly I'm surprised, that we didn't delete it when we cleaned up pv and battery. As far I know `gridConfigured` is currently unused by evcc ui.

MikeTheTux, 2024-04-20:
References:

- https://github.com/evcc-io/evcc/pull/12498
- https://github.com/evcc-io/evcc/issues/13366