evcc-io/evcc #13464: [UI] Seiten mit Auth werden nicht angezeigt (nightly)

mdkeil, 2024-04-16:
### Describe the bug

Durch Auth geschützte Menüteile werden nach erfolgter Passworteingabe nicht geöffnet. Es passiert einfach nichts und ich kann auch nicht erkennen. Welche Logeinstellungen sind notwendig, um den Fehler einzugrenzen?

Wo habe ich es ausprobiert?
- EVCC-Seite im Chromebrowser auf dem Smartphone (Android)
- EVCC-Seite eingebettet in einem iframe in Homeassistent (Android App)

### Steps to reproduce

1. Aktuelle Nightly installieren
2. Bei Aufforderung Passwort vergeben
3. Geschütze UI-Seite (z.B. Konfiguration) öffnen 
4. Passwort eingeben 

### Configuration details

```yaml
nicht notwendig
```

### Log details

```text
nicht vorhanden
```

### What type of operating system are you running?

Docker container

### Version

0.125.0 (39db91c9) nightly

TobiasHuber1980, 2024-04-16:
war bei mir auch so, reboot hat geholfen..... jetzt gehts....

mdkeil, 2024-04-16:
Reboot vom Docker Container hat das Problem bei mir nicht behoben.

TobiasHuber1980, 2024-04-16:
Bei mir hat es anfangs mit Chrome nicht funktioniert. Dann mit Edge getestet, war ok. 
Danach (nach reboot) mit Chrome auch ok. 
Teste mal anderen Browser.

naltatis, 2024-04-17:
Kannst du mal einen Screenshot vom Netzwer- und Konsole-Tab der Browserdevtools posten?

mdkeil, 2024-04-17:
Keine Ahnung, was dort interessant ist.. musste auch vorab den kiwi-Browser installieren, um überhaupt dev-toolz in Android zu haben. Ich kann mir vorstellen, das es damit zu tun hat, das ich die UI via nginx-proxy + Let's Enrypt Zertifikat extern zur Verfügung stelle.

Wenn ich lokal auf die UI zugreife, funktioniert nach Auth auch alles wie erwartet.

[image]
[image]

naltatis, 2024-04-17:
> das ich die evcc-UI via nginx-proxy + Let's Enrypt Zertifikat extern zur Verfügung stelle.

Ist da dann auch noch Basic-Auth zwischen?

mdkeil, 2024-04-17:
lokaler Zugriff ist ohne Basic-Auth-- läuft auch nicht über den proxy. Von extern ist aber via nginx noch die basic-auth (user + pass) vor geschalten.

naltatis, 2024-04-17:
Konnte es nachstellen und fixen. Das Problem war, dass wir den Header, der durch das Basic Auth gesetzt wurde, gelesen haben. Der hatte dann Vorrang gegenüber dem Cookie.

closed 2024-04-18 by naltatis