evcc-io/evcc #13465: EVVC Gui Do not Start

kaschie2, 2024-04-16:
### Describe the bug

before 5 Days all was ok. But now the Addon start   Error: "The add-on is starting, this can take some time..." the wohle time. 

Wenn i Connect my Car it will be Charge with the Last Settings. 

The Logs say nothing about Errors. 

Someone an Iedea?

Sorry for my bad Englisch

### Steps to reproduce

1. Restart 
2.
3.
...

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: XXX

interval: 30s # control cycle interval

sponsortoken: XXX

modbusproxy:
  - port: 5200
    uri: localhost:502
    readonly: true
# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554

mqtt:
  broker: 192.168.178.56:1883
  topic: evcc/site/My_home # root topic for publishing, set empty to disable
  user: XXX
  password: XXX
  
telemetry: false

meters:
- type: template
  template: sungrow-hybrid 
  id: 1  
  host: localhost  
  port: 502  
  usage: grid  
  modbus: tcpip  
  name: grid1
- type: template
  template: sungrow-hybrid 
  id: 1  
  host: localhost  
  port: 502  
  usage: pv  
  modbus: tcpip  
  name: pv2
- type: template
  template: sungrow-hybrid 
  id: 1  
  host: localhost  
  port: 502  
  usage: battery  
  modbus: tcpip  
  name: battery
- type: template
  template: fronius-solarapi-v1 
  usage: pv  
  host: 192.168.178.31  
  name: pv4

chargers:
- type: template
  template: phoenix-ev-eth 
  id: 255  
  host: 192.168.178.90  
  port: 502  
  modbus: tcpip  
  name: wallbox6

vehicles:
- type: template
  template: vw 
  title: ID4  
  user: XXX
  password: XXX
  vin: XXX  
  capacity: 55  
  timeout: 30s  
  name: ev5

loadpoints:
- title: Garage
  charger: wallbox6
  mode: pv
site:
  title: My_home
  meters:
    grid: grid1
    pv:
    - pv2
    - pv4
    battery:
    - battery
  residualPower: 100

tariffs:
  currency: EUR # (default EUR)
  grid:
    # static grid price with price zones)
    type: fixed
    price: 0.2785 # EUR/kWh
  feedin:
    # rate for feeding excess (pv) energy to the grid
    type: fixed
    price: 0.10 # [currency]/kWh
```

### Log details

```text
i dont Know how i run this in Home Assistant
```

### What type of operating system are you running?

HomeAssistant Add-on

### Version

0.125.0

mdkeil, 2024-04-16:
You can try to check the Home Assistent system log.

mdkeil, 2024-04-16:
I dont mean the addon logs.. Look at Settings/System/Logs

closed 2024-04-17 by andig