evcc-io/evcc #13474: Grid type octpusenergy key is called "tariff" but only accepts "product"

teun95, 2024-04-17:
### Describe the bug

The grid type `octopusenergy` has the key `tariff`, but when the tariff code is used an error is displayed in the log. When the product code is used, the configuration works without issues.

Octopus Energy has tariff codes and product codes, which conflicts with how the terminology is used in EVCC. If Octopus Energy  would not have had tariff codes, users would not have been confused and would use product codes as value for `tariff`. However, given that looking up a Octopus Energy product code will also expose the user to the tariff code, the error is the result of expected configuration changes.

For clarification, this is from the [API reference of Octopus Energy](https://developer.octopus.energy/docs/api/#list-tariff-charges).

```
product_code

    The code of the product to be retrieved, for example VAR-17-01-11.
tariff_code

    The code of the tariff to be retrieved, for example E-1R-VAR-17-01-11-A.
```

As you can see, the tariff code is made up of the product code with some additional stuff. I believe that EVCC actually obtains the tariff code using the product code.

### Steps to reproduce

1. Configure octopusenergy as grid type
2. Use the tariff code `E-1R-COSY-22-12-08-A` as value for `tariff`

### Configuration details

```yaml
tariffs:
  currency: GBP
  grid:
    type: octopusenergy
    tariff: E-1R-COSY-22-12-08-A # Tariff code
    region: A # optional
```

### Log details

```text
**Log output when using `E-1R-COSY-22-12-08-A` as value for `tariff`:**

[octopus] ERROR 2024/04/17 16:01:59 unexpected status: 404 (Not Found)
[main  ] ERROR 2024/04/17 16:01:59 failed configuring grid tariff: cannot create tariff type 'octopusenergy': unexpected status: 404 (Not Found)
```
```

### What type of operating system are you running?

HomeAssistant Add-on

### Version

0.125.0

teun95, 2024-04-17:
I think there are two options for a fix:

1. Clarify in the documentation that `tariff` in the case of Octopus Energy refers to the product code. I think this is undesirable as it is more of a workaround.
2. Add an additional more specific key called `tariffcode` or `productcode`and update [/tariff/octopus/api.go](https://github.com/evcc-io/evcc/blob/9d8e2b19d54373e51eb9bf5d18a2696670cf71b8/tariff/octopus/api.go) to reflect this. Tariff can then be deprecated in time. I do not have enough knowledge of the octopus energy api to know whether it makes most sense to ask for the tariffcode or productcode.
3. Add a key for the Octopus Energy api token and obtain the active tariff code in the background. This is more work, but provides a better user experience as the API key can relatively easily be found in the Octopus Energy account details whereas finding the product or tariff details is more involved.

andig, 2024-04-17:
Is this similar to https://github.com/evcc-io/evcc/pull/11555?

duckfullstop, 2024-04-17:
It is - #11555 provides option 3

Tariff and product codes are mightily confusing to pick apart because they're referred to by similar phrases in different places - sorry for any confusion!

I'd suggest we could catch this during parsing of the code (and act appropriately), but option 2 is probably a better move for long term codebase tidiness (alongside option 3)

andig, 2024-04-26:
@duckfullstop would you want to follow up with the PR? Otherwise I'd close here as wontfix.

duckfullstop, 2024-04-26:
I'm happy to provide a fix (life permitting), but I'm not sure whether it makes more sense to add it as part of #11555, or as a separate thing?

andig, 2024-04-26:
As you prefer, no strong opinion here.

duckfullstop, 2024-04-26:
I have pushed a non-breaking change from `tariff` to `productcode` to the branch, ready for review once it's unstaled (unless I broke #11555?)

PR #13637 (2024-04-30): tariff/octopusenergy: Refactor / Support API Keys for tariff data lookup
This PR adds support for the apikey option on the octopusenergy tariff provider. Using an API key allows for automatic detection of the user's relevant tariff, and also allows for some potentially exciting functionality down the road such as user-specific tariff (and load shedding?) flexibility.

Splits a new GraphQL provider and Rest out into separate packages. Rest is still required even with GraphQL being available as GraphQL has a lot of restricted endpoints / retains backwards compatibility.

Changes the `tariff` con [... cut]
closed 2024-04-30 by andig