evcc-io/evcc #13476: Login Tapo P110 schlägt fehl

dominikruffing, 2024-04-17:
### Describe the bug

Hallo, seit dem aktuellsten Update bekommen ich dauerhaft die Fehlermeldung:
`charge meter: request failed: expected 200 OK, got 403 Forbidden. Error message:` für meine Steckdose Tapo P110.
Vor dem Update ging alles. Die Steckdose ist über die Tapo-App auch erreichbar und funktioniert, Zugangsdaten passen auch.

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
chargers:
  - name: tplink
    type: template
    template: tapo
    host: 
    user: 
    password: 
    standbypower: 15
    integrateddevice: true
```

### Log details

```text
charge meter: request failed: expected 200 OK, got 403 Forbidden. Error message:
```

### What type of operating system are you running?

HomeAssistant Add-on

### Version

0.125.0

andig, 2024-04-17:
Bei Tapo hat sich nichts geändert.

closed 2024-04-17 by andig

dominikruffing, 2024-04-18:
Und bei evcc, weil seit das Update auf 125 gemacht wurde geht es nicht
mehr, davor war alles super.

thierolm, 2024-04-18:
ein Trace-Log wäre fein ...

dominikruffing, 2024-04-20:
Trage-Log meinst du den normalen Log aus dem EVCC Add-On?