evcc-io/evcc #13479: Datenbank-Pfad evcc.dB (zumindest) unter Linux nicht fest gesetzt

MrIngenieur, 2024-04-18:
### Describe the bug

Datenbank bei start als Service ist eine andere als mit sudo evcc.
Token für Fahrzeugintegrationen werden nicht korrekt aus der evcc.yaml in die tatsächlich verwendete evcc.db übernommen, wenn EVCC als Service gestartet wird. 
Dies führte mit 0.125.0 dazu, dass bestehende, funktionierende Mercedes-Integrationen nicht mehr funktionierten und kurzfristig nur mit Neuinstallation EVCC behoben werden konnten. Mittlerweile gibt es eine Reparatur-Anleitung, aber besser wäre, wenn das Problem im Vorfeld systematisch ausgeschlossen wird. 
Wenn ich das richtig verstehe, dann müsste beim YAML auslesen "nur" darauf geachtet werden, dass die Token in den korrekten evcc.db-Pfad kopiert werden und dieser Pfad auch per systemd-Aufruf von evcc als Service verwendet wird, korrekt?
Nicht jeder möchte seine Ladehistorie einfach so wegwerfen (bei Neuinstallation) und auch nicht jeder ist so bewandert, dass er einfach irgendwelche Datenbank-Dateien rauslöscht. -> Nutzerfreundlichkeit. 

### Steps to reproduce

#13386 
Vorschlag von @ReneNulschDE als Zitat:
„ Man kann diskutieren ob der aktuelle Weg von evcc optimal ist wie der default-db pfad ermittelt wird. Aktuell ist es der Homepath des aktuellen users + /evcc/evcc.db. Und dieser ist bei Service/sudo/user Aufruf halt unterschiedlich je nach Umgebung. Meiner Meinung wäre es besser die DB im Pfad der Config Datei zu managen, wenn kein Pfad in der config definiert ist ….“

### Configuration details

```yaml
Folgende Pfade für evcc.db sind mir bekannt und je nach dem wie evcc gestartet wird, entscheidet darüber welche .db verwendet wird:

/root/.evcc/evcc.db 
/var/lib/evcc/evcc.db
/home/pi/.evcc/evcc.d
```

### Log details

```text
Token not valid Ausgabe obwohl aus dem Fahrzeug-Backend aktuelle, korrekte Live-Daten kommen. 
Aufruf hier über „sudo evcc vehicle“

[main  ] INFO 2024/04/10 23:49:29 evcc 0.125.0
[main  ] INFO 2024/04/10 23:49:29 using config file: /etc/evcc.yaml
[db    ] INFO 2024/04/10 23:49:29 using sqlite database: /root/.evcc/evcc.db
[mercedes] DEBUG 2024/04/10 23:49:29 identity.NewIdentity - token not valid - Add expiry
[mercedes] DEBUG 2024/04/10 23:49:29 identity.NewIdentity - token not valid - database token check started
Soc:           74%
Capacity:      90.0kWh
Charge status: B
Range:         248km
Odometer:      54237km
Finish time:   2024-04-11 00:00:00 +0200 CEST
Position:      0,0
Features:      []
```

### What type of operating system are you running?

Linux

### Version

0.125.0

andig, 2024-04-18:
> Token für Fahrzeugintegrationen werden nicht korrekt aus der evcc.yaml in die tatsächlich verwendete evcc.db übernommen, wenn EVCC als Service gestartet wird.

Das stimmt so nicht. Die landen dann in einer anderen DB. Auch diese wird tatsächlich verwendet, aber eben in anderem Kontext.

> Dies führte mit 0.125.0 dazu, dass bestehende, funktionierende Mercedes-Integrationen nicht mehr funktionierten und kurzfristig nur mit Neuinstallation EVCC behoben werden konnten.

Auch das ist nicht korrekt. Eine Neuinstallation ändert daran gar nichts. Richtig wäre die Angabe der korrekten DB- entweder in der yaml oder auf der Kommandozeile.

> Meiner Meinung wäre es besser die DB im Pfad der Config Datei zu managen, wenn kein Pfad in der config definiert ist 

Interessanter Vorschlag, entspricht siw aber nicht dem LFS?

MrIngenieur, 2024-04-18:
@andig ich kann dir nur sagen, dass momentan der Token Übertrag von .yaml zu .db nicht in allen Fällen funktioniert und das nicht nur bei mir. Von Mercedes-Fahrzeugen kenne ich es selbst, in #13386 wurde ähnliches auch für Tesla-Fzge berichtet. 

Du kennst dich in den technischen Zusammenhängen besser aus und das ist auch nicht mein Anspruch als einfacher User und Laie, der helfen will das tolle Projekte weiter zu optimieren. 
In #13386 geht @ReneNulschDE ja auch im Detail darauf ein, der ist technisch deutlich tiefer drin als ich. 

Von mir aus schließt die Issue, habs nur gut gemeint um bestehende Hickups langfristig zu lösen, kann mir aber die Lebenszeit für das erstellen eines Issues auch sparen. 
Als einfacher User und Sponsor wünsche ich mir natürlich, dass EVCC als tolles Projekt weiter gedeiht. 
Der Umgang mit konstruktiv gemeinter Kritik, gerade von weniger technisch versierten Usern, sollte aber vielleicht nochmal hinterfragt werden. Wenn man jedesmal als User erstmal aufgezeigt bekommt, was man alles falsch verstanden hat und auf deutsch gesagt "zu doof ist das Tool richtig zu bedienen", verliert man irgendwann auch die Lust sich konstruktiv an der Weiterentwicklung zu beteiligen. 
Wenn EVCC halt auch bei normalen Usern ankommen soll, sollte man das Feedback dieser Usergruppe auch ernst nehmen und genauso wertschätzen wie die User eure tolle Arbeit wertschätzen! 
Langfristig kann man aus meiner Sicht nur so dem Anspruch der evcc-Homepage gerecht werden: "Wir möchten, dass PV-Strom-Laden für alle einfach möglich wird. Im Optimalfall funktioniert das ganz ohne manuelles Regeln und Einstellen"

ReneNulschDE, 2024-04-18:
@MrIngenieur : Öffne das bitte nochmal. Ich kann Deiner Argumentation folgen. Aber nicht deiner Art und Weise, speziell wenn Du von Gefrickel und Balkonkraftwerks-Usern sprichst, ohne Respekt für die Menschen und deren Lebenszeit, wobei mache dies hier for free erbringen. (Solltest Du Dich nicht mehr erinnern, suche nach deinen letzten Posts in dieser Umgebung.)

@andig: Das mit den unterschiedlichen Datenbanken ist wirklich ein Problem. Alle Fehlerreporte in Bezug auf die Mercedes Komponente nach v124.7 waren bezogen auf dieses "Nichtverständnis" von User-Kontexten"/"Datenbank"-Pfaden oder das ein Testen mit `sudo evcc vehicle` den richtigen neuen Refreshtoken in die "falsche" DB schreibt und damit es bei Service Start nicht mehr funktionieren. Wenn Ihr keinen Breaking Change im DB-Verhalten wollt, dann brauchen wir ein anderes Key-Store Konzept für Integrationen mit wechselnden Refreshkeys. HA hatte das gleiche Problem am Anfang, und da habe ich die Keys im Config Folder als extra File gespeichert. Wenn ich mich richtig erinnere, hatten wir das Problem auch während meiner Entwicklung besprochen und uns für die "offizielle" settings API lösung entschieden.

MrIngenieur, 2024-04-18:
@ReneNulschDE da musst du mich falsch verstanden haben, ich hab überhaupt nichts gegen Balkonkraftwerks-User, im Gegenteil, ich bin über jeden froh der sich damit beschäftigt. 

Ich meinte "BALKONE" im Sinne von Workarounds mit vielen Schnittstellen in andere Teilsystemen, wo man ein gleiches Ergebnis mit weniger Fehlerpotential auch direkt erhalten könnte. 
Konkret: Signalkette HomeAssistent -> MQTT -> generischer Vehicle Adapter in Yaml -> Signal-Einzelverknüpfung -> EVCC. 

Mein Ansinnen war: Nutze gerade als unerfahrener User doch lieber die direkte Mercedes-Integration, bevor du diese Schnittstellekette aufbaust und im Falle von Updates etc. auch immer wieder am leben halten musst. Für erfahrene Benutzer geht das. Ich nutze das auch über iobroker und Rest-API, aber mir ging es dort und auch hier halt darum, dass man den "Otto-Normal-Nutzer" nicht zu irren Klimmzügen motivieren sollte, sondern das Nutzererlebnis möglichst reibungslos sein soll. Wenn die Sonne scheint, genieße ich lieber das Wetter und lasse das Auto vollautomatisiert laden, als dass ich MQTT Streams von einem System zum anderen nachverfolge und debugge ;-)

MrIngenieur, 2024-04-18:
> @MrIngenieur : und nochmal zum Verständnis der Logausgabe weiter oben: Die Komponente macht einen mehrstufigen Token-Check und wenn immer ein Fehler auftritt kommt ein Logeintrag.
> 
> 1. Token aus config lesen
> 2. Dem Token eine Ende zuweisen
> 3. Prüfen ob ein Token in der DB ist und wenn ja, dann den Token aus der Config mit dem aktuellen Token überschreiben
> 4. Neuer Laufzeit Check
> 5. Wenn False, hole einen neuen Token
>    In deinem Logbeispiel ist alles ok (im Aufruf-Kontext), der richtige Token ist in der DB `/root/.evcc/evcc.db`  da der Check DB einen gültigen Token enthält.
> 
> Das der gültige Token in der DB gelandet ist, deutet darauf hin, dass Du nach dem Erstellen des Tokens, nochmal `sudo evcc vehicle` oder `sudo evcc` anstatt den Service neu zustarten. (und nein, dass ist keine Fehlerzuweisung nur eine Erklärung)

Verstanden, Danke! Jetzt weiß ich auch was mit „Bedienfehler“ gemeint ist. Wenn ich mich richtig erinnere, steht der Aufruf halt in der Doku ganz oben im allgemeinen Teil so drin um zu testen ob alles läuft. Vielleicht hilft auch schon der Hinweis bei der Mercedes Integration, dass man genau das NICHT machen soll ;-) 
[image]

ReneNulschDE, 2024-04-18:
@MrIngenieur : da musst du mich [hier](https://github.com/evcc-io/evcc/discussions/13277#discussioncomment-9063799) falsch verstanden haben, ich hab überhaupt nichts gegen

Alles gut, Haken dran. Das musste einfach raus... Zurück zum Thema :-)

MrIngenieur, 2024-04-18:
> * Warnung wenn in der `evcc.yaml` keine DB hinterlegt ist.
> * Error wenn in der `evcc.yaml` keine DB hinterlegt ist.
Nochmal kurz mit der User-Brille drauf geschaut:

- Wenn ich mich richtig erinnere habe ich meine Initialbefüllung/Erstellung der YAML mit dem Assistenten/Wizard gemacht. 
- In dem Prozess bin ich nicht beim Thema Datenbank vorbei gekommen bzw. damit konfrontiert worden, was ich erstmal als „gut“ bewerten würde. 
- ich habe auch in der Anleitung zur Installation/Setup nicht gelesen, dass ich eine Datenbank definieren muss, was mir per se als User auch recht ist, wenn ich mich nicht drum kümmern muss. 
- …und vielleicht kapier ich als User auch nicht wofür ich die überhaupt brauche, wo die herkommt und wieso ich die hinterlegen muss ;-)

Bei 1) und 2) müsste man dem User vermutlich sehr genau erklären was nicht passt und warum er ggf. etwas unternehmen sollte. Mit „Datenbank fehlt“ als Hinweistext steht man erstmal wie der Ochs vorm Berg, wenn man gar nicht wusste dass man eine „braucht“ im Sinne hinterlegen muss. 

zu 3) In der Anleitung wird ja auch beschrieben, dass man die generierte yaml mit `sudo mv evcc.yaml /etc ` nach /etc verschieben soll. Damit wäre das aus meiner Sicht der Standard-Fall und wenn ich Linux richtig verstehe findet jeder User, egal ob Sudo, Root, Pi oder per Service unter `/var/lib/evcc/evcc.db ` immer den gleichen Inhalt und kann auch darauf zugreifen, was bei `/root/.evcc/evcc.db  ` ja z.B. für den User Pi erstmal nicht pauschal der Fall ist, auch wenn er sich den Zugriff in der Standardkonfig von Raspi-OS leicht selbst gewähren kann, aber das ist ein anderes Thema.

MrIngenieur, 2024-04-18:
> > Ja, ich habe das drüber auch schon ein bisschen nachgedacht und mir fallen drei Lösungen ein:
> 
> Spontan wäre ich für 3., aber die Auswirkungen sind wahrscheinlich größer und annehmend, dass die Userrate MB cars bei <1% ist (HA hat ca. >800k User mit <8k MB Usern), passt 1. Dazu sollte der Hilfetext bei der MB Komponente noch angepasst werden. Ich schau mir das an. Vorschläge gern hier.

Bzgl. Hilfetext: 
Hier kann man vielleicht auch darüber nachdenken die Reparaturanleitung mit zu hinterlegen, falls doch mal was kaputt geht: 
https://github.com/evcc-io/evcc/discussions/13331#discussioncomment-9145538

andig, 2024-04-18:
Weitere Optionen:

- wir splitten auch die yaml, z.B. indem evcc für den Anwender nicht mehr in /etc nachschaut 
- wir verbieten, bestimmte Fahrzeuge wie Mercedes und Tesla über die yaml zu konfigurieren
- evcc startet nicht mehr wenn es eine zweite Datenbank am Ort des Services findet und keine User-DB explizit konfiguriert ist

All diese Probleme verschwinden, wenn wir yaml-Konfiguration abschaffen. Dann wird man sich als Anwender aber auch wundern, wenn ein `evcc vehicle` plötzlich keine Fahrzeuge mehr findet. Das Problem ist dann nur offensichtlicher...

Vielleicht müssen wir das jetzt auch einfach mal aushalten...

andig, 2024-04-18:
> @andig ich kann dir nur sagen, dass momentan der Token Übertrag von .yaml zu .db nicht in allen Fällen funktioniert und das nicht nur bei mir. Von Mercedes-Fahrzeugen kenne ich es selbst, in https://github.com/evcc-io/evcc/issues/13386 wurde ähnliches auch für Tesla-Fzge berichtet.

@MrIngenieur ich denke wir sind über die Stelle hinaus, aber wie oben geschrieben funktioniert der zu 100%, nur eben nicht so wie Du das erwartest ;)

naltatis, 2024-04-18:
> Vielleicht müssen wir das jetzt auch einfach mal aushalten...

Ja, Prio ist die yaml los zu werden.

DB Backup/Restore evtl. sogar selektiv (Session, Konfiguration, ...) ist etwas über das wir dann eher nachdenken sollten.

ReneNulschDE, 2024-04-18:
Denkt aber bitte an den 2-Step approach (Account Eingabe und dann im zweiten Schritt die Pin) im onboarding. Das war der Grund warum ich überhaupt den evcc token approach gewählt hatte.

MrIngenieur, 2024-04-18:
> * wir verbieten, bestimmte Fahrzeuge wie Mercedes und Tesla über die yaml zu konfigurieren

Wie kann ich diese Fahrzeuge dann konfigurieren? 
Hatte mir nie gewünscht das per Yaml machen zu „dürfen“. 
In der Anleitung kann ich keinen anderen Weg finden.

MrIngenieur, 2024-04-18:
> nur eben nicht so wie Du das erwartest ;)

Weil die Anleitung was anderes suggeriert: https://github.com/evcc-io/evcc/issues/13479#issuecomment-2063308867

ReneNulschDE, 2024-04-18:
Du kannst aktuell es nicht im Frontend konfigurieren, da das Frontend die Funktionalität nicht hat. Hier geht es um Zukunft.

MrIngenieur, 2024-04-18:
Habe ich vermutet, aber die Konfiguration von Tesla Fahrzeugen zu verbieten wäre ja erstmal ein herber Dämpfer für das Projekt.. hoffentlich kommt die Zukunft zeitnah 😉

ReneNulschDE, 2024-04-18:
Darum geht es doch hier auch nicht, die Tesla Komponente kann doch schon über das Frontend konfiguriert werden, da es eine eigene [Webapp](https://docs.evcc.io/docs/devices/vehicles#tesla) für die Tokengenerierung gibt.

MrIngenieur, 2024-04-18:
> Darum geht es doch hier auch nicht, die Tesla Komponente kann doch schon über das Frontend konfiguriert werden, da es eine eigene [Webapp](https://docs.evcc.io/docs/devices/vehicles#tesla) für die Tokengenerierung gibt.

Für die Tokengenerierung schon, aber der muss ja auch in die YAML rein, oder?

MrIngenieur, 2024-04-18:
> Aber ich lese aus den Kommentaren, dass das Core Team dies nicht fokussieren will.

Also komm dann lass uns das hier jetzt beenden. 
Ich habe für **mich** ohnehin längst eine Lösung über die generische Fzg-Konfiguration gefunden. 
Die Issue war eher zur Verbesserung des Verhalten für die Allgemeinheit gedacht. 
Wenn das nicht im Sinne des CoreTeams ist, dann bringts auch nichts. 
vielleicht kann man wenigstens den Hilfetext noch ergänzen, damit nicht noch jemand den gleichen „Bedienfehler“ wie ich macht. 
wenn bis morgen kein Veto kommt, mach ich hier zu.

andig, 2024-04-19:
> Wenn Ihr Input haben wollt, wie es z.B. HA gelöst hat, kann ich gern helfen.

@MrIngenieur Gerne!

andig, 2024-04-19:
> Für die Tokengenerierung schon, aber der muss ja auch in die YAML rein, oder?

Fahrzeuge können seit diversen Releases über das UI angelegt werden.

ReneNulschDE, 2024-04-19:
Aktiviere die experimentellen Funktionen in den Optionen und dann hast Du eine neue Welt an Config Optionen. Mercedes wird explicit nicht unterstützt aktuell (Da die Tokengenerierung einen 2-step approach benötigt).

MrIngenieur, 2024-04-19:
Hab da schonmal rein geschaut, da aber nirgends was dokumentiert steht, hab ich’s produktiv nicht genutzt. Ohne Mercedes-Integration bringt mir die neue Lösung eh nix.

StefanSchoof, 2024-04-19:
Können wir uns bitte um das Thema aus dem Titel konzentrieren und für die anderen Themen eine extra Diskussion eröffnen. Das wird hier unübersichtlich.

andig, 2024-04-21:
@naltatis ich glaube das Kernproblem hier ist gar nicht wo wir die Tokens speichern, sondern wie wir sie erhalten. Das sollte ja idealerweise über das UI passieren- Oauth Login.

@ReneNulschDE bei Tesla ginge das- wie siehts bei Mercedes aus?

closed 2024-05-04 by andig