evcc-io/evcc #13485: Energy-Charts-API

sichtblock144, 2024-04-18:
**Is your feature request related to a problem? Please describe.**
Hi, GPE (Green Planet Energy) bietet seit neustem einen dynamischen Stromtarif an. Clou: Man zahlt den Börsenpreis, GPE kauft aber direkt bei Erzeugern von Wasser- und Windenergie an. Direkthandel und so.. -> https://green-planet-energy.de/privatkunden/dynamischer-stromtarif
Sie bieten zwar keine eigene API, allerdings könne man die nötigen Daten über die "Energy-Charts-API" des Fraunhofer-Instituts abrufen -> https://api.energy-charts.info/#/prices/day_ahead_price_price_get

**Describe the solution you'd like**
Es wäre toll, könnte man den Abruf so in EVCC integrieren oder die Doku dahingehend ergänzen, dass noobs wie ich das einrichten können. 

**Describe alternatives you've considered**
-

**Additional context**
FAQ bei GPE "Steht für den dynamischen Stromtarif eine API (Programmierschnittstelle) zur Verfügung?
Aktuell stellen wir für unseren dynamischen Stromtarif keine eigene API zur Verfügung. Wenn Sie Ihr Smart Home oder Ihr eigenes Energiemanagementsystem (EMS) einbinden möchten, können Sie bereits frei verfügbare Schnittstellen nutzen, die den Day-Ahead-Strompreis abbilden. Eine solche Option ist beispielsweise die [Energy-Charts-API](https://api.energy-charts.info/) des Fraunhofer-Instituts. Diese ermöglicht Ihnen den Zugriff auf aktuelle Strompreisdaten und erleichtert die Integration in Ihr System."

andig, 2024-04-18:
Mir ist nicht klar, warum es Fraunhofer braucht wenn wir Entso-E schon haben? Sollte aber auch kein Problem sein, mittels https://github.com/evcc-io/evcc/pull/13451 kannst Du Dir das selbst bauen. Ein PR für ein Template wäre Klasse, dann wirds für den nächsten Anwender einfacher. Greenplanet kann ich auch gerne an uns wenden, dann übernehmen wir die Umsetzung.

andig, 2024-04-18:
/cc @mdkeil eine schöne Übung für JQ mit Preisen und Timestamps in unterschiedlichen Arrays (facepalm).

closed 2024-04-18 by andig

mdkeil, 2024-04-18:
Ich schau mir das nachher vielleicht mal an.

mdkeil, 2024-04-18:
@sichtblock144 
..kurz reingeschaut und für nicht sinnvoll gehalten.. der gleiche Informationsgehalt wie Entso-E Implementierung, zumal das auch keine Aussage darüber trifft, wie am Ende nun dein persönlicher Preis inkl. Gebühren und Co. aussieht. Das ist rein der Börsenpreis.

andig, 2024-04-18:
Allerdings einfacher benutzbar als Entso und die Gebühren kommen ja einfach in die evcc Config.

sichtblock144, 2024-04-19:
ok, danke

andig, 2024-05-02:
@mdkeil ich bin gerade über https://api.allinpower.nl/troodon/api/p/spot_market/prices/?product_type=ELK gestolpert und habe für das "zip" Problem eine Lösung gefunden:

    jq '[.timestamps, .prices] | transpose  | map({ "timestamp": (.[0]), "price": .[1]})'

mdkeil, 2024-05-02:
da sieht mal wieder, das jq recht mächtig ist-- wenn man sich mal ein wenig damit auseinandergesetzt hat. Somit sollten in Zukunft mit deinem tariff-template einfach neue Anbieter hinzugefügt werden können.