evcc-io/evcc #13496: Dynamische Batterie-Einstellung

michiproep, 2024-04-19:
Hallo zusammen,
das Feature hier löse ich aktuell via Skript, aber es wäre schon cool wenn es direkt integrierbar wäre.
Zum Usecase:
Ich habe 4 Ladepunkte (1 Wallbox, 2 Wärmepumpen und 1 SmartPlug). Das Ziel ist, dass ich so früh wie möglich zum Laden anfangen möchte aber Abends der Speicher doch voll sein sollte, so dass es über Nacht für die Heizung reicht.
Ich stelle im Moment mogens auf 25%/100% und 3 Stunden vor Sonnenuntergang auf (zB) 95%/100%

Wäre es denkbar, dass evcc selbstständig im Laufe des Tages die Batterielevel hoch und runterschraubt?

[image]

andig, 2024-04-19:
Das klingt mir arg spezifisch. Spricht etwas dagegen, diese kundenspezifische Logik außerhalb mit Hilfe der evcc Schnittsstellen umzusetzen?

naltatis, 2024-04-19:
Ich kann den Bedarf durchaus nachvollziehen. Und ja, was da wie geregelt wird, ist sicher sehr spezifisch vom Nutzungsverhalten und der eigenen Umgebung. Haben wir aktuell keine gute Lösung für eingebaut. Geht mMn. eher in Richtung Optimierungsalgorithmus, der auf Basis von Produktionsprognose, Preisvorhersage, Verbrauchserwartung, etc. ... Entscheidungen trifft. Wenn wir diese Datenquellen sauber in der Plattform haben, könnten wir mal darüber nachdenken, ob wir explizite Batterieplanungsalgorithmen im Core mit aufnehmen. Die könnten dann standardisiert Daten bekommen und ne Entscheidung zurückliefern.
Stand heute muss (bzw. kann es ja auch) extern geregelt werden.

michiproep, 2024-04-19:
Hi @andig,
für mich persönlich ist das kein Problem. Ich kann mir das bauen.
Aber so kundenspezifisch ist das nicht. Viele meiner Bekannten sind in der Programmierung nicht so firm und haben auch keinen HomeAssistant oder iobroker etc. am laufen. Die machen das schon manuell. Man muss dazu sagen, dass die dann auch keine Wärmepumpensteuerung haben, aber den ein oder anderen SmartPlug schon.
Das Problem ist einfach, dass viele grundsätzlich möchten, dass der HausAkku voll ist für die Nacht, aber mit einer 100%/100% Einstellung die Kleingeräte nie was tun.

michiproep, 2024-04-19:
> Ich kann den Bedarf durchaus nachvollziehen. Und ja, was da wie geregelt wird, ist sicher sehr spezifisch vom Nutzungsverhalten und der eigenen Umgebung. Haben wir aktuell keine gute Lösung für [...]

Hi @naltatis ,
danke für deinen Kommentar.
So kompliziert wäre das gar nicht. Man bräuchte nur diese Settings:
Sunrise (+offset): x / y levels
Sunset (+ offset): a / b levels
evcc könnte dann bei jedem Takt einfach eine lineare Hochrechnung von x => a und y => b machen und die Werte setzen.
Die UI ist da interessanter. Die müsste eine Linie zeichnen und den aktuellen Punkt darstellen.

naltatis, 2024-04-19:
Verstehe. Bei einer rein sonnenstandsabhängigen Lösung spielen da ja immer noch Faktoren wie Jahreszeit und so rein. Also im Winter hab ich die Limits anders stehen als im Herbst und Sommer. Wenn wir das im Core angehen, muss das mMn. eine sehr flexible Lösung sein, bei der man nicht viele Parameter händisch einstellen muss. Eher den Wunschzustand angeben (Analog Abfahrtszeit+SoC) als die Regelung direkt konfigurieren. Letztendlich ist das ja eine reine Preis- oder CO2-Optimierung.

naltatis, 2024-04-19:
Dass wir den konkreten Use-Case (via Sonnenstand) einbauen sehe ich gerade nicht. Dass eine "intelligente"/dynamische Einstellung auch ohne viel manuelle Arbeit oder externen Scripten möglich sein soll (zumindest perspektivisch) sehe ich sehr wohl. Da fehlt es uns aber aktuell noch etwas an Voraussetzung (Prognose, Historie) und nem guten Konzept. Daher konvertier ich den Issue hier mal in eine Diskussion.