evcc-io/evcc #13512: Planer "verliert Phase" (7,4kW statt 11kW) - Ladevorgänge starten nicht

RustyDust, 2024-04-20:
### Describe the bug

Ich betreibe evcc mit einem 3-phasigen SonnenCharger (ex-Etrel, jetzt Landis+Gyr). In den letzten Tagen habe ich öfter festgestellt, dass der Planer anscheinend eine Phase "vergisst". Das wirkt sich dann so aus, dass in der Ladeplanung "x:yy h @**7.4kW**" statt der erwarteten "@ 11kW" steht.  

[image]

Immer wenn das passiert ist evcc auch nicht mehr in der Lage, den Ladeprozess zu starten - und zwar egal welchen, auch "PV", "Min+PV" und "Schnell" funktionieren dann nicht mehr. 

Die Lösung ist dann, auf die Einstellungen des Ladepunkts zu gehen. Dort stehen dann auch die 7.4 kW:
 
[image]

Wenn ich dann auf 1-phasig umstelle und wieder zurück auf 3-phasig passt es wieder:

[image]

Ich habe keine Idee, was das auslöst. Das Einzige, was ich in den Logs sehe ist, dass evcc jede Nacht zwischen 3:59 und 4:01 keine Verbindung zur PV-Batterie herstellen kann. Keine Ahnung ob das einen Einfluss hat.

Wenn ich da irgendwas bestimmtes debuggen soll mache ich das gerne.

### Steps to reproduce

evcc einfach eine Zeit laufen lassen, ggf. Ladeplan einstellen. Ich habe leider noch kein Schema gefunden, das das Verhalten auslöst, sorry.

### Configuration details

```yaml
network:
  schema: http
  # host: evcc.local # .local suffix announces the hostname on MDNS
  host: 
  port: 7070

# log: debug
log: info
levels:
  cache: error

database:
  dsn: "/opt/evcc/evcc.db"

interval: 30s # control cycle interval

tariffs:
  currency: EUR # (default EUR)
  grid:
    type: tibber
    token: 
  feedin:
    type: fixed
    price: 0.08 # [currency]/kWh
  co2:
    type: grünstromindex
    zip: 

meters:
- type: template
  template: sonnenbatterie
  usage: grid
  host: 1.2.3.4
  port: 8080
  name: sonnen-grid
- type: template
  template: sonnenbatterie
  usage: pv
  host: 1.2.3.4
  port: 8080
  name: sonnen-pv
- type: template
  template: sonnenbatterie
  usage: battery
  host: 1.2.3.4
  port: 8080
  name: sonnen-batterie

chargers:
- type: template
  template: etrel
  host: 192.168.27.26
  port: 502
  name: sonnencharger

loadpoints:
- title: Garage
  charger: sonnencharger
  mode: off

vehicles:
- name: niro
  type: template
  template: kia
  title: Kia Niro EV
  user: 
  password: 
  capacity: 64
  mode: pv
  icon: car

site:
  title: WhoHome
  meters:
    grid: sonnen-grid
    pv:
    - sonnen-pv
    battery:
    - sonnen-batterie
  residualPower: 100
```

### Log details

```text
Lasse ich gerade mitlaufen, folgt sobald es wieder auftritt.
```

### What type of operating system are you running?

Linux

### Version

0.125.0

VolkerK62, 2024-04-20:
Vermutung: 
Du hast vorher PV geladen und deine Wallbox misst die Phasenströme. 
Der Kia lädt bei geringen Strömen gerne nur auf 2 Phasen, obwohl 3p angeschlossen. 
Das interpretiert evcc dann als 2p und schaltet intern entsprechend um. Wenn dann die PV Ladung gestoppt wird, bleibt der Zustand erhalten, solange das Auto angesteckt ist.
Du könntest mal probieren, ob `phases: 3` am vehicle das Problem löst.

RustyDust, 2024-04-20:
> Vermutung: Du hast vorher PV geladen und deine Wallbox misst die Phasenströme. 

Ziemlich wahrscheinlich, ja.

@VolkerK62 Danke für den Hint, probiere ich gleich.

> Der Kia lädt bei geringen Strömen gerne nur auf 2 Phasen, obwohl 3p angeschlossen. 

Gut, den technischen Aspekt dabei verstehe ich.

> Das interpretiert evcc dann als 2p und schaltet intern entsprechend um. 

Das wiederum nicht. Schließlich "weiss" evcc qua Config, was für einen Charger ich habe und dass der 3p ist. Ich mein', der wird ja nicht magisch während des Ladens ausgetauscht und evcc sieht sich plötzlich mit einem neuen Charger konfrontiert, den es noch nie zuvor gesehen hat ;) Und es macht ja für evcc auch eigentlich keinen Unterschied, ob es das Auto nun auf 2 Phasen lädt oder auf 3, wovon halt gerade eine auf Wunsch des Fahrzeugs auf 0A steht. Und wenn das interne Umschalten auf 2p zwingend notwendig ist, dann könnte evcc ja nach Beendigung des auslösenden Ladevorgangs den Charger intern wieder auf dessen "Defaults" zurücksetzen - unabhängig davon, ob das Vehicle abgesteckt wurde oder nicht.

> Wenn dann die PV Ladung gestoppt wird, bleibt der Zustand erhalten, solange das Auto angesteckt ist. Du könntest mal probieren, ob `phases: 3` am vehicle das Problem löst.

Werde ich machen, ich versteh's nur rein Programm-logisch immer noch nicht. Gut, das Auto sagt in dem Fall "bitte nur 2p laden" und ich kann auch nachvollziehen, dass sowohl der Charger als auch evcc diesem Wunsch nachkommen. Was ich nicht verstehe ist, warum evcc das dann auch als "ist jetzt der Default bis zum Abstecken des Autos" übernimmt. Oder kann es den Charger an der Stelle dann nicht mehr passend zu seiner Config "resetten" wenn z.B. der aktuelle PV-Überschuss-Ladevorgang beendet ist?

Genauso "unlogisch" finde ich es dann aus Anwendersicht auch, wenn ich am _Vehicle_ `phases: 3` definieren muss. Was das Vehicle _will_ ist ja unabhängig davon, was mein Charger _real hat_. Eigentlich würde ich ja dann eher evcc mitteilen wollen, dass mein _Charger_ 3 Phasen hat - egal, was es selbst während der Ladevorgänge rausmisst ;) Aber jetzt teste ich das Setting beim Vehicle erstmal und schaue, ob es das Problem löst.

Vielen Dank nochmal für den schnellen Tipp.

VolkerK62, 2024-04-20:
Wie gesagt, ist eine Vermutung. Ohne Logfile nicht gesichert.
Bzgl Phasenerkennung und Verwendung für interne Berechnung siehe https://docs.evcc.io/docs/reference/configuration/loadpoints#phases

RustyDust, 2024-04-20:
Bin schon am loggen. Kurze Frage: sehe ich das richtig, dass durch das Setzen von `phases: 3` beim Vehicle jetzt mein PV-Überschussladen auch erst bei ~4.1 kW Überschuss beginnen würde statt wie bisher bei ~2.4 kW?

VolkerK62, 2024-04-20:
Wenn das funktioniert, dann ja.
Also manchmal möchtest du 2p (bei PV) und manchmal 3p (bei Plan)?
Dann lass es weg und nutz deinen workaround von oben,

andig, 2024-04-20:
Gerne mit Log wieder melden

RustyDust, 2024-05-07:
@andig Hier das Log, Kommentare/Umschaltpunkte sind durch Leerzeilen gekennzeichnet.

[evcc.log.tgz](https://github.com/evcc-io/evcc/files/15242060/evcc.log.tgz)

andig, 2024-05-08:
Also:

```
[lp-1  ] INFO 2024/05/07 20:12:57 car connected
[lp-1  ] DEBUG 2024/05/07 20:12:57 vehicle api refresh
[lp-1  ] DEBUG 2024/05/07 20:12:57 pv timer elapse
[lp-1  ] DEBUG 2024/05/07 20:12:57 pv timer inactive
[lp-1  ] INFO 2024/05/07 20:12:57 start charging ->
[main  ] DEBUG 2024/05/07 20:13:02 vehicle status: C (Kia Niro EV)
[lp-1  ] INFO 2024/05/07 20:13:02 vehicle updated: unknown -> Kia Niro EV
[lp-1  ] DEBUG 2024/05/07 20:13:02 set charge mode: pv
[lp-1  ] DEBUG 2024/05/07 20:13:02 vehicle status: C
[lp-1  ] WARN 2024/05/07 20:13:02 charger out of sync: expected disabled, got enabled
```

Deine WB fängt anscheinend von alleine an obwohl gar keine Ladung möglich sein sollte.

Es zeigt sich Netzbezug:

```
[lp-1  ] DEBUG 2024/05/07 20:13:02 charge power: 10304W
[lp-1  ] DEBUG 2024/05/07 20:13:02 charge currents: [15.1 15.2 15.1]A
[site  ] DEBUG 2024/05/07 20:13:02 pv power: 3W
[site  ] DEBUG 2024/05/07 20:13:02 battery soc: 98%
[site  ] DEBUG 2024/05/07 20:13:02 battery power: 8000W
[site  ] DEBUG 2024/05/07 20:13:02 grid meter: 3212W
```

> Schaltet um auf 2p. Warum so früh? Die Einstellung ist eigentlich, dass die PV-Batterie bis 90% 
zum PV-Laden genutzt werden kann. evcc hätte also auch mit 7,2kw laden können. Wäre halt die 
PV-Batterie schneller leer geworden.

Da schaltet gar nichts um sondern Dein Auto tut das von sich aus:

```
[lp-1  ] DEBUG 2024/05/07 20:13:27 charge power: 2560W
[lp-1  ] DEBUG 2024/05/07 20:13:27 charge currents: [0.512 5.86 5.66]A
```

Kia? Die haben Probleme mit 3p unter 8A. Demzufolge sieht evcc jetzt 2p. Danach wird also auch geplant -> alles richtig.

> Plan eingerichtet und nicht auf die angzeigten Phasen geachtet ... Würde es nicht mehr Sinn machen,
die Planung anhand der von der Wallbox unterstützten Phasen zu berechnen, anstatt die zu nehmen, die
als letztes als aktiv erkannt wurden? Oder zumindest bei nicht plausiblen Werten wie 2p das Maximum
der unterstützten Phasen anzunehmen?

Was nutzt Dir eine 3p WB wenn ein 1p Fahrzeug dran hängt? Deshalb verwenden wir die Erkennung.

Lösung für Dich: `mincurrent: 8`

> Immer wenn das passiert ist evcc auch nicht mehr in der Lage, den Ladeprozess zu starten - und zwar egal welchen, auch "PV", "Min+PV" und "Schnell" funktionieren dann nicht mehr.

Davon ist im Log nix zu sehen und passt auch nicht zum Ablauf oben. Anderes Problem/ neues Issue?

closed 2024-05-08 by andig