evcc-io/evcc #13516: Beim Starten von HA evcc Addon wird der Betriebsmodus des Q-Cells WR von "Backupmodus" auf "Eigenverbrauch" geändert

itwuzel, 2024-04-20:
### Describe the bug

Beim Starten des evcc Addon Version 0.125.0 HomeAssistandVersion  2024.4.3 wird der Betriebsmodus des Q-Cells WR von  "Backupmodus" auf "Eigenverbrauch" geändert.

Einstellungsproblem oder Bug?

Dadurch wird im Q-Cells System u.a. der eingestellte min. SOC für den Backupbetrieb unterschritten.

### Steps to reproduce

1.HA Solax (Qcells) Integration über WaveShare TCP/IP <> Modbus RTU Gateway (Typ: Waveshare RS485 TO POE ETH (B))
2.HA evcc  AddOn ebenfalls über den gleichen WaveShare TCP/IP <> Modbus RTU Gateway mit dem Q-Cells WR Verbunden
3. Q-Cells WR im Betriebsmodus "Backup Mode"
4. starten des Addon
5. Q-Cells WR ist jetzt im Betriebsmodus "Eigenverbrauch"

Aktuell wird evcc nur testweise ohne Wallbox betrieben.

### Configuration details

```yaml
network:
  # schema is the HTTP schema
  # setting to `https` does not enable https, it only changes the way URLs are generated
  schema: http
  # host is the hostname or IP address
  # if the host name contains a `.local` suffix, the name will be announced on MDNS
  # docker: MDNS announcements don't work. host must be set to the docker host's name.
  host: homeassistant.local
  # port is the listening port for UI and api
  # evcc will listen on all available interfaces
  port: 7070

interval: 30s # control cycle interval. Interval <30s can lead to unexpected behavior, see https://docs.evcc.io/docs/reference/configuration/interval

# database configuration for persisting charge sessions and settings
# database:
#   type: sqlite
#   dsn: <path-to-db-file>

# sponsor token enables optional features (request at https://sponsor.evcc.io)
# sponsortoken:

# telemetry enables aggregated statistics
#
# Telemetry allows collecting usage data (grid and green energy, charge power).
# Data is aggregated, no individual charging sessions are tracked. The collected,
# anonymous data can be retrieved using https://api.evcc.io.
#
# See https://github.com/evcc-io/evcc/pull/4343 or details.
#
# For time being, this is only available to sponsors, hence data is associated with
# the sponsor token's identity.
#
# telemetry: true

# log settings
# log: info
log: debug
levels:
  site: debug
  lp-1: debug
  lp-2: debug
  cache: error
  db: error

# modbus proxy for allowing external programs to reuse the evcc modbus connection
# each entry will start a proxy instance at the given port speaking Modbus TCP and
# relaying to the given modbus downstream device (either TCP or RTU, RS485 or TCP)
modbusproxy:
[... 264 lines cut ...]
[lp-1  ] INFO 2024/04/20 14:31:15   charger:     power ✗ energy ✗ currents ✗ phases ✗ wakeup ✗
[lp-1  ] INFO 2024/04/20 14:31:15   meters:      charge ✗
[site  ] DEBUG 2024/04/20 14:31:15 set battery discharge control: true
[lp-1  ] DEBUG 2024/04/20 14:31:15 phase timer inactive
[lp-1  ] DEBUG 2024/04/20 14:31:15 pv timer inactive
[lp-1  ] ERROR 2024/04/20 14:31:18 charger enabled: dial tcp 192.168.22.24:502: connect: no route to host
[site  ] DEBUG 2024/04/20 14:31:18 ----
[lp-1  ] DEBUG 2024/04/20 14:31:18 charge power: 0W
[site  ] DEBUG 2024/04/20 14:31:18 pv power: 3847W
[site  ] DEBUG 2024/04/20 14:31:18 battery soc: 100%
[site  ] DEBUG 2024/04/20 14:31:18 battery power: 0W
[site  ] DEBUG 2024/04/20 14:31:18 grid meter: -3015W
[site  ] DEBUG 2024/04/20 14:31:18 site power: -3015W
[lp-1  ] ERROR 2024/04/20 14:31:21 charger status: dial tcp 192.168.22.24:502: connect: no route to host
[site  ] DEBUG 2024/04/20 14:31:21 set battery mode: normal
[site  ] DEBUG 2024/04/20 14:31:48 ----
```

### What type of operating system are you running?

HomeAssistant Add-on

### Version

0.125.0

andig, 2024-04-20:
Battery control ausschalten!

closed 2024-04-20 by andig

itwuzel, 2024-04-20:
Danke für die schnelle Antwort!

Hmm, sorry bin noch nicht so firm...

Du meinst die Bezüge in der der **site** bzw. **meters** Konfiguration wieder auskommentieren?
Scheint geholfen zu haben.

Bedeutet ich kann die Batterienutzung so nicht steuern, oder?
Bug oder Future?

Gruß
Thomas

VolkerK62, 2024-04-21:
ich denke @andig meinte den Schalter in den Batterieeinstellungen (am unteren Rand).
https://docs.evcc.io/docs/features/battery#entladesperre-beim-schnelladen

itwuzel, 2024-04-24:
Ahh!

Das ist der Trick und hat das Verhalten wie erwartet geändert!
Es wird damit nicht mehr der Betriebsmodus geändert.

Wäre evtl. ein Tipp für die Doku oder FAQ

Vielen Dank
Thomas

andig, 2024-04-24:
Du meinst wir müssen "benutze den Schalter nicht" dokumentieren? Oder was fehlt da konkret?

itwuzel, 2024-04-24:
Evtl. einen Hinweis dass ein aktivieren der  Einstellung bei Q-Cells (und vermutlich auch bei Solax) den Betriebsmodus von "Backup Betrieb" auf Eigenverbrauch umstellt.

andig, 2024-04-24:
Gerne direkt PR fürs Template

itwuzel, 2024-05-01:
Sorry, PR == Pull Request?

Komme mit GitHub oder dem Arbeiten damit nicht zurecht, aber ist ja soweit alles gut.
Bin leider kein Programmierer der sich täglich damit auseinandersetzt.

Schönen 1. Mai Feiertag.

andig, 2024-05-01:
@itwuzel genau, Pull Request. Du kannst das Template https://github.com/evcc-io/evcc/blob/fd653ec838eeedfa40ed26835184649fd705e51a/templates/definition/meter/solax.yaml direkt hier auf Github bearbeiten und speichern. Sollte ganz einfach sein!