evcc-io/evcc #13517: CSV Export Ladevorgänge mit Gastfahrzeug

TobiasHuber1980, 2024-04-20:
### Describe the bug

Wenn man die CSV-Datei der Ladevorgänge herunterlädt, werden die Ladevorgänge als Gastfahrzeug in der CSV-Datei zwar aufgeführt, in der CSV ist jedoch kein Eintrag unter Fahrzeug. Sollte hier nicht auch Gastfahrzeug stehen?
Siehe CSV-Datei E3.
[session.csv](https://github.com/evcc-io/evcc/files/15048514/session.csv)
[](url)

### Steps to reproduce

1. CSV-Datei herunterladen und öffnen.
2.
3.
...

### Configuration details

```yaml
nicht benötigt
```

### Log details

```text
nicht benötigt
```

### What type of operating system are you running?

Linux

### Version

0.125.0

naltatis, 2024-04-20:
Gastfahrzeug bedeutet technisch für uns, dass kein konfiguriertes Fahrzeug zugeordnet ist. Daher ist das erstmal erwartetes Verhalten. 

Da Gastfahrzeug auch ein lokalisierter Text ist (im Gegensatz zu den Fahrzeugnamen) müsste der csv Output je nach Sprache dann auch angepasst sein. Das könnten wir machen. Dann wäre UI und CSV aus Nutzersicht konsistent. Ich find den aktuellen Stand (kein Name) aber auch nicht falsch.

Ich würde zu "lassen wir so" tendieren. @andig wie siehst du das?

andig, 2024-04-20:
So lassen 😌

TobiasHuber1980, 2024-04-20:
Ok kein Problem. War nur eine 💡

closed 2024-04-20 by TobiasHuber1980