evcc-io/evcc #13518: panic: runtime error: slice bounds out of range [1:0]

troem, 2024-04-20:
### Describe the bug

This configuration is running with version evcc/evcc:0.118.11

### Steps to reproduce

1. upgrade evcc docker container from version evcc/evcc:0.118.11 to latest
2. run the container
3. show logs "[panic runtime error: sli](panic: runtime error: slice bounds out of range [1:0])"

### Configuration details

```yaml
network:
  schema: http
  host: evcc.local
  port: 7070

interval: 30s

log: info

meters:
  - name: grid
    type: template
    template: sunspec-inverter
    modbus: tcpip
    id: 240
    usage: grid
    host: 192.168.178.77 # Hostname
    port: 502 # Port

  - name: pv
    type: template
    template: fronius-solarapi-v1
    usage: pv
    host: 192.168.178.77

chargers:
  - name: my_dummy1
    type: custom
    status: # charger status A..F
      source: script
      cmd: /bin/sh -c 'echo b'
    enabled: # charger enabled state (true/false or 0/1)
      source: script
      cmd: /bin/sh -c 'echo false'
    enable: # set charger enabled state (true/false or 0/1)
      source: script
      cmd: /bin/sh -c 'echo false'
    maxcurrent: # set charger max current (A)
      source: script

loadpoints:
  - title: loadpoint # display name for UI
    charger: my_dummy1 # Wallbox Referenz

site:
  title: Zuhause
  meters:
    grid: grid
    pv: pv
```

### Log details

```text
[main  ] INFO 2024/04/20 16:11:48 evcc 0.125.0
[main  ] INFO 2024/04/20 16:11:48 using config file: /etc/evcc.yaml
[main  ] INFO 2024/04/20 16:11:48 starting ui and api at :7070
[db    ] INFO 2024/04/20 16:11:49 using sqlite database: /root/.evcc/evcc.db
[site  ] INFO 2024/04/20 16:12:12 site config:
[site  ] INFO 2024/04/20 16:12:12   meters:      grid ✓ pv ✓ battery ✗
[site  ] INFO 2024/04/20 16:12:12     grid:      power ✓ energy ✓ currents ✓
[site  ] INFO 2024/04/20 16:12:12     pv 1:      power ✓ energy ✗ currents ✗
[lp-1  ] INFO 2024/04/20 16:12:12 loadpoint 1:
[lp-1  ] INFO 2024/04/20 16:12:12   mode:        off
[lp-1  ] INFO 2024/04/20 16:12:12   charger:     power ✗ energy ✗ currents ✗ phases ✗ wakeup ✗
[lp-1  ] INFO 2024/04/20 16:12:12   meters:      charge ✗
[lp-1  ] DEBUG 2024/04/20 16:12:12 phase timer inactive
[lp-1  ] DEBUG 2024/04/20 16:12:12 pv timer inactive
[script] DEBUG 2024/04/20 16:12:12 /bin/sh -c echo false: false
[site  ] DEBUG 2024/04/20 16:12:12 ----
[lp-1  ] DEBUG 2024/04/20 16:12:12 charge power: 0W
[site  ] DEBUG 2024/04/20 16:12:12 pv power: 5199W
[site  ] DEBUG 2024/04/20 16:12:13 grid meter: -4635W
[site  ] DEBUG 2024/04/20 16:12:14 grid powers: [-1707 -1353 -1615]W
[site  ] DEBUG 2024/04/20 16:12:15 grid currents: [-7.28 -6.73 -6.89]A
[site  ] DEBUG 2024/04/20 16:12:15 site power: -4635W
[script] DEBUG 2024/04/20 16:12:15 /bin/sh -c echo b: b
[lp-1  ] DEBUG 2024/04/20 16:12:15 charger status: B
[lp-1  ] INFO 2024/04/20 16:12:15 car connected
[lp-1  ] DEBUG 2024/04/20 16:12:15 pv timer elapse
[lp-1  ] DEBUG 2024/04/20 16:12:15 pv timer inactive
[script] DEBUG 2024/04/20 16:12:15 /bin/sh -c echo false: false
[lp-1  ] DEBUG 2024/04/20 16:12:15 pv charge current: 6.72A = 0A + 6.72A (-4635W @ 3p)
[lp-1  ] DEBUG 2024/04/20 16:12:15 site power -4635W <= 0W enable threshold
[lp-1  ] DEBUG 2024/04/20 16:12:15 pv enable in 0s
[lp-1  ] DEBUG 2024/04/20 16:12:15 pv enable timer elapsed
panic: runtime error: slice bounds out of range [1:0]

goroutine 15 [running]:
github.com/evcc-io/evcc/provider.(*Script).exec(0x5d200a0, {0x0, 0x0})
	github.com/evcc-io/evcc/provider/script.go:115 +0x474
github.com/evcc-io/evcc/provider.(*Script).IntSetter.func1(0x6)
	github.com/evcc-io/evcc/provider/script.go:224 +0xfc
github.com/evcc-io/evcc/charger.(*Charger).MaxCurrent(0x60e44e0, 0x6)
	github.com/evcc-io/evcc/charger/charger.go:169 +0x30
github.com/evcc-io/evcc/core.(*Loadpoint).setLimit(0x616e008, 0x4018000000000000)
	github.com/evcc-io/evcc/core/loadpoint.go:754 +0x15c
github.com/evcc-io/evcc/core.(*Loadpoint).Update(0x616e008, 0xc0b21b6660000000, 0x0, 0x0, 0x0, 0x3ff0000000000000, 0x0, 0x0)
	github.com/evcc-io/evcc/core/loadpoint.go:1655 +0xc9c
github.com/evcc-io/evcc/core.(*Site).update(0x6344128, {0x3230bd8, 0x616e008})
	github.com/evcc-io/evcc/core/site.go:790 +0x514
github.com/evcc-io/evcc/core.(*Site).Run(0x6344128, 0x5b5c340, 0x6fc23ac00)
	github.com/evcc-io/evcc/core/site.go:905 +0x2c8
github.com/evcc-io/evcc/cmd.runRoot.func4()
	github.com/evcc-io/evcc/cmd/root.go:284 +0x3c
created by github.com/evcc-io/evcc/cmd.runRoot in goroutine 1
	github.com/evcc-io/evcc/cmd/root.go:283 +0x1bf0
```

### What type of operating system are you running?

Docker container

### Version

evcc version 0.125.0

andig, 2024-04-21:
Closed in https://github.com/evcc-io/evcc/commit/1797fcce6452e09d047dfd0b9ef0d39c19f1e86b

closed 2024-04-21 by andig

troem, 2024-04-21:
Super, thank you for the quick reaction and your fix ... I added the "cmd" in my config and it is working now