evcc-io/evcc #13531: Phasenwechsel funktioniert nicht zuverlässig  , manchmal braucht es einen evcc restart

67-Peter, 2024-04-21:
### Describe the bug

Grundsätzlich läuft der 1P -  3P Phasenwechsel , heute lief der Phasenwechsel den ganzen Morgen. Plötzlich hört er dann aber auf und bleibt auf 1P stehen obwohl genügend PV Leistung vom Dach kommt.
Auch wenn ich jetzt auf SCHNELL stelle bleibt es bei 1P.
Wenn ich evcc restarte  lädt mein ID.4 wieder mit 3P:

Das Problem tritt immer wieder auf.

Ich habe eine Keba P30x mit einem 2-Phasenschütz.

### Steps to reproduce

1. Warten bis 1/3 P nicht mehr automatisch läuft  (PV MODE)
2. PV -> SCHNELL
3. sudo systemctl restart evcc

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

plant: xxxxxxxxxxxxxx

sponsortoken: xxxxxxxxxxxx
interval: 10s # control cycle interval

meters:
- name: grid1
  type: template
  template: growatt-hybrid-tlxh
  usage: grid
  modbus: tcpip
  id: 1
  host: 192.168.178.11
  port: 502
- name: Solar
  type: template
  template: growatt-hybrid-tlxh
  id: 1
  host: 192.168.178.11
  port: 502
  usage: pv
  modbus: rs485tcpip
- name: battery1
  type: template
  template: growatt-hybrid-tlxh
  id: 1
  host: 192.168.178.11
  port: 502
  usage: battery
  modbus: tcpip
  capacity: 10

chargers:
- type: template
  template: keba-modbus
  host: 192.168.178.108
  name: mykeba
  modbus: tcpip
  id: 255
  port: 502

[... 130 lines cut ...]
Apr 21 13:39:56 evcc evcc[1999]: [lp-1  ] DEBUG 2024/04/21 13:39:56 wake-up timer: start
Apr 21 13:40:05 evcc evcc[1999]: [site  ] DEBUG 2024/04/21 13:40:05 ----
Apr 21 13:40:05 evcc evcc[1999]: [lp-1  ] DEBUG 2024/04/21 13:40:05 charge power: 0W
Apr 21 13:40:05 evcc evcc[1999]: [site  ] DEBUG 2024/04/21 13:40:05 pv power: 11086W
Apr 21 13:40:06 evcc evcc[1999]: [site  ] DEBUG 2024/04/21 13:40:06 battery soc: 94%
Apr 21 13:40:06 evcc evcc[1999]: [site  ] DEBUG 2024/04/21 13:40:06 battery power: -2836W
Apr 21 13:40:06 evcc evcc[1999]: [site  ] DEBUG 2024/04/21 13:40:06 grid meter: -7826W
Apr 21 13:40:06 evcc evcc[1999]: [site  ] DEBUG 2024/04/21 13:40:06 site power: -10562W
Apr 21 13:40:06 evcc evcc[1999]: [lp-1  ] DEBUG 2024/04/21 13:40:06 charge currents: [0 0 0]A
Apr 21 13:40:06 evcc evcc[1999]: [lp-1  ] DEBUG 2024/04/21 13:40:06 charge total import: 4042.387kWh
Apr 21 13:40:06 evcc evcc[1999]: [lp-1  ] DEBUG 2024/04/21 13:40:06 charger status: B
Apr 21 13:40:06 evcc evcc[1999]: [lp-1  ] DEBUG 2024/04/21 13:40:06 pv charge current: 45.9A = 0A + 45.9A (-10562W @ 1p)
Apr 21 13:40:06 evcc evcc[1999]: [lp-1  ] DEBUG 2024/04/21 13:40:06 pv timer reset
Apr 21 13:40:06 evcc evcc[1999]: [lp-1  ] DEBUG 2024/04/21 13:40:06 pv timer inactive
Apr 21 13:40:06 evcc evcc[1999]: [lp-1  ] DEBUG 2024/04/21 13:40:06 max charge current: 16A
Apr 21 13:40:15 evcc evcc[1999]: [site  ] DEBUG 2024/04/21 13:40:15 ----
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.125.0

andig, 2024-04-21:
Das trifft sich gut. Das Problem haben wir analog für die Alfen auch vor ein paar Minuten gelöst.

closed 2024-04-21 by andig

67-Peter, 2024-04-21:
Super , dann warte ich mal ab.

stiefel46, 2024-04-26:
Habe das gleiche Problem. Bei mir hilft auch kein Neustart. 3-phasig läuft nur durch Brücken der X2-Kontakte. Für einphasiges Überschussladen muss ich die Brücke wieder entfernen.

Bis vor ein paar Tagen lief alles wunderbar.

andig, 2024-04-26:
@stiefel46 die Aussage ist nicht verwertbar:
- was hast Du geändert?
- hast Du eine neue Version? 
- auf welcher ging es noch?
- gibts ein Logfile?
Mach bitte ein Issue. Hier sind wir fertig!