evcc-io/evcc #13535: Sehr häufig: charger logic error: phases mismatch (got 0, expected 1)

userwithoutpassword, 2024-04-21:
### Describe the bug

Ich habe eine Easee Wallbox und bekomme sehr häufig die Meldung 

charger logic error: phases mismatch (got 0, expected 1)

Ladung steht auf "Solar" und 1phasig über das UI ausgewählt.

Hatte es "früher" auch mal in der yaml drin stehen und dort mittlerweile gelöscht, aber dennoch taucht die Meldung in regelmäßigen Abständen auf.

### Steps to reproduce

siehe oben

### Configuration details

```yaml
siehe oben
```

### Log details

```text
siehe oben
```

### What type of operating system are you running?

Linux

andig, 2024-04-21:
Ohne Log nix zu machen

closed 2024-04-21 by andig

userwithoutpassword, 2024-04-21:
Ja letztemal als ich ein log erstellt habe ging das Tesla Token anschließend nicht mehr...