evcc-io/evcc #13538: Support structured logging

andi0b, 2024-04-21:
**Is your feature request related to a problem? Please describe.**

I need to analyze quite a big amount of evcc logs, because of Tesla token issues. 

The db trace logs are multline log messages which are hard to process with common log stacks (I'm using Grafana Agent / Loki / Grafana on Kubernetes).

**Describe the solution you'd like**

Support structured logging in JSON (or similar) format. Make it an optional configuration parameter, so people who are not using a logging pipeline can still read their logs.

Take those log messages for example

```log
[site  ] DEBUG 2024/04/21 22:11:14 site power: 1782W
[db    ] TRACE 2024/04/21 22:10:45
		SELECT SUM(charged_kwh * price_per_kwh) / SUM(charged_kwh) AS AvgPrice 
		FROM sessions 
		WHERE finished >= "1924-05-16 22:10:45.208" 
		AND charged_kwh > 0 
		AND price_per_kwh IS NOT NULL 1 <nil>
```

expected structured log:
```jsonl
{ "timestamp": "2024-04-21T20:11:14Z", "level": "debug", "message": "site power: 1782W" }
{ "timestamp": "2024-04-21T20:10:45Z", "level": "trace", "message": "SELECT SUM(charged_kwh * price_per_kwh) / SUM(charged_kwh) AS AvgPrice \nFROM sessions\nWHERE finished >= \"1924-05-16 22:10:45.208\" ..." }
```

this could be enabled in the evcc.yaml for example with this kind of parameter:
```yaml
# default (current behavior)
logFormat: plain

# json logging
logFormat: json
```

The format itself is not so important as long as it's possible to process it with common logging stacks. timestamp, level and message should be separate fields. Line breaks should be escaped or quoted.

Additional structured logging fields would be a nice thing to have, but more like a nice to have, see [slog](https://pkg.go.dev/golang.org/x/exp/slog) docs for some examples.

**Describe alternatives you've considered**

Alternatives are fixing it with complex logshipper configurations, for example [Promtail multiline stage](https://grafana.com/docs/loki/latest/send-data/promtail/stages/multiline/)

**Additional context**

Grafana (Agent/Loki) screenshot of current logs, the problem is they are not recognized as one multi-line log message, instead as multiple single line log messages. The lines might even get sorted the other way (newest on top) and log messages completely mixed up. Filtering/querying might get quite the hassle.

[image]

andig, 2024-04-23:
> The db trace logs are multline log messages which are hard to process with common log stacks (I'm using Grafana Agent / Loki / Grafana on Kubernetes).

@andi0b that's a developer issue. We're generally aiming for broader audience, not devs. Still, structured logging is in the backlog but not a priority. Closing here as "will do later when time". For transparency: biggest obstacle if that we'll need to change about every single file in the codebase...

andig, 2024-04-23:
PS.: if you wanna help with that please contact us on Slack!

closed 2024-04-23 by andig

andi0b, 2024-04-23:
On the first glance it looked like a simple change. Will look into it, and might drop a PR. I have zero Go experience, so it might take me a few hours.

andi0b, 2024-04-23:
@andig you're right, your logging library (jWalterWeatherman) has a hardcoded log format, I thought I could tweak that, never seen a logger with hardcoded format before. I will think about possible solutions.

Changing all the logger invocations probably isn't the best way to move forward. Although it might make sense to optimize the DB logging.

andig, 2024-04-23:
I‘m looking into slog but still need to find an approach that will a) allow stepwise transition and b) still accept printf messages. Happy to collaborate on this.