evcc-io/evcc #13541: Backup-Funktion (Import/Export)

TobiasHuber1980, 2024-04-22:
Ist mir gerade als Idee gekommen. 

Wie wäre es mit einer Backup-Funktion in evcc?

1. Als z.B. die evcc.yaml & evcc.db als Datei zum Download. 
2. Bzw. noch weiter gedacht eine Import/Export-Funktion?
3. Oder noch weiter Gedacht die Möglichkeit die "Datensicherung" auf einer Cloud-Lösung zu speichern? Idealerweise direkt bei evcc online.

andig, 2024-04-23:
Kann man sicher machen, ist aber keine Priorität. Perspektivisch geht es im das Kopieren einer Datei- das sollte handhabar sein 👍🏻

closed 2024-04-23 by andig

dachris1337, 2025-05-07:
Nachdem es bei Backup von evcc "nur" um .yaml und .db geht wäre es doch hilfreich eine Sicherung in die Cloud (google drive o.ä.) anlegen zu können