evcc-io/evcc #13543: go-E & Renault Zoe: Not correctly phase switching phases and using grid power - charge power must not be negative: -374

Tigraine, 2024-04-22:
### Describe the bug

With v0.123.2 I was seeing a bunch of `charge power must not be negative: -374` errors in the logs so I upgraded to v0.125.0 to see if the issue was resolved there (didn't want to start debugging something that might be resolved already).

After the upgrade I noticed this very weird behaviour that made my Charge power jump all over the place that seems to be related to evcc not switching between phases correctly and instead oscillating the Charge Current between min/max.
It seems that when I have *just enough* power to do 3p charging (so something around ~4.5kW) the following happens:

[image]

As you can see (this is from the associated Influx DB) the Charger starts jumping between -374 charge power and then full 11 kW. Which then immediately causes EVCC to throttle it down again and we're back to -374. 
(Don't ask me about the -374 - it's always exactly -374 - also before the v0.125.0 upgrade)
I resolved the oscillation by manually setting the phases to 1p in the EVCC UI at 13:09

This makes the Automatic Phase switching really unusable right now and I am not sure if the car was even charging during the time of the oscillation.

Also notice that it was correctly charging at 3p before and then just randomly started the jump. No idea why - as the PV power was mostly stable before that)
[image]

Also the only variation I see in Battery power and Grid Power is then related to evcc starting to go crazy and oscillating between 11kW charging and no-charging.

I did spend some time investigating the logs and the pattern looks like this:

`WARN 2024/04/22 13:01:27 charge power must not be negative: -374` start occuring in the logs.
I also see that the PV current starts being all over the place:
```
DEBUG 2024/04/22 13:01:08 pv charge current: 14.1A = 7A + 7.14A (-4929W @ 3p)
DEBUG 2024/04/22 13:01:17 pv charge current: 7.72A = 14A + -6.28A (4331W @ 3p)
DEBUG 2024/04/22 13:01:28 pv charge current: 14.8A = 7A + 7.84A (-5412W @ 3p)
DEBUG 2024/04/22 13:01:38 pv charge current: 6.44A = 14A + -7.56A (5217W @ 3p)
```

What happened here was that EVCC started oscillating the `max charge current` between 6A and 13A on each refresh - all while the `phase scale1p in 5m0s` keeps getting reset:

```
Apr 22 13:08:18 raspberrypi evcc[768059]: [lp-1  ] DEBUG 2024/04/22 13:08:18 start phase scale1p timer
Apr 22 13:08:18 raspberrypi evcc[768059]: [lp-1  ] DEBUG 2024/04/22 13:08:18 phase scale1p in 5m0s
Apr 22 13:08:28 raspberrypi evcc[768059]: [lp-1  ] DEBUG 2024/04/22 13:08:28 phase timer inactive
Apr 22 13:08:38 raspberrypi evcc[768059]: [lp-1  ] DEBUG 2024/04/22 13:08:38 start phase scale1p timer
Apr 22 13:08:38 raspberrypi evcc[768059]: [lp-1  ] DEBUG 2024/04/22 13:08:38 phase scale1p in 5m0s
Apr 22 13:08:48 raspberrypi evcc[768059]: [lp-1  ] DEBUG 2024/04/22 13:08:48 phase timer inactive
Apr 22 13:08:58 raspberrypi evcc[768059]: [lp-1  ] DEBUG 2024/04/22 13:08:58 start phase scale1p timer
Apr 22 13:08:58 raspberrypi evcc[768059]: [lp-1  ] DEBUG 2024/04/22 13:08:58 phase scale1p in 5m0s
```

(before any confusion as of the config happens - I had the `disable:` timer set to 5m at the time but removed it since as it didn't change anything. Also the phases were set to "automatic" via the UI - the config has it at `1`)

Hope this helps in maybe uncovering an issue - let me know if I can help debug this further.
Switching to 1p manually fixed the issue.
But this should not really be happening.
I'll try downgrading to 0.123.2 now and see if the issue persists.

PS: I know a interval < 30 can lead to issues. But i've been running this setup now for months and never had an issue on 0.123.2 that's why I left the refresh interval at the `10s`

### Steps to reproduce

hopefully provided all the details in the bug description.

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: raspberrypi.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

[... 233 lines cut ...]
Apr 22 13:01:47 raspberrypi evcc[768059]: [lp-1  ] DEBUG 2024/04/22 13:01:47 charge power: -372W
Apr 22 13:01:47 raspberrypi evcc[768059]: [lp-1  ] WARN 2024/04/22 13:01:47 charge power must not be negative: -372
Apr 22 13:01:47 raspberrypi evcc[768059]: [site  ] DEBUG 2024/04/22 13:01:47 pv power: 5272W
Apr 22 13:01:47 raspberrypi evcc[768059]: [site  ] DEBUG 2024/04/22 13:01:47 battery soc: 100%
Apr 22 13:01:47 raspberrypi evcc[768059]: [site  ] DEBUG 2024/04/22 13:01:47 battery power: 165W
Apr 22 13:01:47 raspberrypi evcc[768059]: [site  ] DEBUG 2024/04/22 13:01:47 grid meter: -5426W
Apr 22 13:01:47 raspberrypi evcc[768059]: [site  ] DEBUG 2024/04/22 13:01:47 grid currents: [-10.6 -10.8 -10.5]A
Apr 22 13:01:48 raspberrypi evcc[768059]: [site  ] DEBUG 2024/04/22 13:01:48 site power: -5261W
Apr 22 13:01:48 raspberrypi evcc[768059]: [lp-1  ] DEBUG 2024/04/22 13:01:48 charge currents: [6.54 6.61 6.61]A
Apr 22 13:01:48 raspberrypi evcc[768059]: [lp-1  ] DEBUG 2024/04/22 13:01:48 detected active phases: 3p
Apr 22 13:01:48 raspberrypi evcc[768059]: [lp-1  ] DEBUG 2024/04/22 13:01:48 charge total import: 1717.314kWh
Apr 22 13:01:48 raspberrypi evcc[768059]: [lp-1  ] DEBUG 2024/04/22 13:01:48 charger status: C
Apr 22 13:01:48 raspberrypi evcc[768059]: [lp-1  ] DEBUG 2024/04/22 13:01:48 phase timer inactive
Apr 22 13:01:48 raspberrypi evcc[768059]: [lp-1  ] DEBUG 2024/04/22 13:01:48 pv charge current: 13.6A = 6A + 7.62A (-5261W @ 3p)
Apr 22 13:01:48 raspberrypi evcc[768059]: [lp-1  ] DEBUG 2024/04/22 13:01:48 max charge current: 13A
Apr 22 13:01:57 raspberrypi evcc[768059]: [site  ] DEBUG 2024/04/22 13:01:57 ----
```

### What type of operating system are you running?

Linux

### Version

v0.125.0

VolkerK62, 2024-04-22:
Negative charge power is a bug in go-E.  https://github.com/evcc-io/evcc/issues/13378
it is fixed in the actual nightly version.

The other problem is ZOE. In 3p it doesn´t like charge current below 8A (older model below 10A).

andig, 2024-04-23:
> After the upgrade I noticed this very weird behaviour that made my Charge power jump all over the place 

Is there a log from 0.123 that does not have these jumps? Otherwise I'd follow with @VolkerK62 that >8A min current should fix it.

RTTTC, 2024-04-29:
> Also the phases were set to "automatic" via the UI - the config has it at 1

I would comment out any settings in yaml, that you are planning to change in UI

Tigraine, 2024-04-29:
Thank you very much for the replies. Good to hear this is fixed already.
Also I didn't know that about the Zoe - thank you for that.
I've now changed the phases to fixed 1p.
Gotta say - been driving the Zoe for 2 years now and the longer you get to live with that car the more it annoys you to no end. (No passenger heating while fast-charging takes the cake though)

Thanks everyone - I'll close the bug.

closed 2024-04-29 by Tigraine

bockha, 2024-05-04:
With the current Renault Kangoo E-Tech I have the same trouble. It will not charge on 3p if the wallbox limits the current below 8A. If the wallbox sets the limit to 8A, the car will start charging with 4kW.

hpbock, 2024-05-04:
With the current Renault Kangoo E-Tech I have the same trouble. It will not charge on 3p if the wallbox limits the current below 8A. If the wallbox sets the limit to 8A, the car will start charging with 4kW.

andig, 2024-05-04:
This issue has been closed as Go-E bug. Open new issue if problem persists.